margelo/react-native-vision-camera #1101: 🐛  This declaration is opt-in and its usage should be marked with

AmirHMousavi, 2022-06-21:
### What were you trying to do?

Building the app for Android with` ./gradlew build` command

Environment data:
```sh
 "react": "17.0.2",
 "react-native": "0.68.2",
 
 # gradle
 
 buildscript {
    ext {
        buildToolsVersion = "33.0.0"
        minSdkVersion = 21
        compileSdkVersion = 31
        targetSdkVersion = 31

        if (System.properties['os.arch'] == "aarch64") {
            // For M1 Users we need to use the NDK 24 which added support for aarch64
            ndkVersion = "24.0.8215888"
        } else {
            // Otherwise we default to the side-by-side NDK version from AGP.
            ndkVersion = "21.4.7075529"
        }
    }
  }
  
  distributionUrl=https\://services.gradle.org/distributions/gradle-7.4.2-all.zip
#  app/build.gradle
 android{
    defaultConfig {
          multiDexEnabled true
          
# machine
MacbookPro , Apple M1 Pro
macOS Monterey 12.4
 ```
 

### What happened instead?

> Task :react-native-vision-camera:lintDebug FAILED

### Relevant log output

```shell
> Task :react-native-vision-camera:lintDebug FAILED
Lint found 3 errors, 0 warnings. First failure:

/<project-path>/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView.kt:431: Error: This declaration is opt-in and its usage should be marked with

The full lint text report is located at:
  /<project-path>/node_modules/react-native-vision-camera/android/build/intermediates/lint_intermediate_text_report/debug/lint-results-debug.txt
w: Detected multiple Kotlin daemon sessions at build/kotlin/sessions

the lint-result.debug.txt content

/<project-path>/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView.kt:431: Error: This declaration is opt-in and its usage should be marked with
'@androidx.camera.camera2.interop.ExperimentalCamera2Interop' or '@OptIn(markerClass = androidx.camera.camera2.interop.ExperimentalCamera2Interop.class)' [UnsafeOptInUsageError from jetified-annotation-experimental-1.1.0]
            Camera2Interop.Extender(previewBuilder)
                           ~~~~~~~~
/<project-path>/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView.kt:432: Error: This declaration is opt-in and its usage should be marked with
'@androidx.camera.camera2.interop.ExperimentalCamera2Interop' or '@OptIn(markerClass = androidx.camera.camera2.interop.ExperimentalCamera2Interop.class)' [UnsafeOptInUsageError from jetified-annotation-experimental-1.1.0]
              .setCaptureRequestOption(CaptureRequest.CONTROL_AE_TARGET_FPS_RANGE, Range(fps, fps))
               ~~~~~~~~~~~~~~~~~~~~~~~
/<project-path>/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView.kt:433: Error: This declaration is opt-in and its usage should be marked with
'@androidx.camera.camera2.interop.ExperimentalCamera2Interop' or '@OptIn(markerClass = androidx.camera.camera2.interop.ExperimentalCamera2Interop.class)' [UnsafeOptInUsageError from jetified-annotation-experimental-1.1.0]
              .setCaptureRequestOption(CaptureRequest.SENSOR_FRAME_DURATION, frameDuration)
               ~~~~~~~~~~~~~~~~~~~~~~~

   Explanation for issues of type "UnsafeOptInUsageError":
   This API has been flagged as opt-in with error-level severity.

   Any declaration annotated with this marker is considered part of an
   unstable or
   otherwise non-standard API surface and its call sites should accept the
   opt-in
   aspect of it either by using @OptIn or by being annotated with that marker
   themselves, effectively causing further propagation of the opt-in aspect.

   Vendor: Android Open Source Project (jetified-annotation-experimental-1.1.0)
   Identifier: jetified-annotation-experimental-1.1.0
   Feedback: https://issuetracker.google.com/issues/new?component=192731

3 errors, 0 warnings
```

### Device

Android 

### VisionCamera Version

2.13.5

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

chukiatt, 2022-11-09:
i have same

chukiatt, 2022-11-09:
are you solved this ?

finger-guns, 2022-12-13:
no updates on this one ?

flybluewolf, 2022-12-15:
modify android/build.gradle file
`allprojects {
    // append here
    afterEvaluate {
        if (getPlugins().hasPlugin('android') ||
            getPlugins().hasPlugin('android-library')) {
            configure(android.lintOptions) {
                abortOnError false
                checkReleaseBuilds false
            }
        }
    }
}`
and run './gradlew clean'

malinjr07, 2023-02-17:
> modify android/build.gradle file
>
> ```
> allprojects {
>     // append here
>     afterEvaluate {
>         if (getPlugins().hasPlugin('android') ||
>             getPlugins().hasPlugin('android-library')) [...]

Thank you @flybluewolf . Your solution worked. Here's my package.json:

`{
  "dependencies": {
    "@react-native-masked-view/masked-view": "^0.2.8",
    "@react-navigation/native": "^6.1.4",
    "@react-navigation/stack": "^6.3.14",
    "react": "18.2.0",
    "react-native": "0.71.3",
    "react-native-gesture-handler": "^2.9.0",
    "react-native-safe-area-context": "^4.5.0",
    "react-native-screens": "^3.20.0",
    "react-native-view-shot": "^3.5.0",
    "react-native-vision-camera": "^2.15.4",
    "react-native-zephyr": "^1.1.0"
  },
  "devDependencies": {
    "@babel/core": "^7.20.0",
    "@babel/preset-env": "^7.20.0",
    "@babel/runtime": "^7.20.0",
    "@react-native-community/eslint-config": "^3.2.0",
    "@tsconfig/react-native": "^2.0.2",
    "@types/jest": "^29.2.1",
    "@types/react": "^18.0.24",
    "@types/react-test-renderer": "^18.0.0",
    "babel-jest": "^29.2.1",
    "eslint": "^8.19.0",
    "jest": "^29.2.1",
    "metro-react-native-babel-preset": "0.73.7",
    "prettier": "^2.4.1",
    "react-test-renderer": "18.2.0",
    "typescript": "4.8.4"
  },
  "jest": {
    "preset": "react-native"
  }
}`

Update `AndroidManifest.xml`:

`<uses-permission android:name="android.permission.CAMERA" />`

Finally, my `build.gradle` file (directory: `android/build.gradle`):

` buildscript {
    ext {
        buildToolsVersion = "33.0.0"
        kotlinVersion = "1.7.0"
        minSdkVersion = 21
        compileSdkVersion = 33
        targetSdkVersion = 33

        // We use NDK 23 which has both M1 support and is the side-by-side NDK version from AGP.
        ndkVersion = "23.1.7779620"
    }
    repositories {
        google()
        mavenCentral()
    }
    dependencies {
        classpath("com.android.tools.build:gradle:7.3.1")
        classpath("com.facebook.react:react-native-gradle-plugin")
    }
    allprojects {
    // append here
    afterEvaluate {
        if (getPlugins().hasPlugin('android') ||
            getPlugins().hasPlugin('android-library')) {
            configure(android.lintOptions) {
                abortOnError false
                checkReleaseBuilds false
            }
        }
    }
}
}`

Note: I have added `kotlinVersion = "1.7.0"` and 

`allprojects {
    // append here
    afterEvaluate {
        if (getPlugins().hasPlugin('android') ||
            getPlugins().hasPlugin('android-library')) {
            configure(android.lintOptions) {
                abortOnError false
                checkReleaseBuilds false
            }
        }
    }
}`

In the **gradle.build** file, nothing else.

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

closed 2023-09-30 by mrousavy