margelo/react-native-vision-camera #1102: 🐛 Camara not closed on Android when container is disposed

fleuverouge, 2022-06-22:
### What were you trying to do?

I need to use the camera for other tasks (different libraries). After capturing photos using react-native-vision-camera, the app navigates back to the previous screen (hence the camera view's container is unmounted).

### Reproduceable Code

```tsx
<Camera
                            ref={cameraRef}
                            style={StyleSheet.absoluteFill}
                            device={device}
                            photo={true}
                            isActive={isAppForeground && isFocused}
                            orientation="portrait"
                            onError={onCameraError}
                        />
```

### What happened instead?

I found out that after navigating back, the CameraService is still running, which blocks me from using it (getting CameraService::connect X (PID 3140) rejected (existing client(s) with higher priority)). The logs below keep being printed.

### Relevant log output

```shell
Camera2-FrameProcessorBase
FrameProcessorBase: Error waiting for new frames: Connection timed out (-110)
```

### Device

Samsung Galaxy A8

### VisionCamera Version

2.13.3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

fleuverouge, 2022-06-23:
Hi @AmirHMousavi I don't think there's no different to my code:

`
import { useFocusEffect, useIsFocused } from '@react-navigation/native';
import useIsAppForeground from '@hooks/useIsAppForeground';
...
const isFocused = useIsFocused();
const isAppForeground = useIsAppForeground();
...
<Camera
                            ref={cameraRef}
                            style={StyleSheet.absoluteFill}
                            device={device}
                            photo={true}
                            isActive={isAppForeground && isFocused}
                            orientation="portrait"
                            onError={onCameraError}
                        />
`

fleuverouge, 2022-06-27:
Update: looks like react-navigation/native doesn't turn off isFocused when the navigating back. I had to listen to 'beforeRemove' to update the value:

```
  useEffect(() => {
        const unsubscribeFocused = navigation.addListener('focus', () => {
            console.log('[Camera] Page focus');
            setIsScreenFocused(true);
        });

        const unsubscribeBlur = navigation.addListener('blur', () => {
            console.log('[Camera] Page blur');
            setIsScreenFocused(false);
        });
        const unsubsribeRemove = navigation.addListener('beforeRemove', e => {
            console.log('[Camera] Page removed');
            setIsScreenFocused(false);
            if (Platform.OS === 'android') {
                e.preventDefault();
                setTimeout(() => {
                    navigation.dispatch(e.data.action);
                }, 500);
            }
        });

        return () => {
            unsubscribeFocused();
            unsubscribeBlur();
            unsubsribeRemove();
        };
    }, [navigation]);
```

closed 2022-06-27 by fleuverouge

mrousavy, 2023-11-18:
Hey! I just found out that I really forgot to close and dispose the locked Camera resources, so I just fixed that in this PR: https://github.com/mrousavy/react-native-vision-camera/pull/2174 
Now the Camera fully & synchronously closes all resources (CameraSession, CameraDevice, OpenGL context, Video & Photo outputs, Photo Synchronizer) once the view gets removed from the React view hierarchy ("unmounted"), and things like Flash, Torch, NFC, and other Camera components should be working again.

There is still a small issue that causes once Camera component to turn into a blackscreen when navigating back and forth between two Camera components, that's a pretty rare edge case but I will still try to fix that soon when I have some free time.

If you appreciate my time, expertise and dedication to this project, pleas [💖 consider sponsoring me on GitHub 💖](https://github.com/sponsors/mrousavy) to support the development of this project.