margelo/react-native-vision-camera #1104: I got this error "Use of undeclared identifier 'jsi'; did you mean 'facebook::jsi'" After upgraded from React Native 0.68.2 to 0.69

sunchanras, 2022-06-24:
### Question

I created a new React Native 0.69 app for reproducing purpose and installed only react-native-vision-camera library for testing and I got this "Use of undeclared identifier 'jsi'; did you mean 'facebook::jsi'"

### What I tried

package-lock.json && rm -rf yarn.lock && rm -rf node_modules
rm -rf ios/Podfile.lock && rm -rf ios/Pods
npm i  # or "yarn"
cd ios && pod repo update && pod update && pod install

### VisionCamera Version

2.13.5

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

ricardonazario, 2022-06-24:
I'm having the same issue after upgrading from react-native 0.67.2 to 0.69.0.

mm-webx, 2022-06-29:
[image]
I have other issue with 0.69.1

junioroga, 2022-07-01:
> [image]
> I have other issue with 0.69.1

Same problem here

icc-guerrero, 2022-07-02:
> > [image]
> > I have other issue with 0.69.1
> 
> Same problem here

Seems to be 0.69.1, not vision camera: https://github.com/facebook/react-native/issues/34102

dminkovsky, 2022-07-05:
Similar/same issue here https://github.com/Shopify/react-native-skia/pull/629. Solved temporarily in my codebase by putting in the `facebook::` namespace qualifier in JSConsoleHelper.mm. Also had to copy the bridging headers.

demogar, 2022-09-14:
If it's helpful to anyone, I had to create a local patch meanwhile this gets resolved.  In order to do that, you can always use [patch-package)(https://www.npmjs.com/package/patch-package).  Just add to the `JSConsoleHelper.mm`, as described by @dminkovsky [here)(https://github.com/mrousavy/react-native-vision-camera/issues/1104#issuecomment-1174467006) the `facebook::` namespace.

I was even able to fix it for an `expo eas build`.

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

closed 2023-09-30 by mrousavy