margelo/react-native-vision-camera #1110: 🐛 Execution failed for task ':react-native-vision-camera:compileDebugKotlin'.

Noma98, 2022-06-29:
### What were you trying to do?

When I run `npm i react-native-vision-camera` and `npm run android`, the error `':react-native-vision-camera:compileDebugKotlin'` appears.

I have downgraded the react-native-vision-camera to 2.13.0 and raised the react-native version to 0.68.2, but it still doesn't work.

### Reproduceable Code

```tsx
Gradle JDK : Android Studio default JDK, version 11.0.10
--------------------------------------------------------
* package.json
react-native: ^0.65.1
react-native-vision-camera: ^2.13.5
--------------------------------------------------------
*gradle-wrapper.properties
distributionBase=GRADLE_USER_HOME
distributionPath=wrapper/dists
distributionUrl=https\://services.gradle.org/distributions/gradle-6.7.1-all.zip
zipStoreBase=GRADLE_USER_HOME
zipStorePath=wrapper/dists
-------------------------------------------------------------------
*build.gradle(android)
// Top-level build file where you can add configuration options common to all sub-projects/modules.

buildscript {
    ext {
        RNNKotlinVersion = "1.3.61"
        buildToolsVersion = "30.0.2"
        minSdkVersion = 21
        compileSdkVersion = 31
        targetSdkVersion = 31
        ndkVersion = "21.4.7075529"
        versions = [
            'gradle'                    : '4.2.2',
            'kotlin'                    : '1.4.10',
        ]
        androidXCore = "1.7.0"
    }
    repositories {
        google()
        jcenter()
    }
    dependencies {
        classpath("com.android.tools.build:gradle:${versions.gradle}")
        classpath "org.jetbrains.kotlin:kotlin-gradle-plugin:${versions.kotlin}"
        // NOTE: Do not place your application dependencies here; they belong
        // in the individual module build.gradle files
        classpath 'com.google.gms:google-services:4.3.10'
    }
}

allprojects {
    repositories {
        mavenLocal()
        maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url("$rootDir/../node_modules/react-native/android")
        }
        maven {
            // Android JSC is installed from npm
            url("$rootDir/../node_modules/jsc-android/dist")
        }

        google()
        jcenter()
        flatDir {
            dirs 'libs'
        }
        maven { url 'https://www.jitpack.io' }
[... 293 lines cut ...]
info Run CLI with --verbose flag for more details.

### Relevant log output

```shell
e: /Users/jinheui/hjh/success/ktp/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/utils/withPromise.kt: (14, 13): Class 'kotlin.Unit' was compiled with an incompatible version of Kotlin. The binary version of its metadata is 1.6.0, expected version is 1.4.0.
The class is loaded from /Users/jinheui/.gradle/caches/transforms-2/files-2.1/dd29d72351235421935757ac7419235d/jetified-kotlin-stdlib-1.6.10.jar!/kotlin/Unit.class

/Users/jinheui/.gradle/caches/transforms-2/files-2.1/6adc402019d26e4e20d80e41f02ced82/jetified-kotlin-stdlib-common-1.6.10.jar!/META-INF/kotlin-stdlib-common.kotlin_module: Module was compiled with an incompatible version of Kotlin. The binary version of its metadata is 1.6.0, expected version is 1.4.0.
```

### Device

Pixel 2 API 29 (virtual device)

### VisionCamera Version

2.13.5

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

tarikko, 2022-08-02:
I have a similar problem, it was compiling fine last month and now it fails all the time, it is also possible that something in my Linux environment changed...

tarikko, 2022-08-04:
An update, I completely deleted my ndk and cmake compiler from the android sdk folder and reinstalled them from android studio, It was a really frustrating problem, but anyway that fixed it.
note: I installed ndk version 21.4.7075529 and 24.0.8215888 and all cmake versions.

hope this helps !!

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy