margelo/react-native-vision-camera #1117: ❓

drbarazi, 2022-07-03:
### Question

is there any option to automatically stop video by file size limit like fileSizeLimit in react native camera?

### VisionCamera Version

2.13.5

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-07-04:
Hi, no, there is not (yet).

closed 2022-07-04 by mrousavy

drbarazi, 2022-07-04:
okey, thanks for reply :)

rodrigobertolottizinsp, 2022-09-08:
Hi! Any plans on adding this?

mrousavy, 2022-09-12:
Hi! No, I don't need this feature in my apps. If you want, I can develop the feature for you by our (https://margelo.io) normal hourly rates, contact us at hello@margelo.io. :)