margelo/react-native-vision-camera #1118: 🐛 Error building in 0.69.1: "Expected directory 'node_modules/react-native/android' to contain exactly one file, however, it contains more than one file"

viniciusyoshioka, 2022-07-03:
### What were you trying to do?

After upgrading react-native from 0.68.2 to 0.69.1, I try to build to test if there was no erros in the upgrade.

### What happened instead?

The build failed. Even after update the library to the latest version, the error continued.

### Relevant log output

```shell
> Configure project :react-native-vision-camera
VisionCamera: node_modules/ found at: /home/user/Área de Trabalho/docscan/node_modules
VisionCamera: Building with Hermes...
Warning: The 'kotlin-android-extensions' Gradle plugin is deprecated. Please use this migration guide (https://goo.gle/kotlin-android-extensions-deprecation) to start working with View Binding (https://developer.android.com/topic/libraries/view-binding) and the 'kotlin-parcelize' plugin.
VisionCamera: Frame Processors are enabled! Building C++ part...
WARNING:DSL element 'dexOptions' is obsolete and should be removed.
It will be removed in version 8.0 of the Android Gradle plugin.
Using it has no effect, and the AndroidGradle plugin optimizes dexing automatically.

Deprecated Gradle features were used in this build, making it incompatible with Gradle 8.0.

You can use '--warning-mode all' to show the individual deprecation warnings and determine if they come from your own scripts or plugins.

See https://docs.gradle.org/7.3.3/userguide/command_line_interface.html#sec:command_line_warnings
6 actionable tasks: 6 up-to-date

FAILURE: Build failed with an exception.

* Where:
Build file '/home/user/Área de Trabalho/docscan/node_modules/react-native-vision-camera/android/build.gradle' line: 231

* What went wrong:
A problem occurred evaluating project ':react-native-vision-camera'.
> Expected directory '/home/user/Área de Trabalho/docscan/node_modules/react-native/android' to contain exactly one file, however, it contains more than one file.

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 11s

error Failed to install the app. Make sure you have the Android development environment set up: https://reactnative.dev/docs/environment-setup.
Error: Command failed: ./gradlew app:installDebug -PreactNativeDevServerPort=8081

FAILURE: Build failed with an exception.

* Where:
Build file '/home/user/Área de Trabalho/docscan/node_modules/react-native-vision-camera/android/build.gradle' line: 231

* What went wrong:
A problem occurred evaluating project ':react-native-vision-camera'.
> Expected directory '/home/user/Área de Trabalho/docscan/node_modules/react-native/android' to contain exactly one file, however, it contains more than one file.
```

### Device

Samsung S20 FE (Android 12)

### VisionCamera Version

2.13.1 and 2.13.5

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

sinfrid, 2022-07-04:
> Hi,
> 
> you could wait for this PR to merge #1116
> 
> or use a patch from the attachment [react-native-vision-camera+2.13.5.txt](https://github.com/mrousavy/react-native-vision-camera/files/9035854/react-native-vision-camera%2B2.13.5.txt)

How to apply the patch?

nobane, 2022-07-06:
Thank you for the patch @psycheangel!

@sinfrid and anyone else who wants to use this patch: you need to put the patch a folder called `patches` in the root of your project, install [patch-package](https://www.npmjs.com/package/patch-package), and then run `npx patch-package` to apply the patch. 

For instance, if the txt file were saved to `~/Downloads`:

    cd <project-directory>
    mkdir patches
    mv ~/Downloads/react-native-vision-camera+2.13.5.txt patches/react-native-vision-camera+2.13.5.patch
    npm i patch-package
    npx patch-package

closed 2022-07-06 by viniciusyoshioka

carlosdan-el, 2022-07-26:
> > > Hi,
> > > you could wait for this PR to merge #1116
> > > or use a patch from the attachment [react-native-vision-camera+2.13.5.txt](https://github.com/mrousavy/react-native-vision-camera/files/9035854/react-native-vision-camera%2B2.13.5.txt)
> > 
> > 
> > How to apply the patch?
> 
> rename the file to react-native-vision-camera+2.13.5.patch use patch-package library

BTW, I'ts works in the last version of react native/react-native-vision-camera. Thank you!

JulianErnest, 2022-07-28:
Still have the issue. Project: 
React Native - 0.69.1
React Native Vision Camera - 2.14.0
React Native Reanimated - 2.9.1

mocanSergiu666, 2022-07-28:
React Native - 0.69.3
React Native Vision Camera - 2.14.0
React Native Reanimated - 2.9.1
__________________________________
Same issue, could someone take a look, please?

jimshidbachu, 2022-07-29:
> Thank you for the patch @psycheangel!
>
> @sinfrid and anyone else who wants to use this patch: you need to put the patch a folder called `patches` in the root of your project, install [...]

the patch used, but then having another issues

> Task :react-native-vision-camera:compileDebugKotlin
w: /.../node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt: (7, 44): 'RCTEventEmitter' is deprecated. Deprecated in Java
w: /.../node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt: (13, 28): 'RCTEventEmitter' is deprecated. Deprecated in Java
w: /.../node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt: (13, 57): 'receiveEvent(Int, String!, WritableMap?): Unit' is deprecated. Deprecated in Java
w: /.../node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt: (31, 28): 'RCTEventEmitter' is deprecated. Deprecated in Java
w: /.../node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt: (31, 57): 'receiveEvent(Int, String!, WritableMap?): Unit' is deprecated. Deprecated in Java
w: /.../node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt: (42, 28): 'RCTEventEmitter' is deprecated. Deprecated in Java
w: /.../node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt: (42, 57): 'receiveEvent(Int, String!, WritableMap?): Unit' is deprecated. Deprecated in Java
w: /.../node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt: (48, 28): 'RCTEventEmitter' is deprecated. Deprecated in Java
w: /.../node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt: (48, 57): 'receiveEvent(Int, String!, WritableMap?): Unit' is deprecated. Deprecated in Java
w: /.../node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView.kt: (27, 44): 'RCTEventEmitter' is deprecated. Deprecated in Java
w: /.../node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraViewManager.kt: (11, 25): Parameter 'reactContext' is never used
w: /.../node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraViewModule.kt: (67, 11): 'onCatalystInstanceDestroy(): Unit' is deprecated. Overrides deprecated member in 'com.facebook.react.bridge.NativeModule'. Deprecated in Java

> Task :react-native-vision-camera:configureCMakeDebug[arm64-v8a]
C/C++: debug|arm64-v8a :CMake Warning (dev) in CMakeLists.txt:
C/C++: debug|arm64-v8a :  No project() command is present.  The top-level CMakeLists.txt file must
C/C++: debug|arm64-v8a :  contain a literal, direct call to the project() command.  Add a line of
C/C++: debug|arm64-v8a :  code such as
C/C++: debug|arm64-v8a :    project(ProjectName)
C/C++: debug|arm64-v8a :  near the top of the file, but after cmake_minimum_required().
C/C++: debug|arm64-v8a :  CMake is pretending there is a "project(Project)" command on the first
C/C++: debug|arm64-v8a :  line.
C/C++: debug|arm64-v8a :This warning is for project developers.  Use -Wno-dev to suppress it.
C/C++: debug|arm64-v8a :CMake Warning at CMakeLists.txt:148 (message):
C/C++: debug|arm64-v8a :  VisionCamera linking: FOR_HERMES=false

> Task :react-native-vision-camera:configureCMakeDebug[armeabi-v7a]
C/C++: debug|armeabi-v7a :CMake Warning (dev) in CMakeLists.txt:
C/C++: debug|armeabi-v7a :  No project() command is present.  The top-level CMakeLists.txt file must
C/C++: debug|armeabi-v7a :  contain a literal, direct call to the project() command.  Add a line of
C/C++: debug|armeabi-v7a :  code such as
C/C++: debug|armeabi-v7a :    project(ProjectName)
C/C++: debug|armeabi-v7a :  near the top of the file, but after cmake_minimum_required().
C/C++: debug|armeabi-v7a :  CMake is pretending there is a "project(Project)" command on the first
C/C++: debug|armeabi-v7a :  line.
C/C++: debug|armeabi-v7a :This warning is for project developers.  Use -Wno-dev to suppress it.
C/C++: debug|armeabi-v7a :CMake Warning at CMakeLists.txt:148 (message):
C/C++: debug|armeabi-v7a :  VisionCamera linking: FOR_HERMES=false

> Task :react-native-vision-camera:configureCMakeDebug[x86]
C/C++: debug|x86 :CMake Warning (dev) in CMakeLists.txt:
C/C++: debug|x86 :  No project() command is present.  The top-level CMakeLists.txt file must
C/C++: debug|x86 :  contain a literal, direct call to the project() command.  Add a line of
C/C++: debug|x86 :  code such as
C/C++: debug|x86 :    project(ProjectName)
C/C++: debug|x86 :  near the top of the file, but after cmake_minimum_required().
C/C++: debug|x86 :  CMake is pretending there is a "project(Project)" command on the first
C/C++: debug|x86 :  line.
C/C++: debug|x86 :This warning is for project developers.  Use -Wno-dev to suppress it.
C/C++: debug|x86 :CMake Warning at CMakeLists.txt:148 (message):
C/C++: debug|x86 :  VisionCamera linking: FOR_HERMES=false

> Task :react-native-vision-camera:configureCMakeDebug[x86_64]
C/C++: debug|x86_64 :CMake Warning (dev) in CMakeLists.txt:
C/C++: debug|x86_64 :  No project() command is present.  The top-level CMakeLists.txt file must
C/C++: debug|x86_64 :  contain a literal, direct call to the project() command.  Add a line of
C/C++: debug|x86_64 :  code such as
[... cut ...]
    at makeError (/.../node_modules/@react-native-community/cli-platform-android/node_modules/execa/index.js:174:9)
    at /.../node_modules/@react-native-community/cli-platform-android/node_modules/execa/index.js:278:16
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at async runOnAllDevices (/.../node_modules/@react-native-community/cli-platform-android/build/commands/runAndroid/runOnAllDevices.js:109:5)
    at async Command.handleAction (/.../node_modules/@react-native-community/cli/build/index.js:192:9)

here what shows after patch and run

viniciusyoshioka, 2022-07-29:
@jimshidbachu I got a solution to this new error and the .patch to fix it is here https://github.com/mrousavy/react-native-vision-camera/issues/1125#issuecomment-1183987100

fynally, 2022-07-31:
I followed the patch process as stated above but when I tried to clean my project, 
it got stucked this >> Preparing "Install NDK (Side by side) 21.4.7075529 (revision: 21.4.7075529)".<<<
Please what will I do?
.......................................................................

"patch-package": "^6.4.7",
"react-native": "0.69.1",
"react-native-reanimated": "^2.9.1",
"vision-camera-dynamsoft-barcode-reader": "^0.4.3"
"react-native-vision-camera": "^2.14.0",
..........................................................................................................................

PS C:\Users\---\Desktop\---\---\android> ./gradlew clean
Starting a Gradle Daemon (subsequent builds will be faster)

> Configure project :react-native-reanimated
AAR for react-native-reanimated has been found
C:\Users\---\Desktop\---\---\node_modules\react-native-reanimated\android\react-native-reanimated-69-jsc.aar

> Configure project :react-native-vision-camera
VisionCamera: node_modules/ found at: C:\Users\---\Desktop\---\---\node_modules
VisionCamera: Building with JSC...
Warning: The 'kotlin-android-extensions' Gradle plugin is deprecated. Please use this migration guide (https://goo.gle/kotlin-android-extensions-deprecation) to start working with View Binding (https://developer.android.com/topic/libraries/view-binding) and the 'kotlin-parcelize' plugin.
VisionCamera: Frame Processors are enabled! Building C++ part...
WARNING:DSL element 'dexOptions' is obsolete and should be removed.
It will be removed in version 8.0 of the Android Gradle plugin.
Using it has no effect, and the AndroidGradle plugin optimizes dexing automatically.
[CXX1101] NDK at C:\Users\---\AppData\Local\Android\Sdk\ndk\21.4.7075529 did
not have a source.properties file
Checking the license for package NDK (Side by side) 21.4.7075529 in C:\Users\----\AppData\Local\Android\Sdk\licenses
License for package NDK (Side by side) 21.4.7075529 accepted.
Preparing "Install NDK (Side by side) 21.4.7075529 (revision: 21.4.7075529)".
<===========--> 87% CONFIGURING [13m 59s]
> :react-native-vision-camera

lfrallon, 2022-08-03:
> ### **Preparing "Install NDK (Side by side) 21.4.7075529 (revision: 21.4.7075529)"**
> 
> I followed the patch process as stated above but when I tried to clean my project, it got stucked here >> _Preparing "Install NDK (Side by side) 21.4.7075529 (revision: 21.4.7075529)"_
> ### Please what will I do?
> 
> .......................................................................
> 
> "patch-package": "^6.4.7", "react-native": "0.69.1", "react-native-reanimated": "^2.9.1", "vision-camera-dynamsoft-barcode-reader": "^0.4.3" "react-native-vision-camera": "^2.14.0", ..........................................................................................................................
> 
> ```
> PS C:\Users\---\Desktop\---\---\android> ./gradlew clean
> Starting a Gradle Daemon (subsequent builds will be faster)
> 
> > Configure project :react-native-reanimated
> AAR for react-native-reanimated has been found
> C:\Users\---\Desktop\---\---\node_modules\react-native-reanimated\android\react-native-reanimated-69-jsc.aar
> 
> > Configure project :react-native-vision-camera
> VisionCamera: node_modules/ found at: C:\Users\---\Desktop\---\---\node_modules
> VisionCamera: Building with JSC...
> Warning: The 'kotlin-android-extensions' Gradle plugin is deprecated. Please use this migration guide (https://goo.gle/kotlin-android-extensions-deprecation) to start working with View Binding (https://developer.android.com/topic/libraries/view-binding) and the 'kotlin-parcelize' plugin.
> VisionCamera: Frame Processors are enabled! Building C++ part...
> WARNING:DSL element 'dexOptions' is obsolete and should be removed.
> It will be removed in version 8.0 of the Android Gradle plugin.
> Using it has no effect, and the AndroidGradle plugin optimizes dexing automatically.
> [CXX1101] NDK at C:\Users\---\AppData\Local\Android\Sdk\ndk\21.4.7075529 did
> not have a source.properties file
> Checking the license for package NDK (Side by side) 21.4.7075529 in C:\Users\----\AppData\Local\Android\Sdk\licenses
> License for package NDK (Side by side) 21.4.7075529 accepted.
> Preparing "Install NDK (Side by side) 21.4.7075529 (revision: 21.4.7075529)".
> <===========--> 87% CONFIGURING [59m 59s]
> > :react-native-vision-camera
> ```

Same in "react-native": "0.69.3".

danieltvaz, 2022-08-04:
> 

After applying the patch, can I delete the "patches" folder securely ?

TaheriSDE, 2022-08-09:
> Thank you for the patch @psycheangel!
>
> @sinfrid and anyone else who wants to use this patch: you need to put the patch a folder called `patches` in the root of your project, install [...]

Thank you it works for me

princefishthrower, 2022-10-19:
With react native 0.70.3 and react-native-vision-camera 2.13.5 building after applying the patch did NOT working for me.

thomasdierickx, 2023-07-12:
* Where:
Build file '/Users/.../Desktop/.../project/node_modules/react-native-mmkv-storage/android/build.gradle' line: 138

* What went wrong:
A problem occurred evaluating project ':react-native-mmkv-storage'.
> Expected directory '/Users/.../Desktop/.../project/node_modules/react-native-mmkv-storage/android/../../../node_modules/react-native/android' to contain exactly one file, however, it contains no files.

Is their a possible fix for this error? Or can i get some guidance at least please?