margelo/react-native-vision-camera #1119: How to custom photo size to 4:3, 16:9 or freesize❓

truongngodang, 2022-07-04:
### Question

I want to custom size image when take photo.

### VisionCamera Version

2.13.5

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

julianklumpers, 2022-07-04:
I achieved this with the following.

For the camera view to get the wanted aspect ratio, for example 4:3 or 16:9. You need to get the screen width first. You can achieve this with `useWindowDimensions` from `react-native`

example:

```js
const { width } = useWindowDimensions()

...

return (
  <Camera
    ref={cameraRef}
    style={{
    ...StyleSheet.absoluteFillObject,
    width,
    height: width * (16/9), // or width * (4/3)
    }}
    device={device}
    isActive
    photo
    orientation="portrait"
    preset="photo"
  />
)
```

For your photo resolution, you can use something with `expo-image-manipulator` to resize your photo. For example, if your photo is 3024x4032 in 4:3 ratio, and you want the resolution for 16:9 resolution. You can do something like this: 4032 / (16/9) = 2268. Then use image manipulator to resize your image.

mrousavy, 2022-07-04:
Exactly what @julianklumpers said ☝️ 

Also, read [the docs on formats](https://mrousavy.com/react-native-vision-camera/docs/guides/formats)

closed 2022-07-04 by mrousavy

Pavaneshpratap, 2023-06-01:
> I achieved this with the following.
> 
> For the camera view to get the wanted aspect ratio, for example 4:3 or 16:9. You need to get the screen width first. You can achieve this with `useWindowDimensions` from `react-native`
> 
> example:
> 
> ```js
> const { width } = useWindowDimensions()
> 
> ...
> 
> return (
>   <Camera
>     ref={cameraRef}
>     style={{
>     ...StyleSheet.absoluteFillObject,
>     width,
>     height: width * (16/9), // or width * (4/3)
>     }}
>     device={device}
>     isActive
>     photo
>     orientation="portrait"
>     preset="photo"
>   />
> )
> ```
> 
> For your photo resolution, you can use something with `expo-image-manipulator` to resize your photo. For example, if your photo is 3024x4032 in 4:3 ratio, and you want the resolution for 16:9 resolution. You can do something like this: 4032 / (16/9) = 2268. Then use image manipulator to resize your image.
> 
> Hope this helps

with this method we only cropped that image or particular camera view
but ideal behaviour will be camera should be the cover more area in 4:3 in comparison to 16:9

is there any method so that we can set the camera with 4:3 aspect ratio like the native camera behaviour ?

mrousavy, 2023-06-01:
Yes, by selecting a Format. Read [the docs on formats](https://mrousavy.com/react-native-vision-camera/docs/guides/formats).

Arpine-javakhkuhi, 2023-11-23:
How to specify a Format for Vision Camera version 2.15.4?