margelo/react-native-vision-camera #1121: 🐛 'react/bridging/CallbackWrapper.h' file not found

luisfuertes, 2022-07-06:
### What were you trying to do?

Im trying to install the library, i think i follow all steps and all iOS troubleshooting section. I have tried with react-native-reanimated and without it. But always have the same issue

[image]

### Reproduceable Code

```tsx
"react": "18.0.0",
    "react-native": "0.69.1",
    "react-native-reanimated": "^2.9.1",
    "react-native-vision-camera": "^2.13.5"
```

### What happened instead?

iOS build error

### Relevant log output

```shell
/myApp/node_modules/react-native/ReactCommon/react/nativemodule/core/ReactCommon/TurboModuleUtils.h:16:10: 'react/bridging/CallbackWrapper.h' file not found
```

### Device

iPhone 11 simulator and iPhone Xr device

### VisionCamera Version

"react-native-vision-camera": "^2.13.5"

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

luisfuertes, 2022-07-06:
And on Android 
```
A problem occurred evaluating project ':react-native-vision-camera'.
> Expected directory '/myApp/node_modules/react-native/android' to contain exactly one file, however, it contains more than one file.

```

kierancrown, 2022-07-06:
I am also experiencing both these issues since upgrading to RN 0.69

luisfuertes, 2022-07-07:
@kierancrown do you know some hotfix for that?

kierancrown, 2022-07-07:
> @kierancrown do you know some hotfix for that?
> 
> EDIT: I init my project with react native 0.68 version to avoid that issue xd

I actually managed to get this working by applying two patches.

[RN-Vision-Patches.zip](https://github.com/mrousavy/react-native-vision-camera/files/9061852/RN-Vision-Patches.zip)

I can't get the QR Code frame processor working however. [See this issue here](https://github.com/rodgomesc/vision-camera-code-scanner/issues/75).

luisfuertes, 2022-07-07:
I apply only the pach for react-native-vision-camera and it didnt work :(

kierancrown, 2022-07-07:
> Thanks!
> 
> I'm going to implement my own frame processor so no problem. I recommend you to implement your own frame processor to read QRs, its very easy

Do you have any links or articles? You might need to apply both. I'll check my code and get back to you!

luisfuertes, 2022-07-07:
With example project and this docs is very easy
https://github.com/mrousavy/react-native-vision-camera/tree/main/example

https://mrousavy.com/react-native-vision-camera/docs/guides/frame-processors-plugins-ios
https://mrousavy.com/react-native-vision-camera/docs/guides/frame-processors-plugins-android

luisfuertes, 2022-07-07:
I start new project on RN 0.69.1 and add your patches, but i get the same error :s

I add some info
```
    node v16.15.1
    cocoapods v1.11.3

    "react": "18.0.0",
    "react-native": "0.69.1",
    "react-native-vision-camera": "^2.13.5"
```

kierancrown, 2022-07-07:
> I start new project on RN 0.69.1 and add your patches, but i get the same error :s
> 
> I add some info
> 
> ```
>     node v16.15.1
>     cocoapods v1.11.3
> 
>     "react": "18.0.0",
>     "react-native": "0.69.1",
>     "react-native-vision-camera": "^2.13.5"
> ```

Try this https://github.com/facebook/react-native/issues/34102#issuecomment-1176370957

kierancrown, 2022-07-07:
> I start new project on RN 0.69.1 and add your patches, but i get the same error :s
> 
> I add some info
> 
> ```
>     node v16.15.1
>     cocoapods v1.11.3
> 
>     "react": "18.0.0",
>     "react-native": "0.69.1",
>     "react-native-vision-camera": "^2.13.5"
> ```

No worries, I haven't tested with Android yet, still looking into a frame processor.

luisfuertes, 2022-07-07:
```
#import "CVWrapper.h"
#import <Foundation/Foundation.h>
#import <VisionCamera/FrameProcessorPlugin.h>
#import <VisionCamera/Frame.h>

@interface FrameProcessorPlugin : NSObject
@end

@implementation FrameProcessorPlugin

static inline id frameProcessorPlugin(Frame* frame, NSArray* arguments) {
  CVPixelBufferRef frameBuffer = CMSampleBufferGetImageBuffer(frame.buffer);
  CIImage *ciImage = [CIImage imageWithCVPixelBuffer:frameBuffer];

  NSDictionary* options;
  CIContext* context = [CIContext context];
  options = @{ CIDetectorAccuracy : CIDetectorAccuracyHigh }; // Slow but thorough
  //options = @{ CIDetectorAccuracy : CIDetectorAccuracyLow}; // Fast but superficial

  CIDetector* qrDetector = [CIDetector detectorOfType:CIDetectorTypeQRCode
                                              context:context
                                              options:options];
  if ([[ciImage properties] valueForKey:(NSString*) kCGImagePropertyOrientation] == nil) {
      options = @{ CIDetectorImageOrientation : @1};
  } else {
      options = @{ CIDetectorImageOrientation : [[ciImage properties] valueForKey:(NSString*) kCGImagePropertyOrientation]};
  }

  NSArray * features = [qrDetector featuresInImage:ciImage
                                           options:options];

  if (features != nil && features.count > 0) {
      for (CIQRCodeFeature* qrFeature in features) {
          NSLog(@"QRFeature.messageString : %@ ", qrFeature.messageString);
          return @{
                    @"qr": qrFeature.messageString,
          };
      }
  }
  

}

VISION_EXPORT_FRAME_PROCESSOR(qrPlugin)

@end

```

Try your own frame processor like this on iOS

closed 2022-07-08 by luisfuertes

hengkx, 2022-07-21:
0.69.2 same error

mocanSergiu666, 2022-07-25:
"react": "18.0.0",
"react-native": "0.69.2",
"react-native-vision-camera": "^2.14.0"
It's the same error :. (

mocanSergiu666, 2022-07-26:
Is anyone working on this issue? :)

KZCOWIOT, 2022-09-28:
i am using react-native 0.69.5 and still facing the same issue
'react/bridging/CallbackWrapper.h' file not found

anveshbabu, 2022-12-11:
Hi any one Help this 
i am using react-native 0.70.6 and still facing the same issue

'react/bridging/CallbackWrapper.h' file not found
[image]

luchozamora, 2023-02-06:
@anveshbabu make sure you are using `SWIFT_VERSION = 5.2` in your project.
From react-native-camera's [troubleshooting website](https://mrousavy.com/react-native-vision-camera/docs/guides/troubleshooting):
```
3. iii. If the SWIFT_VERSION value is set, make sure it is set to 5.2 or higher.
```

SamiChab, 2023-08-06:
Any update on this ? I am also facing this same issue
react-native 0.72.3
react-native-vision-camera 2.15.5

asifiqubal, 2023-08-13:
> Any update on this ? I am also facing this same issue react-native 0.72.3 react-native-vision-camera 2.15.5

Same with me.

jonjamz, 2023-08-13:
I started having this issue after upgrading from Expo SDK 48 to 49. From what I've read, the issue is with RN 0.72. I did not have any problems while using RN 0.71. Had to revert for now.

vitas61, 2023-08-16:
fix for rn 0.72.4,  react-native-vision-camera 2.15.5 and react-native-reanimated 2.17.0

with use_frameworks!

update your Podfile

`
post_install do |installer|
       installer.pods_project.targets.each do |target|
..

target.build_configurations.each do |config|
                 config.build_settings['HEADER_SEARCH_PATHS'] ||= '$(inherited) '
                 config.build_settings['HEADER_SEARCH_PATHS'] << '"${PODS_ROOT}/../../node_modules/react-native/ReactCommon" '
                 if target.name === "RNReanimated"
                    config.build_settings['CLANG_CXX_LANGUAGE_STANDARD'] = 'c++17'
                 end
end

..

`

vitas61, 2023-08-16:
> How can I do this using Expo?

@jonjamz 
[https://github.com/mrousavy/react-native-vision-camera/issues/1413#issuecomment-1612635330](https://github.com/mrousavy/react-native-vision-camera/issues/1413#issuecomment-1612635330)

subirats345, 2023-08-21:
Same for me, on react-native 0.70.6 and react-native-vision-camera 2.15.6

rosendosalazarbta, 2023-08-22:
Any update on this ? I'm also facing this same issue.
react-native: 0.72.4
react-native-vision-camera: 2.15.6

sravis02, 2023-09-14:
same for me with 
    "react-native": "0.72.4",
    "react-native-vision-camera": "^3.0.0",

brascene, 2023-09-20:
Same here:
"react-native": "0.72.4",
"react-native-vision-camera": "^2.16.1"

jeemercado, 2023-09-23:
> fix for rn 0.72.4, react-native-vision-camera 2.15.5 and react-native-reanimated 2.17.5
> 
> with use_frameworks!
> 
> update your Podfile
> 
> ` post_install do |installer| installer.pods_project.targets.each do |target| ..
> 
> target.build_configurations.each do |config| config.build_settings['HEADER_SEARCH_PATHS'] ||= '$(inherited) ' config.build_settings['HEADER_SEARCH_PATHS'] << '"${PODS_ROOT}/../../node_modules/react-native/ReactCommon" ' if target.name === "RNReanimated" config.build_settings['CLANG_CXX_LANGUAGE_STANDARD'] = 'c++17' end end
> 
> ..
> 
> `

this worked for me thanks!

junaid7898, 2023-11-18:
> fix for rn 0.72.4, react-native-vision-camera 2.15.5 and react-native-reanimated 2.17.5
>
> with use_frameworks!
>
> update your Podfile
>
> ` post_install do |installer| installer.pods_project.targets.each [...]

when i do pos-install after updating my pod file with this code, it is taking a very long time and I am keep getting this 
"Setting REACT_NATIVE build settings
Setting CLANG_CXX_LANGUAGE_STANDARD to c++17 on /Users/my-project/ios/my-project.xcodeproj"
and it keeps happening again and again, it seems it's a never ending for loop. Any solution for this
Here is my pod file code

  post_install do |installer|
    installer.pods_project.targets.each do |target|
    # https://github.com/facebook/react-native/blob/main/packages/react-native/scripts/react_native_pods.rb#L197-L202
      react_native_post_install(
        installer,
        config[:reactNativePath],
        :mac_catalyst_enabled => false
      )
      __apply_Xcode_12_5_M1_post_install_workaround(installer)
      target.build_configurations.each do |config|
        config.build_settings['HEADER_SEARCH_PATHS'] ||= '$(inherited) '
        config.build_settings['HEADER_SEARCH_PATHS'] << '"${PODS_ROOT}/../../node_modules/react-native/ReactCommon" '
        if target.name === "RNReanimated"
          config.build_settings['CLANG_CXX_LANGUAGE_STANDARD'] = 'c++17'
        end
      end
    end  
  end

sekhuat, 2024-01-14:
I also faced the same issue and I am using react-native-vision-camera ver 2.16.1. I solved the issue by adding `s.dependency "ReactCommon"` into the `node_modules/react-native-vision-camera/VisionCamera.podspec` at the last line as shown below.

```
Pod::Spec.new do |s|
  ...
  s.dependency "React-callinvoker"
  s.dependency "React"
  s.dependency "React-Core"
  s.dependency "ReactCommon"    <== Add this line
end
```

After made the changes, I execute `pod install` command then recompile the app again. Hope this will help those who using V2 of react-native-vision-camera.

AndersonD4vid, 2024-01-29:
> s.dependency "ReactCommon"

Thank youuuuuuuu, this really helped me!