margelo/react-native-vision-camera #1124: How to fix android error, when creating custom plugin❓

Lucianofc138, 2022-07-06:
### Question

I been working on an example app for android using this library, which will need some  native code for a frame processor plugin. I managed to make everything work with a community plugin (https://github.com/rodgomesc/vision-camera-face-detector), but now when I'm trying to create my own plugin, following the tutorial in the docs for kotlin I get this error:

-------
`
> Task :app:compileDebugJavaWithJavac FAILED

Deprecated Gradle features were used in this build, making it incompatible with Gradle 8.0.

You can use '--warning-mode all' to show the individual deprecation warnings and determine if they come from your own scripts or plugins.

See https://docs.gradle.org/7.3.3/userguide/command_line_interface.html#sec:command_line_warnings
86 actionable tasks: 80 executed, 6 up-to-date
Warning: This version only understands SDK XML versions up to 2 but an SDK XML file of version 3 was encountered. This can happen if you use versions of Android Studio and the command-line tools that were released at different times.
Warning: unexpected element (uri:"", local:"base-extension"). Expected elements are <{}codename>,<{}layoutlib>,<{}api-level>
/<BASE_DIR>/RNCamera0/android/app/src/main/java/com/rncamera0/MainApplication.java:31: error: cannot find symbol
          packages.add(new QRCodeFrameProcessorPluginPackage());
                           ^
  symbol: class QRCodeFrameProcessorPluginPackage
Note: /<BASE_DIR>/RNCamera0/android/app/src/debug/java/com/rncamera0/ReactNativeFlipper.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
1 error

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:compileDebugJavaWithJavac'.
> Compilation failed; see the compiler error output for details.

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 54s

error Failed to install the app. Make sure you have the Android development environment set up: https://reactnative.dev/docs/environment-setup.
Error: Command failed: ./gradlew app:installDebug -PreactNativeDevServerPort=8081
Warning: This version only understands SDK XML versions up to 2 but an SDK XML file of version 3 was encountered. This can happen if you use versions of Android Studio and the command-line tools that were released at different times.
Warning: unexpected element (uri:"", local:"base-extension"). Expected elements are <{}codename>,<{}layoutlib>,<{}api-level>
/<BASE_DIR>/RNCamera0/android/app/src/main/java/com/rncamera0/MainApplication.java:31: error: cannot find symbol
          packages.add(new QRCodeFrameProcessorPluginPackage());
                           ^
  symbol: class QRCodeFrameProcessorPluginPackage
Note: /<BASE_DIR>/RNCamera0/android/app/src/debug/java/com/rncamera0/ReactNativeFlipper.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
1 error

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:compileDebugJavaWithJavac'.
> Compilation failed; see the compiler error output for details.

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 54s

    at makeError (/<BASE_DIR>/RNCamera0/node_modules/execa/index.js:174:9)
    at /<BASE_DIR>/RNCamera0/node_modules/execa/index.js:278:16
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at async runOnAllDevices (/<BASE_DIR>/RNCamera0/node_modules/@react-native-community/cli-platform-android/build/commands/runAndroid/runOnAllDevices.js:109:5)
    at async Command.handleAction (/<BASE_DIR>/RNCamera0/node_modules/@react-native-community/cli/build/index.js:192:9)
info Run CLI with --verbose flag for more details.
error Command failed with exit code 1.

info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.

`
(THIS IS THE END OF THE ERROR)
-------
-------

The only thing I did following the basic example I'm not sure about is, I wrote the files for the plugin and the package in the same folder as MainActivity.java, is that okay? Also I didn't have the option to directly create a java or kotlin file, I had to click new scratch file to create both of them. (See image) I don't know if that could be related to the fact that the directory isn't mark in any specific way such as module or resources root.

Disclaimer: it's my first time interacting with android native code, so it could be something really basic, I apologize in advance if that's the case.

[image]
.

Thanks for reading and for any help any of you could give me!

### What I tried

I tried following the tutorial a few times, first customizing some stuff, but then following every part including the names of the plugins and packages.

I tried I few things searching for this part of the error:
`error: cannot find symbol
          packages.add(new QRCodeFrameProcessorPluginPackage());
`
But nothing worked.

### VisionCamera Version

2.13.5

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

skam22, 2022-07-15:
the documentation and tutorials are assuming that you're installing a packaged plugin from outside your project.

when you write your own local plugin, first need to declare it in `android/settings.gradle` and then add as a dependency to `android/app/build.gradle`.

first start by creating a directory located at `/android/app/src/main/java/com/rncamera0/qrcodeframeprocessor`

inside that directory add your custom frame processor files:

`/android/app/src/main/java/com/rncamera0/qrcodeframeprocessor/QRCodeFrameProcessorPlugin.kt`
`/android/app/src/main/java/com/rncamera0/qrcodeframeprocessor/QRCodeFrameProcessorPluginPackage.kt`

the top line of both of those files should read:

`package com.rncamera0.qrcodeframeprocessor`

in your `/android/app/main/java/com/rncamera0/MainApplication.java` file, import the package:

`import com.rncamera0.qrcodeframeprocessor.QRCodeFrameProcessorPluginPackage`

add it inside the getPackages() block:

```
@Override
        protected List<ReactPackage> getPackages() {
          @SuppressWarnings("UnnecessaryLocalVariable")
          List<ReactPackage> packages = new PackageList(this).getPackages();
          // Packages that cannot be autolinked yet can be added manually here, for example:
          // packages.add(new MyReactNativePackage());
          packages.add(new QRCodeFrameProcessorPluginPackage());  // <-- ADD THIS LINE
          return packages;
        }
```

in your `/android/settings.gradle` add these two lines after `include ':app'`:

```
...
include ':app'

include ":qrcodeframeprocessor" // <-- ADD THIS LINE
project(":qrcodeframeprocessor").projectDir = new File(rootProject.projectDir, '/app/src/main/java/com/rncamera0/qrcodeframeprocessor') // <-- ADD THIS LINE
```

in your `/android/app/build.gradle` add this line to dependencies block:

```
dependencies {
   ... // other dependencies
  implementation project(':qrcodeframeprocessor') // <-- ADD THIS LINE
}
```

Lucianofc138, 2022-07-16:
> the documentation and tutorials are assuming that you're installing a packaged plugin from outside your project.

That makes sense, actually that was the alternative I was trying while I couldn't solve this, since I knew it worked with the community plugins.

I did what you wrote above and for the first time in more than a week I got a different error while building:

```
error Failed to install the app. Make sure you have the Android development environment set up: https://reactnative.dev/docs/environment-setup.
Error: Command failed: ./gradlew app:installDebug -PreactNativeDevServerPort=8081

FAILURE: Build failed with an exception.

* What went wrong:
Could not determine the dependencies of task ':app:mergeDebugAssets'.
> Could not resolve all task dependencies for configuration ':app:debugRuntimeClasspath'.
   > Could not resolve project :qrcodeframeprocessor.
     Required by:
         project :app
      > No matching configuration of project :qrcodeframeprocessor was found. The consumer was configured 
      to find a runtime of a component, preferably optimized for Android, as well as attribute
      'com.android.build.api.attributes.BuildTypeAttr' with value 'debug' but:
          - None of the consumable configurations have attributes.

```
I found a few similar errors online that were solved by adding `missingDimensionStrategy 'react-native-camera', 'general'` or `missingDimensionStrategy 'store', 'play'` in the defaultConfig of `android/app/build.gradle`, so I tried that, and also `missingDimensionStrategy 'qrcodeframeprocessor', 'general'`, but none of that worked. So I will be trying to fix this new error.

@skam22 Thanks you so much for the help! And if you have any further recommendation I will be happy to read them.

mrousavy, 2022-07-18:
Did you try to use the template for a new RN library? That's how the QR Code plugin was made

skam22, 2022-07-19:
since you're using kotlin files (.kt) don't forget to have this line towards the top of your `android/app/build.gradle`

```
apply plugin: "kotlin-android"
```

that could be the missing link for you since you're using a .kt frame processor.  Unless you specifically tell the compiler to include kotlin files it's not going to be able to find the class declaration.

Lucianofc138, 2022-07-20:
@mrousavy don't worry! Regarding to this issue, no, I didn't use any template, since I was trying implement the plugin for the frame processor inside the same project code. 

However, I was working on creating an external package that could fulfill my needs (in parallel/another branch), which I did with bob, but it didn't work for me (my IDE didn't even recognize the example function). Is that (bob) what you mean when talking about the template? If that's not the case, were could I find the template?
Anyway for the moment I'm testing some basic functionalities with some code I forked from a community library that already included kotlin (vision-camer-ocr), so I'm just editing its code, but I know that's not the best long term solution.

@skam22 I actually tried that before, but for some reason now the build process worked, probably because I changed some other configs in build.gradle files thanks to you and other person in reddit. Nevertheless, now the Camera preview is black (before it was wroking fine), so I will be trying to check which change make it stop working, and I hope that wasn't necessary to make it build.

Thanks again for all the help!

aboozaid, 2022-10-11:
@mrousavy building a plugin is such a painful process, I think this is important to invest some time at to make it a bit easier

mrousavy, 2022-10-12:
> I think this is important to invest some time at to make it a bit easier

Sure, I think it is important to invest some money so I can work on this! :)

aboozaid, 2023-03-26:
> May I know are you manage to solve the problem? I have the exact same
> problem as you when trying to implement the plugin for the frame processor
> inside the same project code.

If I remember correctly I think I did not solve it.

skam22, 2023-03-26:
Yes - the process I laid out above is how we’re using a locally defined frame processor.

michealChin, 2023-03-26:
> * What went wrong:
> Could not determine the dependencies of task ':app:mergeDebugAssets'.
>      Required by:
>          project :app
>       to find a runtime of a component, preferably optimized for [...]

I follow exactly like what you say to update the android/settings.gradle and then add as a dependency to android/app/build.gradle but I encounter below error message.

* What went wrong:
Could not determine the dependencies of task ':app:mergeDebugAssets'.
> Could not resolve all task dependencies for configuration ':app:debugRuntimeClasspath'.
   > Could not resolve project :qrcodeframeprocessor.
     Required by:
         project :app
      > No matching configuration of project :qrcodeframeprocessor was found. The consumer was configured 
      to find a runtime of a component, preferably optimized for Android, as well as attribute
      'com.android.build.api.attributes.BuildTypeAttr' with value 'debug' but:
          - None of the consumable configurations have attributes.

I then trying to follow the sample code from "https://github.com/mrousavy/Colorwaver", which is using a locally defined frame processor but he didn't do what you have suggested but also end up with the same error message. I have been struggling for this for a weeks, searching for the answer in stackoverflow and this forum, some time I think maybe migrating to flutter will be a good choice, however I want to give a one last chance before making such decision.

aboozaid, 2023-03-26:
I don’t think that’s a good decision as every framework has its own pros
and cons. Being struggling for weeks doesn’t mean that rn very poor however
you can try to reach to the owner and ask for guidelines to solve the
problem

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

closed 2023-09-30 by mrousavy

mrousavy, 2023-10-06:
Hey!

JFYI; **VisionCamera V3 now includes a QR/Barcode Scanner!** 😍  Check out the [CodeScanner Documentation](https://www.react-native-vision-camera.com/docs/guides/code-scanning) 🚀 

Try upgrading to V3 if you can, it has a much more stable yet flexible foundation.

If you appreciate me dedicating my free time to improving VisionCamera and implementing features like the Code Scanner, please consider [**sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy) to show your support.