margelo/react-native-vision-camera #1125: 🐛 Error building: "2 files found with path 'lib/arm64-v8a/libfolly_runtime.so' from inputs"

viniciusyoshioka, 2022-07-06:
### What were you trying to do?

I was trying to build the app

### What happened instead?

The build fails

### Relevant log output

```shell
> Task :app:mergeDebugNativeLibs FAILED
Execution optimizations have been disabled for task ':app:mergeDebugNativeLibs' to ensure correctness due to the following reasons:
  - Gradle detected a problem with the following location: '/home/user/Área de Trabalho/docscan/android/app/build/intermediates/merged_jni_libs/debug/out'. Reason: Task ':app:mergeDebugNativeLibs' uses this output of task ':app:copyDebugReactNativeVectorIconFonts' without declaring an explicit or implicit dependency. This can lead to incorrect results being produced, depending on what order the tasks are executed. Please refer to https://docs.gradle.org/7.3.3/userguide/validation_problems.html#implicit_dependency for more details about this problem.

Deprecated Gradle features were used in this build, making it incompatible with Gradle 8.0.

You can use '--warning-mode all' to show the individual deprecation warnings and determine if they come from your own scripts or plugins.

See https://docs.gradle.org/7.3.3/userguide/command_line_interface.html#sec:command_line_warnings

Execution optimizations have been disabled for 8 invalid unit(s) of work during this build to ensure correctness.
Please consult deprecation warnings for more details.
295 actionable tasks: 24 executed, 271 up-to-date

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:mergeDebugNativeLibs'.
> A failure occurred while executing com.android.build.gradle.internal.tasks.MergeNativeLibsTask$MergeNativeLibsTaskWorkAction
   > 2 files found with path 'lib/arm64-v8a/libfolly_runtime.so' from inputs:
      - /home/user/Área de Trabalho/docscan/node_modules/react-native-vision-camera/android/build/intermediates/library_jni/debug/jni/arm64-v8a/libfolly_runtime.so
      - /home/user/.gradle/caches/transforms-3/1eed3d8d6e8f118b544df7150c917187/transformed/jetified-react-native-0.69.1-debug/jni/arm64-v8a/libfolly_runtime.so
     If you are using jniLibs and CMake IMPORTED targets, see
     https://developer.android.com/r/tools/jniLibs-vs-imported-targets

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 1m

error Failed to install the app. Make sure you have the Android development environment set up: https://reactnative.dev/docs/environment-setup.
Error: Command failed: ./gradlew app:installDebug -PreactNativeDevServerPort=8081

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:mergeDebugNativeLibs'.
> A failure occurred while executing com.android.build.gradle.internal.tasks.MergeNativeLibsTask$MergeNativeLibsTaskWorkAction
   > 2 files found with path 'lib/arm64-v8a/libfolly_runtime.so' from inputs:
      - /home/user/Área de Trabalho/docscan/node_modules/react-native-vision-camera/android/build/intermediates/library_jni/debug/jni/arm64-v8a/libfolly_runtime.so
      - /home/user/.gradle/caches/transforms-3/1eed3d8d6e8f118b544df7150c917187/transformed/jetified-react-native-0.69.1-debug/jni/arm64-v8a/libfolly_runtime.so
     If you are using jniLibs and CMake IMPORTED targets, see
     https://developer.android.com/r/tools/jniLibs-vs-imported-targets

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 1m

    at makeError (/home/user/Área de Trabalho/docscan/node_modules/execa/index.js:174:9)
    at /home/user/Área de Trabalho/docscan/node_modules/execa/index.js:278:16
    at processTicksAndRejections (internal/process/task_queues.js:95:5)
    at async runOnAllDevices (/home/user/Área de Trabalho/docscan/node_modules/@react-native-community/cli-platform-android/build/commands/runAndroid/runOnAllDevices.js:109:5)
    at async Command.handleAction (/home/user/Área de Trabalho/docscan/node_modules/react-native/node_modules/@react-native-community/cli/build/index.js:192:9)
info Run CLI with --verbose flag for more details.
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
```

### Device

Samsung S20 FE (Android 12)

### VisionCamera Version

2.13.5

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

rvuyyuru1, 2022-07-12:
found any solution for this

viniciusyoshioka, 2022-07-14:
> found any solution for this ?

This patch worked for me [react-native-vision-camera+2.13.5.txt](https://github.com/mrousavy/react-native-vision-camera/files/9108558/react-native-vision-camera%2B2.13.5.txt)

GittiMcHub, 2022-07-28:
I first faced https://github.com/mrousavy/react-native-vision-camera/issues/1118 
Then applied the patch and faced the error stated here.

I tried with 2.13.5 + patch and 2.14.0 + patch but doesn't work for me. I keep getting same error. Did you do anything else?

GittiMcHub, 2022-07-28:
@hahachiwah and @rvuyyuru1 
I additionally applied this https://github.com/mrousavy/react-native-vision-camera/pull/1144/files
to myProject/node_modules/react-native-vision-camera/android/build.gradle

viniciusyoshioka, 2022-07-29:
> I first faced #1118 Then applied the patch and faced the error stated here.
> 
> I tried with 2.13.5 + patch and 2.14.0 + patch but doesn't work for me. I keep getting same error. Did you do anything else?

I forgot to specify, but this .patch is different than the one in this comment https://github.com/mrousavy/react-native-vision-camera/issues/1118#issuecomment-1173229139, this only contains the changes related to pull requests #1144.

closed 2022-08-10 by mrousavy

babaiyu, 2022-08-17:
Hi, this issue is still happen in my project from latest version "2.14.1".

- react-native => "0.69.4"
- react-native-vision-camera => "2.14.1"

viniciusyoshioka, 2022-08-20:
Hi @babaiyu
This issue was fixed, but wasn't released yet to npm. You can install it with: `npm install https://github.com/mrousavy/react-native-vision-camera`

babaiyu, 2022-08-22:
> Hi @babaiyu This issue was fixed, but wasn't released yet to npm. You can install it with: `npm install https://github.com/mrousavy/react-native-vision-camera`

Thanks for the solution, @viniciusyoshioka 
But after install from github url i got failed build from source. Could you teach me how to install library from github source?

What i do:
```
yarn add https://github.com/mrousavy/react-native-vision-camera
cd node_modules/react-native-vision-camera
yarn install && yarn build
```

And finally i got this error
```
src/Camera.tsx:476:8 - error TS2786: 'NativeCameraView' cannot be used as a JSX component.
  Its instance type 'Component<NativeCameraViewProps, {}, any> & Readonly<NativeMethods>' is not a valid JSX element.
    The types returned by 'render()' are incompatible between these types.
      Type 'React.ReactNode' is not assignable to type 'import("/home/babaiyu/Desktop/personal/simponi-app/node_modules/@types/react/index").ReactNode'.
        Type '{}' is not assignable to type 'ReactNode'.

476       <NativeCameraView
           ~~~~~~~~~~~~~~~~

Found 1 error in src/Camera.tsx:476

✖ Failed to build definition files.
bob build
```

viniciusyoshioka, 2022-08-24:
> Could you teach me how to install library from github source?

Hi, of course!
To install the library using GitHub I ran:
```
npm install https://github.com/mrousavy/react-native-vision-camera
npx react-native run-android
```
Thats what I did and it worked for me. I think is not needed to build from source manually.

kicksent, 2022-09-01:
I have never installed packages via github. Does this work for production builds too? When will this be updated on npm so we can use the traditional installation method? @viniciusyoshioka Thanks!!

marsakt, 2022-09-06:
Hi @viniciusyoshioka,

I would like to ask you about the status, I have the latest versions of packages, but still I have this issue, when I tried to build the application. I am using EXPO and the error is:

[stderr] FAILURE: Build failed with an exception.
[stderr] * What went wrong:
[stderr] Execution failed for task ':app:mergeReleaseNativeLibs'.
[stderr] > A failure occurred while executing com.android.build.gradle.internal.tasks.MergeNativeLibsTask$MergeNativeLibsTaskWorkAction
[stderr]    > 2 files found with path 'lib/arm64-v8a/libfolly_runtime.so' from inputs:
[stderr]       - /home/expo/workingdir/build/node_modules/react-native-vision-camera/android/build/intermediates/library_jni/release/jni/arm64-v8a/libfolly_runtime.so
[stderr]       - /home/expo/.gradle/caches/transforms-3/04cf608f9945247f5366e97511924550/transformed/jetified-react-native-0.69.5-release/jni/arm64-v8a/libfolly_runtime.so
[stderr]      If you are using jniLibs and CMake IMPORTED targets, see
[stderr]      https://developer.android.com/r/tools/jniLibs-vs-imported-targets
[stderr] * Try:
[stderr] > Run with --stacktrace option to get the stack trace.
[stderr] > Run with --info or --debug option to get more log output.
[stderr] > Run with --scan to get full insights.
[stderr] * Get more help at https://help.gradle.org
[stderr] BUILD FAILED in 12m 11s

"react-native": "0.69.5",
"react-native-vision-camera": "^2.14.1",
"expo": "~46.0.9",
"expo-camera": "~12.3.0",
"expo-dev-client": "~1.2.1",
"expo-file-system": "~14.1.0",
"expo-image-manipulator": "~10.4.0",
"expo-image-picker": "~13.3.1",
"expo-location": "~14.3.0",
"expo-media-library": "~14.2.0",
"expo-modules-core": "~0.11.5",
"expo-sensors": "~11.4.0",
"expo-splash-screen": "~0.16.2",
"expo-status-bar": "~1.4.0",
"expo-structured-headers": "^2.2.1",
"eas-cli": "2.1.0",
"expo-cli": "6.0.5",

I got the same error in EXPO Cloud & also locally when I try to build application.

Any suggestion how to fix this error? 

Thank you very much !!

viniciusyoshioka, 2022-09-06:
> Hi @viniciusyoshioka,
>
> I would like to ask you about the status, I have the latest versions of packages, but still I have this issue, when I tried to build the application. I am using EXPO and the [...]

Hi!
react-native-vision-camera version 2.14.1 doesn't contain the fix for this issue. It wasn't released yet. To use it you can install the library from its repository with `npm install https://github.com/mrousavy/react-native-vision-camera`

And i think the configuration in app.json can be removed after that

marsakt, 2022-09-06:
sorry for stupid qeustion, I see here https://github.com/mrousavy/react-native-vision-camera/pull/1144 that it was merge. Can I find somewhere when and possibly in which version it will be released ??

Thank you

Productivix, 2022-09-13:
hi, if that can help : 
I have this failure, but _only on Linux_ , **not on Windows 10.**
I have this message exactly: 
`A failure occurred while executing com.android.build.gradle.internal.tasks.MergeNativeLibsTask$MergeNativeLibsTaskWorkAction
   > 2 files found with path 'lib/arm64-v8a/libfolly_runtime.so' from inputs:
      - /myhome/dev_RN-Node-sauv-GIT/test_rn_qrscan/node_modules/react-native-vision-camera/android/build/intermediates/library_jni/debug/jni/arm64-v8a/libfolly_runtime.so
      - /myhome/.gradle/caches/transforms-3/df0743fddd24fb4ce9da9c85e374cb8b/transformed/jetified-react-native-0.70.0-debug/jni/arm64-v8a/libfolly_runtime.so
     If you are using jniLibs and CMake IMPORTED targets, see
     https://developer.android.com/r/tools/jniLibs-vs-imported-targets`

I have on both machines this config : 
`"dependencies": {
    "react": "18.1.0",
    "react-native": "0.70.0",
    "react-native-reanimated": "^2.10.0",
    "react-native-vision-camera": "^2.14.1",
    "vision-camera-code-scanner": "^0.2.0"
  },
  "devDependencies": {
    "@babel/core": "^7.12.9",
    "@babel/runtime": "^7.12.5",
    "@react-native-community/eslint-config": "^2.0.0",
    "babel-jest": "^26.6.3",
    "eslint": "^7.32.0",
    "jest": "^26.6.3",
    "metro-react-native-babel-preset": "^0.72.1",
    "react-test-renderer": "18.1.0"
  },`

Productivix, 2022-09-13:
After reading the link [https://developer.android.com/r/tools/jniLibs-vs-imported-targets`](https://developer.android.com/r/tools/jniLibs-vs-imported-targets%60)
I think the issue comes from the line 137 of buid.gradle of the module (react-native-vision-camera) 
with : 

` if (ENABLE_FRAME_PROCESSORS) {
      externalNativeBuild {
        cmake {
          cppFlags "-fexceptions", "-frtti", "-std=c++1y", "-DONANDROID"
          abiFilters 'x86', 'x86_64', 'armeabi-v7a', 'arm64-v8a'
          arguments '-DANDROID_STL=c++_shared',
                  "-DREACT_NATIVE_VERSION=${REACT_NATIVE_VERSION}",
                  "-DNODE_MODULES_DIR=${nodeModules}",
                  "-DFOR_HERMES=${FOR_HERMES}",
                  "-DJS_RUNTIME_DIR=${jsRuntimeDir}"
        }`

Productivix, 2022-09-13:
solved by : 
https://github.com/software-mansion/react-native-reanimated/issues/3342#issuecomment-1171660650

RRaideRR, 2022-10-15:
Unfortunately, I run into that issue as well. 

I'm using expo with the following dependencies.

```
"expo": "~46.0.16",
"react-native": "0.69.6",
"react-native-vision-camera": "^2.14.1",
```

Running `expo prebuild` works on my machine. However, the eas cloud build fails with the above mentioned error. 

I also tried to add the following to my app.json - still the same behavior (prebuild works, eas cloud build fails)

```
      ["expo-build-properties",
        {
          "android": {
            "packagingOptions": {
              "pickFirst": [
                "/armeabi-v7a/libfolly_runtime.so",
                "/x86/libfolly_runtime.so",
                "/arm64-v8a/libfolly_runtime.so",
                "/x86_64/libfolly_runtime.so"
              ]
            }
          }
        }
      ],
```
 
 Anyone has a helping hand here?

mikedepeel-movista, 2022-10-21:
We've run into this on linux dev machines as well, and while installing from github works around the issue, then it breaks the running of jest mock, etc.  

Is the intent to get this fix in a release, or is it just a temporary fix and we're waiting on something else?

Thanks for the work on this, no pressure intended by any means, only looking to better understand the situation.

Productivix, 2022-10-22:
hi, 
on Linux , and Windows,
with only the stack : 
`"react": "18.1.0",
    "react-native": "0.70.3",
 "react-native-vision-camera": "^2.14.1"
`
I mean without Reanimated ,(I removed the plugin)
 I have removed the added lines  of packagingOptions in app/build.gradle.
and no problems to build.
hope it helps

mayeyun, 2023-12-13:
没有人能解决这个问题，不知道react-native官方能否解决这个问题！

CMangoDH, 2024-12-06:
For the problem "NativeCameraView' cannot be used as a JSX component.",add "// @ts-expect-error" can resolve the error.
```
public render(): React.ReactNode {
    // We remove the big `device` object from the props because we only need to pass `cameraId` to native.
    const { device, frameProcessor, frameProcessorFps, ...props } = this.props;
    return (
      // @ts-expect-error because the type declarations are kinda wrong, no?
      <NativeCameraView
        {...props}
        frameProcessorFps={frameProcessorFps === 'auto' ? -1 : frameProcessorFps}
        cameraId={device.id}
        ref={this.ref}
        onViewReady={this.onViewReady}
        onInitialized={this.onInitialized}
        onError={this.onError}
        onFrameProcessorPerformanceSuggestionAvailable={this.onFrameProcessorPerformanceSuggestionAvailable}
        enableFrameProcessor={frameProcessor != null}
      />
    );
  }
```

PR #1144 (2022-08-10): fix: Build error "2 files found with path '.../libfolly_runtime.so'"
## What
This PR fixes an build error after upgrading to react-native 0.69.1 after the fix in the issue #1118

## Changes
* Adds libfolly_runtime.so library to exclusion list in build.gradle

## Tested on
* Samsung S20 FE, Android 12

## Related issues
* Fixes #1125