margelo/react-native-vision-camera #1126: 🐛

jhongyi, 2022-07-07:
### What were you trying to do?

I get the build error in iOS, and still can't resolve it.
- "react": "17.0.2"
- "react-native": "0.68.2"
- "react-native-vision-camera": "^2.13.5"

Things I have done

I deleted my node_modules folder
rm -rf package-lock.json && rm -rf yarn.lock && rm -rf node_modules
rm -rf ios/Podfile.lock && rm -rf ios/Pods
yarn
cd ios && pod repo update && pod update && pod install

[image]

### Reproduceable Code

```tsx
Undefined symbols for architecture x86_64:
  "typeinfo for reanimated::Scheduler", referenced from:
      typeinfo for vision::VisionCameraScheduler in VisionCameraScheduler.o
  "facebook::react::LongLivedObject::LongLivedObject()", referenced from:
      facebook::react::CallbackWrapper::CallbackWrapper(facebook::jsi::Function&&, facebook::jsi::Runtime&, std::__1::shared_ptr<facebook::react::CallInvoker>) in JSIUtils.o
  "facebook::react::LongLivedObject::~LongLivedObject()", referenced from:
      facebook::react::CallbackWrapper::CallbackWrapper(facebook::jsi::Function&&, facebook::jsi::Runtime&, std::__1::shared_ptr<facebook::react::CallInvoker>) in JSIUtils.o
      facebook::react::CallbackWrapper::~CallbackWrapper() in JSIUtils.o
  "facebook::react::LongLivedObjectCollection::get()", referenced from:
      facebook::react::CallbackWrapper::createWeak(facebook::jsi::Function&&, facebook::jsi::Runtime&, std::__1::shared_ptr<facebook::react::CallInvoker>) in JSIUtils.o
  "typeinfo for facebook::react::LongLivedObject", referenced from:
      typeinfo for facebook::react::CallbackWrapper in JSIUtils.o
  "facebook::react::LongLivedObjectCollection::add(std::__1::shared_ptr<facebook::react::LongLivedObject>) const", referenced from:
      facebook::react::CallbackWrapper::createWeak(facebook::jsi::Function&&, facebook::jsi::Runtime&, std::__1::shared_ptr<facebook::react::CallInvoker>) in JSIUtils.o
  "facebook::jsi::detail::throwJSError(facebook::jsi::Runtime&, char const*)", referenced from:
      std::__1::shared_ptr<facebook::jsi::HostObject> facebook::jsi::Object::asHostObject<facebook::jsi::HostObject>(facebook::jsi::Runtime&) const in JSIUtils.o
  "vtable for reanimated::Scheduler", referenced from:
      reanimated::Scheduler::Scheduler() in VisionCameraScheduler.o
      reanimated::Scheduler::~Scheduler() in VisionCameraScheduler.o
  NOTE: a missing vtable usually means the first non-inline virtual member function has no definition.
  "facebook::jsi::Object::getPropertyAsObject(facebook::jsi::Runtime&, char const*) const", referenced from:
      ___43+[JSConsoleHelper getLogFunctionForBridge:]_block_invoke_2 in JSConsoleHelper.o
  "vtable for facebook::jsi::HostObject", referenced from:
      facebook::jsi::HostObject::HostObject() in FrameProcessorUtils.o
  NOTE: a missing vtable usually means the first non-inline virtual member function has no definition.
  "typeinfo for facebook::jsi::HostObject", referenced from:
      typeinfo for FrameHostObject in FrameHostObject.o
      convertJSIValueToObjCObject(facebook::jsi::Runtime&, facebook::jsi::Value const&, std::__1::shared_ptr<facebook::react::CallInvoker>) in JSIUtils.o
  "typeinfo for facebook::jsi::JSError", referenced from:
      FrameHostObject::assertIsFrameStrong(facebook::jsi::Runtime&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) in FrameHostObject.o
      FrameHostObject::get(facebook::jsi::Runtime&, facebook::jsi::PropNameID const&)::$_1::operator()(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long) const in FrameHostObject.o
      GCC_except_table6 in FrameProcessorUtils.o
      GCC_except_table3 in JSConsoleHelper.o
  "facebook::jsi::HostObject::set(facebook::jsi::Runtime&, facebook::jsi::PropNameID const&, facebook::jsi::Value const&)", referenced from:
      vtable for FrameHostObject in FrameHostObject.o
  "facebook::jsi::HostObject::~HostObject()", referenced from:
      FrameHostObject::~FrameHostObject() in FrameHostObject.o
  "facebook::jsi::JSError::~JSError()", referenced from:
      FrameHostObject::assertIsFrameStrong(facebook::jsi::Runtime&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) in FrameHostObject.o
      FrameHostObject::get(facebook::jsi::Runtime&, facebook::jsi::PropNameID const&)::$_1::operator()(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long) const in FrameHostObject.o
  "facebook::react::LongLivedObject::allowRelease()", referenced from:
      facebook::react::CallbackWrapper::allowRelease() in JSIUtils.o
  "facebook::jsi::JSError::JSError(facebook::jsi::Runtime&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >)", referenced from:
      facebook::jsi::JSError::JSError(facebook::jsi::Runtime&, char const*) in FrameHostObject.o
  "facebook::react::LongLivedObjectCollection::remove(facebook::react::LongLivedObject const*) const", referenced from:
      facebook::react::CallbackWrapper::allowRelease() in JSIUtils.o
  "facebook::jsi::Value::Value(facebook::jsi::Value&&)", referenced from:
      facebook::jsi::Value::operator=(facebook::jsi::Value&&) in JSIUtils.o
  "facebook::jsi::Object::getPropertyAsFunction(facebook::jsi::Runtime&, char const*) const", referenced from:
      ___43+[JSConsoleHelper getLogFunctionForBridge:]_block_invoke_2 in JSConsoleHelper.o
  "facebook::jsi::Value::~Value()", referenced from:
      facebook::jsi::Value::Value<facebook::jsi::Function>(facebook::jsi::Function&&) in FrameHostObject.o
      facebook::jsi::Value::Value<facebook::jsi::String>(facebook::jsi::String&&) in FrameHostObject.o
      ____Z42convertJSIFunctionToFrameProcessorCallbackRN8facebook3jsi7RuntimeERKNS0_8FunctionE_block_invoke in FrameProcessorUtils.o
      facebook::jsi::Value facebook::jsi::Function::callWithThis<facebook::jsi::Object>(facebook::jsi::Runtime&, facebook::jsi::Object const&, facebook::jsi::Object&&) const in FrameProcessorUtils.o
      facebook::jsi::Function::callWithThis(facebook::jsi::Runtime&, facebook::jsi::Object const&, facebook::jsi::Value const*, unsigned long) const in FrameProcessorUtils.o
      facebook::jsi::Value::Value(facebook::jsi::Runtime&, facebook::jsi::Object const&) in FrameProcessorUtils.o
      ___43+[JSConsoleHelper getLogFunctionForBridge:]_block_invoke_2 in JSConsoleHelper.o
      ...
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

### What happened instead?

Build failed

### Device

iPhone 8 (iOS 14.3)

### VisionCamera Version

2.13.5

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

caferyukseloglu, 2022-07-24:
[Check This](https://github.com/mrousavy/react-native-vision-camera/pull/1109)

somiasHtec, 2022-07-25:
Guys, I managed to find a solution. When I implemented firebase into my project, it requested from me to have `use_frameworks! `in Podfile. That was the time I started getting this error. So I deleted `use_frameworks!` and added   `use_modular_headers!`. 

After doing this, I didn't had any errors except some glog issue, so I had to add this in Podfile next to pods:
`  pod 'glog', :podspec => '../node_modules/react-native/third-party-podspecs/glog.podspec', :modular_headers => false
`

The app is finally working now!

caferyukseloglu, 2022-07-25:
> Guys, I managed to find a solution. When I implemented firebase into my project, it requested from me to have `use_frameworks! `in Podfile. That was the time I started getting this error. So I [...]

When I did that FirebaseStore gave error for me evet with custom headers :/

somiasHtec, 2022-07-25:
> > Guys, I managed to find a solution. When I implemented firebase into my project, it requested from me to have `use_frameworks! `in Podfile. That was the time I started getting this error. So I deleted `use_frameworks!` and added `use_modular_headers!`.
> > After doing this, I didn't had any errors except some glog issue, so I had to add this in Podfile next to pods: ` pod 'glog', :podspec => '../node_modules/react-native/third-party-podspecs/glog.podspec', :modular_headers => false`
> > The app is finally working now!
> 
> When I did that FirebaseStore gave error for me even with custom headers :/. Clean build folder Pods and re-instaal

Could you write your error here?

RahulMore-Codes, 2022-07-28:
> Guys, I managed to find a solution. When I implemented firebase into my project, it requested from me to have `use_frameworks! `in Podfile. That was the time I started getting this error. So I [...]

Worked for me also. Thanks 👍🏻

cashcount, 2022-08-15:
> Guys, I managed to find a solution. When I implemented firebase into my project, it requested from me to have `use_frameworks! `in Podfile. That was the time I started getting this error. So I [...]

What a lovely solution. I've been struggling with this for nearly a day! Thank you.

BMwanza, 2022-09-08:
@RahulMore-Codes 

Hello Rahul,

Could you kindly post your PodFile? I am attempting your solution, however, after removing `use_frameworks!` and adding `use_modular_headers!` I still get the glog error so I suspecting I am not putting `pod 'glog', :podspec => '../node_modules/react-native/third-party-podspecs/glog.podspec', :modular_headers => false` in its proper place

Thanks in advance

LazyAfternoons, 2022-09-20:
Same issue here. I tried the propsed solution but it didn't work. My pod file: 
`require_relative '../node_modules/react-native/scripts/react_native_pods'
require_relative '../node_modules/@react-native-community/cli-platform-ios/native_modules'

platform :ios, '13.0'
install! 'cocoapods', :deterministic_uuids => false

production = ENV["PRODUCTION"] == "1"

target 'justallarme' do
  pod 'glog', :podspec => '../node_modules/react-native/third-party-podspecs/glog.podspec', :modular_headers => false
  use_modular_headers!
  config = use_native_modules!

  # Flags change depending on the env values.
  flags = get_default_flags()

  # React Native Maps dependencies
  rn_maps_path = '../node_modules/react-native-maps'
  pod 'react-native-google-maps', :path => rn_maps_path

  use_react_native!(
    :path => config[:reactNativePath],
    # to enable hermes on iOS, change `false` to `true` and then install pods
    :production => production,
    :hermes_enabled => flags[:hermes_enabled],
    :fabric_enabled => flags[:fabric_enabled],
    :flipper_configuration => FlipperConfiguration.enabled,
    # An absolute path to your application root.
    :app_path => "#{Pod::Config.instance.installation_root}/.."
  )

  target 'justallarmeTests' do
    inherit! :complete
    # Pods for testing
  end

  post_install do |installer|
    react_native_post_install(installer)
    __apply_Xcode_12_5_M1_post_install_workaround(installer)
  end
end
`

krisidmisso, 2022-09-30:
> Guys, I managed to find a solution. When I implemented firebase into my project, it requested from me to have `use_frameworks! `in Podfile. That was the time I started getting this error. So I [...]

Can you please share your PodFile? I have the same issue with Firebase installed. Followed the setup example by @mikehardy https://github.com/mikehardy/rnfbdemo/blob/main/make-demo.sh . The app was ok, but Im not able to run the app anymore with Vision Camera in iOS

mikehardy, 2022-09-30:
It appears that react-native-vision-camera may not work with `use_frameworks` then, and that's a react-native-vision-camera issue.

Attempting to work around the issue by removing `use_frameworks` and adding `modular_headers` stuff is a temporary workaround at best

@mrousavy have you tried building any compile-test rig for react-native-vision-camera with `use_frameworks! :linkage => :static` in the Podfile? Seems it may have some symbol location (read as: header pathing...) issue ?

krisidmisso, 2022-09-30:
@mikehardy thank you for the quick reply. Do you suggest any workaround for a specific library (i.e.: VisionCamera) to skip `use_frameworks!`?

Below is my Podfile:

```
require_relative '../node_modules/react-native/scripts/react_native_pods'
require_relative '../node_modules/@react-native-community/cli-platform-ios/native_modules'

platform :ios, '12.4'
install! 'cocoapods', :deterministic_uuids => false

production = ENV["PRODUCTION"] == "1"

target 'APPNAME' do

  config = use_native_modules!
  config = use_frameworks!
  $RNFirebaseAsStaticFramework = true

  # Flags change depending on the env values.
  flags = get_default_flags()

  use_react_native!(
    :path => config[:reactNativePath],
    # to enable hermes on iOS, change `false` to `true` and then install pods
    :production => production,
    :hermes_enabled => flags[:hermes_enabled],
    :fabric_enabled => flags[:fabric_enabled],
#    :flipper_configuration => FlipperConfiguration.enabled(["Debug"], { 'Flipper' => '0.159.0' }),
    :flipper_configuration => FlipperConfiguration.disabled,
    # An absolute path to your application root.
    :app_path => "#{Pod::Config.instance.installation_root}/.."
  )
  # Firebase BoM -> specify global firebase version
  # TODO: update firebase version here too
  $FirebaseSDKVersion = '9.4.0'

  target 'APPNAMETests' do
    inherit! :complete
    # Pods for testing
  end

  post_install do |installer|
    react_native_post_install(installer)
    installer.pods_project.targets.each do |target|
      if (target.name.eql?('FBReactNativeSpec'))
        target.build_phases.each do |build_phase|
          if (build_phase.respond_to?(:name) && build_phase.name.eql?('[CP-User] Generate Specs'))
            target.build_phases.move(build_phase, 0)
          end
        end
      end
    end

    installer.pods_project.targets.each do |target|
      target.build_configurations.each do |config|
        config.build_settings["ENABLE_BITCODE"] = "NO"
      end
    end
    __apply_Xcode_12_5_M1_post_install_workaround(installer)
  end
end

```
I tried:
- removing `use_framworks!`
- removing `use_framworks!` and adding `use_modular_headers!` in its place
- adding the following code to skip the use of `use_frameworks!` for a specific library:
```
dynamic_frameworks = ['VisionCamera'] # <- swift libraries names

# Make all the other frameworks into static frameworks by overriding the static_framework function to return true
pre_install do |installer|
  installer.pod_targets.each do |pod|
    if !dynamic_frameworks.include?(pod.name)
      puts "Overriding the static_framework? method for #{pod.name}"
      def pod.static_framework?;
        true
      end
      def pod.build_type;
        Pod::BuildType.static_library
      end
    end
  end
end
```

mikehardy, 2022-09-30:
Sorry, I do not propose workarounds to `use_frameworks`, I work with maintainers to fix `use_frameworks` compilation in their modules
Avoiding it is not (I repeat *not*) a viable option for any users of react-native-firebase and that's my focus

viljark, 2022-10-20:
After bunch of debugging, I found out **use_frameworks** is also causing iOS runtime crash "`Frame Processors are not enabled`" when trying to use frame processors. 

It seems even if the build succeeds (I'm on RN 0.69, without flipper & without hermes), use_frameworks usage required by react-native-firebase@15 will cause issues in runtime.

I also tried to [set the related packages to be static](https://github.com/rodgomesc/vision-camera-code-scanner/issues/62#issuecomment-1272598027) in the podfile, but it did not help.

mikehardy, 2022-10-20:
@mrousavy not sure if you are aware of the use_frameworks issue or not? I wish I could wave a magic wand and remove the requirement for use_frameworks in react-native-firebase but I cannot. Would be great if this package supported it - usually it's just header + linkage location differences that need resolving

KonstantinZhukovskij, 2022-10-25:
any news here?

closed 2022-10-25 by jhongyi

bglgwyng, 2022-10-29:
Is this issue resolved?
I just tried `use_frameworks!` with react-native-vision-camera@2.15.1 but it failed. The error message was the same.

mikehardy, 2022-10-29:
@bglgwyng I saw it closed and looked for anything in recent commits that would indicate a change had been made that would effect it. I did not see anything so I was surprised it was closed, myself. But I'm not a maintainer here so I don't really know. I think it is likely still an issue.

bglgwyng, 2022-10-31:
I tried a very similar workaround here https://github.com/ICoi/VisionCameraExample/blob/main/ios/Podfile#L10 with what @krisidmisso suggested. The build succeded, but I got the following error message

```
 ERROR  frame-processor/unavailable: Frame Processors are not enabled. See https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting, js engine: hermes
```

So I commented frame processor things in the code and the app worked fine. If migration from static to dynamic library is a hard work, then to enhance this workaround to work well with frame processor seems the best option for now.

bglgwyng, 2022-11-02:
@hb-webdev Sorry. I shared a private one. You can see this one https://github.com/bglgwyng/VisionCameraExample.
The latest commit also solved the frame processor issue by setting RNReanimated as static.

I haven't tested with firebase yet. I think I'll do it today.

mrousavy, 2022-11-02:
yea RNReanimated pod is the issue here, it can't be used with frameworks afaik - maybe there's a workaround

bglgwyng, 2022-11-02:
https://github.com/bglgwyng/FirebaseExample I tried firebase + vision camera + frame processor here. It works fine. Firestore client is ok and the camera is too.
 @mrousavy My workaround was just setting `RNAnimated` to be built as a static library and there seems no problem for now.

mikehardy, 2022-11-02:
Nice!

I see this block, doing the thing you mention:

```ruby

  pre_install do |installer|
    installer.pod_targets.each do |pod|
      if pod.name.eql?('vision-camera-code-scanner') || pod.name.eql?('VisionCamera') || pod.name.eql?('RNReanimated')
        def pod.build_type
          Pod::BuildType.static_library
        end
      end
    end
  end
```

Could be similar to our workaround in react-native-firebase [here](https://github.com/invertase/react-native-firebase/blob/d654d00e06f7d638828653cfaa64ef9ad5e343d9/packages/app/RNFBApp.podspec#L36-L41), though they look a little different:

```ruby
  if defined?($RNFirebaseAsStaticFramework)
    Pod::UI.puts "#{s.name}: Using overridden static_framework value of '#{$RNFirebaseAsStaticFramework}'"
    s.static_framework = $RNFirebaseAsStaticFramework
  else
    s.static_framework = false
  end
```

It is exactly the same as the workaround (which also works) documented in react-native-permissions:
https://github.com/zoontek/react-native-permissions#workaround-for-use_frameworks-issues

So...possibly could be solved in podspec here (somehow, though not aware of an elegant way yet) *or* could be a documentation solution similar to react-native-permissions (though there is a little issue tracker traffic related to it when people aren't aware)...

krisidmisso, 2022-12-17:
@hb-webdev thanks to @bglgwyng and @mikehardy I was able to run my application with Firebae and Camera libs today.
Here is my final Podfile:

```
require_relative '../node_modules/react-native/scripts/react_native_pods'
require_relative '../node_modules/@react-native-community/cli-platform-ios/native_modules'

$FirebaseSDKVersion = '10.2.0'

platform :ios, '12.4'
install! 'cocoapods', :deterministic_uuids => false

production = ENV["PRODUCTION"] == "1"

# see react-native-permission for more info
use_frameworks! :linkage => :static

$RNFirebaseAsStaticFramework = true

# Convert all permission pods into static libraries
pre_install do |installer|
  Pod::Installer::Xcode::TargetValidator.send(:define_method, :verify_no_static_framework_transitive_dependencies) {}

  installer.pod_targets.each do |pod|
    if pod.name.eql?('RNPermissions') || pod.name.start_with?('Permission-') || pod.name.eql?('vision-camera-code-scanner') || pod.name.eql?('VisionCamera') || pod.name.eql?('RNReanimated')
      def pod.build_type;
        Pod::BuildType.static_library
      end
    end
  end
end

target 'MyApp' do

  config = use_native_modules!

    # Flags change depending on the env values.
    flags = get_default_flags()
    flags[:hermes_enabled] = false

    use_react_native!(
      :path => config[:reactNativePath],
      # to enable hermes on iOS, change `false` to `true` and then install pods
      :production => production,
      :hermes_enabled => flags[:hermes_enabled],
      :fabric_enabled => flags[:fabric_enabled],
      :flipper_configuration => FlipperConfiguration.disabled,
      # An absolute path to your application root.
      :app_path => "#{Pod::Config.instance.installation_root}/.."
    )

    permissions_path = '../node_modules/react-native-permissions/ios'
    pod 'Permission-Camera', :path => "#{permissions_path}/Camera"

  target 'MyAppTests' do
    inherit! :complete
    # Pods for testing
  end

  post_install do |installer|

  #disable bitcode as it will be depricated in Xcode14
    installer.pods_project.targets.each do |target|
      target.build_configurations.each do |config|
        config.build_settings['ENABLE_BITCODE'] = 'NO'
      end
    end

    react_native_post_install(installer)
    __apply_Xcode_12_5_M1_post_install_workaround(installer)
    installer.pods_project.build_configurations.each do |config|
      config.build_settings["EXCLUDED_ARCHS[sdk=iphonesimulator*]"] = "arm64"
    end
  end
end

```

Note the long condition:
```
...
if pod.name.eql?('RNPermissions') || pod.name.start_with?('Permission-') || pod.name.eql?('vision-camera-code-scanner') || pod.name.eql?('VisionCamera') || pod.name.eql?('RNReanimated')
...
```
Replace it with packages of your app that have issues with `use_frameworks!`

Hope this helps.

VictorThales, 2022-12-21:
@krisidmisso, Thanks Bro, your podfile example help me to make build sucessfull, i lost any hours with this.
I runned before build starts too:
` src/ios $ pod clean && pod deintegrate && pod install`