margelo/react-native-vision-camera #1132: How to change Android path name❓

somiasHtec, 2022-07-11:
### Question

When I'm taking a photo and I want to save it, it has this path:

`/data/user/0/com.myapp/cache/mrousavy8999427678902473567.jpg`

Is there a way to remove **mrousavy** from the path ?

### VisionCamera Version

2.13.3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

micm0, 2022-07-20:
Hello @mrousavy, do you plan to add a path property to TakePhotoOptions ?

fudr, 2022-09-11:
> Ah whoops probably left that in there back when I was developing the lib - no there's currently no way to change my branding there it has to be edited in the code 🙈

Hi, how do I get the final path after CameraRoll.save ?

mindrootstech, 2022-09-14:
Did you get the Image from path "/data/user/0/com.myapp/cache/mrousavy8999427678902473567.jpg" ?
I got a warning in android, working fine in iOS
Warning: Image source "file path" doesn't exist

any solution?
please suggest to me what I should do.

mrousavy, 2022-09-14:
Try with `file://` prefix

mindrootstech, 2022-09-14:
> Try with `file://` prefix

Thanks, working fine

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

closed 2023-09-30 by mrousavy