margelo/react-native-vision-camera #1134: 🐛 Unable to build  in  react native 0.69.1

rvuyyuru1, 2022-07-11:
### What were you trying to do?

error Failed to install the app. Make sure you have the Android development environment set up: 

FAILURE: Build failed with an exception.

..\node_modules\react-native-vision-camera\android\build.gradle' line: 231A problem occurred evaluating project ':react-native-vision-camera'.

> Expected directory '..\node_modules/react-native/android' to contain exactly one file, however, it contains more than one file.

### What happened instead?

unable to build the app

### Device

android 

### VisionCamera Version

2.13.5

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Manojkumar-Ainqa, 2022-07-12:
Hi @rvuyyuru1. Is the above issue solved for you?

rvuyyuru1, 2022-07-12:
> Hi @rvuyyuru1. Is the above issue solved for you? No...  using react-native-camera

rvuyyuru1, 2022-07-12:
#1116 check this

Manojkumar-Ainqa, 2022-07-12:
okay @rvuyyuru1, I think the react-native-camera is not working for react native 0.69.1. We need to downgrade the react-native version to 68. Also, the above PR is not merged right.

rvuyyuru1, 2022-07-12:
yep

rvuyyuru1, 2022-07-12:
i will check with it and update back

rvuyyuru1, 2022-07-12:
no Task :react-native-vision-camera:configureCMakeDebug[arm64-v8a] FAILED for me

Manojkumar-Ainqa, 2022-07-12:
You're using the react-native version 0.69.1 right?

Manojkumar-Ainqa, 2022-07-12:
Is the npx patch-package command running successfully?
Reference: [#1083](https://github.com/mrousavy/react-native-vision-camera/issues/1083)

rvuyyuru1, 2022-07-12:
#908 this helped me

Manojkumar-Ainqa, 2022-07-13:
Hi @nica0012 , Yes I'm also facing the same problem for iOS. But the patched package is working for Android alone. I'm also waiting for the iOS fix :)

aha85b, 2022-07-19:
> I cannot build on this version with IOS.
> 
> I am getting the following error in xCode:
> 
> `'react/bridging/CallbackWrapper.h' file not found

Same issue.

hengkx, 2022-07-21:
0.69.2 Same issue.

bglgwyng, 2022-07-24:
I found that `react-native-vision-camera` alone worked well, but after installing `react-native-reanimated`, it started to emit the error message reported here.

https://github.com/bglgwyng/visionCameraReport
It only contains scaffolded code from `react-native` cli and `react-native-vision-camera` and `react-native-reanmiated` added to package.json. Clone this repo and try `npm i && npm run android`, then it will fail with the same error. Then you can remove `react-native-reanimated` dependency, and it will work well.

 I guess the problem is with `react-native-reanimated`.

danielMuffin, 2022-07-26:
rn 0.69.3 fix this

tokkozhin, 2022-07-26:
RN 0.69.3 fix this

BartoszDombrow, 2022-07-26:
> RN 0.69.3 fix this

Still same issue

doublex, 2022-07-26:
0.69.3 works for me

BartoszDombrow, 2022-07-26:
> 0.69.3 works for me

Did you disable frame processors or just upgrade RN to 0.69.3?

doublex, 2022-07-26:
Fresh install (0.69.2 failed)

alexco2, 2022-07-26:
I still have the same problem with version 0.69.3 with this error

```
Build file '.../node_modules/react-native-vision-camera/android/build.gradle' line: 231

A problem occurred evaluating project ':react-native-vision-camera'.
> Expected directory '.../node_modules/react-native/android' to contain exactly one file, however, it contains more than one file.
```

Which is this file

` def rnAAR = fileTree("${nodeModules}/react-native/android").matching({ it.include "**/**/*.aar" }).singleFile`

alkunjo, 2022-08-01:
> @Manojkumar-Ainqa Thank you, I have been keeping my eyes peeled for the next update. I think I'll wait this week otherwise I will downgrade to RN v68.

Hi @nica0012 have you got it work using RN 68? I've tried on newly created RN 0.68.2 and got the same issue
I installed it with `react-native-reanimated`

Zhi0105, 2022-08-04:
> Hi @nica0012 , Yes I'm also facing the same problem for iOS. But the patched package is working for Android alone. I'm also waiting for the iOS fix :)

same issue

criss02-cs, 2022-08-05:
@alexco2 I have the same issue

alexco2, 2022-08-10:
Fixed with newest release of vision-camera

closed 2022-08-10 by mrousavy

jason840917gmail, 2023-01-24:
Same issue with RN 0.71.1