margelo/react-native-vision-camera #1138: 🐛 An unknown camera error occured.

Jerald042615, 2022-07-12:
### What were you trying to do?

Can't detect external camera/usb camera in my react native app.

Please help thanks.

### Reproduceable Code

```tsx
An unknown camera error occured.
```

### What happened instead?

An unknown camera error occured.

### Relevant log output

```shell
An unknown camera error occured.
```

### Device

iMin

### VisionCamera Version

Latest Version

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-07-18:
Hi, you did not share the relevant log output, how do you expect me to help you? I cannot see your screen from over there, and I don't have your device. I don't have that error, and I don't have any logs at all.

mrousavy, 2022-07-18:
If you want me to help you you need to provide enough details. **Read** the stuff I write, in the issue template it says that you need to scan ADB logcat for logs.

closed 2022-07-18 by mrousavy

liancp, 2023-01-28:
An unknown camera error occured.

liancp, 2023-01-28:
2023-01-28 14:21:42.888  1292-2082  System.err              com.mrs                              W  java.lang.NullPointerException
2023-01-28 14:21:42.889  1292-2082  System.err              com.mrs                              W  	at com.mrousavy.camera.utils.CameraCharacteristicsUtilsKt.getDeviceTypes(CameraCharacteristicsUtils.kt:30)
2023-01-28 14:21:42.889  1292-2082  System.err              com.mrs                              W  	at com.mrousavy.camera.CameraViewModule$getAvailableCameraDevices$1.invokeSuspend(CameraViewModule.kt:184)
2023-01-28 14:21:42.889  1292-2082  System.err              com.mrs                              W  	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
2023-01-28 14:21:42.889  1292-2082  System.err              com.mrs                              W  	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
2023-01-28 14:21:42.889  1292-2082  System.err              com.mrs                              W  	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:571)
2023-01-28 14:21:42.889  1292-2082  System.err              com.mrs                              W  	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:750)
2023-01-28 14:21:42.889  1292-2082  System.err              com.mrs                              W  	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:678)
2023-01-28 14:21:42.889  1292-2082  System.err              com.mrs                              W  	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:665)
2023-01-28 14:21:43.094  1292-4059  unknown:Re...gWebSocket com.mrs                              W  Couldn't connect to "ws://localhost:8081/message?device=rk3399-Android10%20-%2010%20-%20API%2029&app=com.mrs&clientid=DevSupportManagerImpl", will silently retry
2023-01-28 14:21:43.408  1292-1656  ReactNativeJS           com.mrs                              I  FaceIdentification Component | Camera permission status: authorized
2023-01-28 14:21:44.318  1292-1656  ReactNativeJS           com.mrs                              I  Device Utils | 设备型号: rk3399-Android10
2023-01-28 14:21:44.902  1292-1656  ReactNativeJS           com.mrs                              W  Possible Unhandled Promise Rejection (id: 24):
                                                                                                    unknown/unknown: [unknown/unknown] An unknown camera error occured.
                                                                                                    [unknown/unknown]: [unknown/unknown] An unknown camera error occured.
                                                                                                        at _construct (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:20966:39)
                                                                                                        at apply (native)
                                                                                                        at _construct (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:20971:28)
                                                                                                        at Wrapper (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:20933:25)
                                                                                                        at apply (native)
                                                                                                        at _createSuperInternal (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:229753:382)
                                                                                                        at call (native)
                                                                                                        at CameraError (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:229761:26)
                                                                                                        at apply (native)
                                                                                                        at _createSuperInternal (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:229753:382)
                                                                                                        at apply (native)
                                                                                                        at CameraRuntimeError (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:229810:27)
                                                                                                        at tryParseNativeCameraError (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:229834:38)
                                                                                                        at getAvailableCameraDevices$ (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:229625:66)
                                                                                                        at call (native)
[... cut ...]
                                                                                                        at __guard (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:2237:15)
                                                                                                        at flushedQueue (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:2062:21)
                                                                                                        at invokeCallbackAndReturnFlushedQueue (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:2056:33)