margelo/react-native-vision-camera #1139: 🐛 After adding react-native-vision-camera app does not build(event after checking troubleshooting and existing issues)

qeorqer, 2022-07-12:
### What were you trying to do?

I need to add `react-native-vision-camera` to my ios app. I followed through the installation guide from the docs(installed library, then pods, added `react-native-reanimated` to my `package.json` as well). Checked everything from the troubleshooting page and it did not helped, the app still doesn't build. Please tell if you need any other information from my side.
Also, I've seen the similar issue with `folly/Optional.h`, but the solution did not help me(or I did not understood its usage)

Dependencies and versions:
"react-native": "0.63.3",
"react-native-reanimated": "^2.9.1",
"react-native-vision-camera": "^2.13.5",

SWIFT_VERSION = 5.2;
platform :ios, '12.0'

### What happened instead?

After running `npx react-native run-ios`
[image]

After running Build from xcode
[image]

### Device

iPhone 11 (15.4)

### VisionCamera Version

2.13.5

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

MatheusHCP, 2022-07-13:
I'm having the same problem after adding the library to my project.

aha85b, 2022-07-14:
I am having the same issue, I created a new react native project using npx react-native init Project then I installed react-native-vision-camera and followed all their steps, troubleshooting, etc. and still getting the same error. react native version is 0.69.1 - This error I am getting with iOS only. I had no issue with Android.

Ashok-Marimuthu, 2022-07-19:
I am also facing the same issue. after installing the **react-native-reanimated**, when I build the android using **./gradlew clean** cmd then I got the below error.

`> Configure project :react-native-vision-camera
VisionCamera: node_modules/ found at: /Users/ashok/Project/DIGID_RN_MOBILE_AGENT/node_modules
VisionCamera: Building with JSC...
Warning: The 'kotlin-android-extensions' Gradle plugin is deprecated. Please use this migration guide (https://goo.gle/kotlin-android-extensions-deprecation) to start working with View Binding (https://developer.android.com/topic/libraries/view-binding) and the 'kotlin-parcelize' plugin.
VisionCamera: Frame Processors are enabled! Building C++ part...
WARNING:DSL element 'dexOptions' is obsolete and should be removed.
It will be removed in version 8.0 of the Android Gradle plugin.
Using it has no effect, and the AndroidGradle plugin optimizes dexing automatically.
[CXX1104] NDK from ndk.dir at /Users/ashok/Library/Android/sdk/ndk/21.3.6528147 had version [21.3.6528147] which disagrees with android.ndkVersion [21.4.7075529]
[CXX1104] NDK from ndk.dir at /Users/ashok/Library/Android/sdk/ndk/21.3.6528147 had version [21.3.6528147] which disagrees with android.ndkVersion [21.4.7075529]

FAILURE: Build failed with an exception.

* What went wrong:
A problem occurred configuring project ':react-native-vision-camera'.
> [CXX1104] NDK from ndk.dir at /Users/ashok/Library/Android/sdk/ndk/21.3.6528147 had version [21.3.6528147] which disagrees with android.ndkVersion [21.4.7075529]
`

timoschlueter, 2022-10-03:
I ran into this issue today as well. Has anyone managed to get this resolved yet?

qeorqer, 2022-10-03:
@timoschlueter I didn't find the solution. I used react-native-camera library instead, it worked perfectly for me

timoschlueter, 2022-10-03:
@qeorqer thats the library i wanted to replace with this one, since `react-native-camera` seems no longer to be maintained.
But I figured out how to get around the issue: https://github.com/amplitude/Amplitude-ReactNative/issues/52#issuecomment-803857119
Simply adding the Swift Bridging headers solved the problem for me. `react-native-vision-camera` works without any problem now :)

abdugaffor-abdurahimov, 2023-03-10:
In my case, Hermes was disabled  after I enabled the build fixed 
changed podfile `use_react_native`
```   :hermes_enabled => false``` => ```:hermes_enabled => flags[:hermes_enabled]```

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

closed 2023-09-30 by mrousavy