margelo/react-native-vision-camera #1143: 🐛 The App close itself without any error when camera isActive = true

fathonitr, 2022-07-13:
### What were you trying to do?

Tried to open the camera to use the barcode feature, the app then closed it self without any error being detected on the emulator, metro and adb logcat.  
When I change the value from isActive to false, the issue gone, no more closing without error. But of course, no camera.

### Reproduceable Code

```tsx
import React, { useState, useEffect } from 'react';
import { View, Button, Text, StatusBar, StyleSheet, Alert } from 'react-native';
import { useCameraDevices, Camera } from 'react-native-vision-camera';
import { useScanBarcodes, BarcodeFormat } from 'vision-camera-code-scanner';
import { useStore } from '@utils/states';
import { useIsFocused } from '@react-navigation/native';

const SerialNumberScanner = ({ navigation }) => {
  const isFocused = useIsFocused();
  const devices = useCameraDevices();
  const device = devices.back;
  const setResultSerialNumber = useStore(state => state.setResultSerialNumber);
  const resultSerialNumber = useStore(state => state.resultSerialNumber);
  const [frameProcessor, barcodes] = useScanBarcodes([
    BarcodeFormat.ALL_FORMATS, // You can only specify a particular format
  ]);
  const [hasPermission, setHasPermission] = useState(false);
  const [tempResult, setTempResult] = useState('');

  useEffect(() => {
    (async () => {
      const status = await Camera.requestCameraPermission();
      console.log("#######################################");
      setHasPermission(status === 'authorized');
    })();
  }, []);
  useEffect(() => {
    /**
     * Length of barcode is checked before setResultSerialNumber
     * so it doesn't store empty string everytime barcodes is changed
     */
    if (barcodes.length > 0) {
      barcodes.forEach(barcode => setTempResult(barcode.displayValue ?? ''));
    } else {
      setTempResult('');
    }
  }, [barcodes]);

  return (
    device != null &&
    hasPermission && (
      <View style={styles.body}>
        <StatusBar barStyle="light-content" backgroundColor="#000000" />
        <Camera
          style={StyleSheet.absoluteFill}
          device={device}
          isActive={true}
          frameProcessor={frameProcessor}
          frameProcessorFps={5}
          audio={false}
        />

        {barcodes.map((barcode, idx) => (
          <Text key={idx} style={styles.barcodeTextURL}>
            {barcode.displayValue}
          </Text>
        ))}

        <View style={styles.buttons}>
          {tempResult !== '' && (
            <>
              <Text style={styles.result}>{tempResult}</Text>
              <Button
                title="Got It"
                color="#1eb900"
                onPress={() => [
                  setResultSerialNumber(tempResult),
                  console.log(tempResult),
                ]}
              />
            </>
          )}
```

### What happened instead?

I still have a backup local branch where everything works, but I also have a new branch with the newest update from the main branch. The second one is where this weird thing happend.
I've compare side by side between these two branch, there are no differences on Android/IoS folder, so no changes on Build.Gradle and AndroidManifest.
The plugin for Reanimated in Babel.Config is at the end of the list, I've also imported the Reanimated in the index.js.

I've tried:
1. Remove node_modules and yarn.lock
2. Run yarn cache clean
3. Run yarn
4. Run cd android/
5. Run ./gradlew clean 
6. Run cd ..
7. Wipe the data from my emulator device on android studio's device manager
8. Run yarn react-native start --resetCache
9. Run yarn react-native run-android

Changing the isActive value to false stop the app from closing itself.

### Relevant log output

```shell
ADB Logcat, the metro record no error at all, and the emulator also show no error pop-up.
```

### Device

Android Emulator, Android 11

### VisionCamera Version

2.13.5

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-07-18:
Hey! There has to be an error in `adb logcat`, make sure to run `adb logcat` **before launching the app**, then launching the app, and after the crash copy every line that happened in-between. Then share that here.

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy