margelo/react-native-vision-camera #1145: Type mismatch in type QrCode🐛

Abscessus89, 2022-07-14:
### What were you trying to do?

I try to get content from QR code. 

frame looks like 
{
"boundingBox":{
"bottom":596,
"left":163,
"right":309,
"top":456
},
"content":{
"content":"JWT token",
"type":7
},
"cornerPoints":[
{
"x":163,
"y":467
},
{
"x":295,
"y":456
},
{
"x":309,
"y":586
},
{
"x":182,
"y":596
}
],
"displayValue":"JWT token",
"rawValue":"JWT token"
}

so need to add "content" property or extend type QrCode from another type with these properties included, or change spelling from data to content. "Data" field doesn't exist in "content field".

### Reproduceable Code

```tsx
const frameProcessor = useFrameProcessor((frame) => {
    'worklet';
    const qrcode = scanQRCodes(frame);
    if (!!qrcode[0]) {
      authStore.loginQr(qrcode[0].content?.content);
    }
  }, []);
```

### What happened instead?

Property 'content' does not exist on type '{ type: QrCodeType.UNKNOWN | QrCodeType.ISBN | QrCodeType.TEXT; data: string; } | { type: QrCodeType.CONTACT_INFO; data: QrCodeContactInfo; } | ... 7 more ... | { ...; }'.
  Property 'content' does not exist on type '{ type: QrCodeType.UNKNOWN | QrCodeType.ISBN | QrCodeType.TEXT; data: string; }'

### Device

HUAWEI P30 lit

### VisionCamera Version

"react-native-vision-camera": "^2.13.5"

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-07-18:
Hey! This is not an issue of VisionCamera, but rather of the third party Frame Processor Plugin you are using.

closed 2022-07-18 by mrousavy

mrousavy, 2023-10-06:
Hey!

JFYI; **VisionCamera V3 now includes a QR/Barcode Scanner!** 😍  Check out the [CodeScanner Documentation](https://www.react-native-vision-camera.com/docs/guides/code-scanning) 🚀 

Try upgrading to V3 if you can, it has a much more stable yet flexible foundation.

If you appreciate me dedicating my free time to improving VisionCamera and implementing features like the Code Scanner, please consider [**sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy) to show your support.