margelo/react-native-vision-camera #1148: Why everytime i run `react-native run-android` i need to download an enormous tar in `:downloadGlog` task❓

hkar19, 2022-07-17:
### Question

this particular task is making my build time so long and uses enormous data to download:

```bash
> Task :react-native-vision-camera:downloadGlog
Download https://github.com/google/glog/archive/v0.3.5.tar.gz
<===========--> 90% EXECUTING [1m 29s]
> IDLE
> :react-native-vision-camera:downloadBoost > boost_1_76_0.tar.gz > 7.99 MB/124.24 MB downloaded
> IDLE
> IDLE
> IDLE
> IDLE
```

is there any way i can mitigate this such as no need to download when i already have it in my machine?

### VisionCamera Version

2.13.5

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

xyylb, 2022-07-23:
我也遇到了和你一样的问题，查了相关资料，然后在/android/gradle.properties尾部增加了以下设置，后面构建速度快了不少

# 开启编译缓存
org.gradle.caching=true
# 构建缓存
android.enableBuildCache = true

# 分配更大的内存
org.gradle.jvmargs=-Xmx4g -XX:MaxMetaspaceSize=1g -Dfile.encoding=UTF-8

# 开启并行
org.gradle.parallel=true

# 守护进程
# Gradle 是基于 JVM 的构建系统，JVM 的启动和初始化需要时间，开启 Gradle Daemon 守护进程可以节省这些时间
org.gradle.daemon = true

# 按需配置
org.gradle.configureondemand=true

hkar19, 2022-07-25:
> 我也遇到了和你一样的问题，查了相关资料，然后在/android/gradle.properties尾部增加了以下设置，后面构建速度快了不少
>
> # 开启编译缓存
> org.gradle.caching=true
>
> # 构建缓存
> android.enableBuildCache = true
>
> # 分配更大的内存
> org.gradle.jvmargs=-Xmx4g [...]

can you please explain in english?

i tried using google translate to understand your reply, and applied it in my gradle.

this is what i got

```bash
A problem occurred evaluating project ':app'.
> Failed to apply plugin 'com.android.internal.application'.
   > The option 'android.enableBuildCache' is deprecated.
     The current default is 'false'.
     It was removed in version 7.0 of the Android Gradle plugin.
     The Android-specific build caches were superseded by the Gradle build cache (https://docs.gradle.org/current/userguide/build_cache.html)

```

xyylb, 2022-07-25:
Gradle 7.0 不支持这项设置 android.enableBuildCache = true 需要删除

hkar19, 2022-07-25:
> Gradle 7.0 不支持这项设置 android.enableBuildCache = true 需要删除

so i applied everything you suggested, but i commented on
`org.gradle.caching=true` and `android.enableBuildCache = true` as both are deprecated.

my gradle task still do the `dowloadBooster`  everytime i installed new module.

is there a workaround for Gradle 7.0?

SectionTN, 2022-08-15:
### Temporarily Workaround:
- Go to `node_modules/react-native-vision-camera/android/build/downloads`
- Copy all `*.tar.gz` to a safe folder like `/home/${yourUsername}/gradleThings`
- When you install a new library and you know that gradle will download glog again, just copy them from the safe folder.
- Gradle should found and extract them as he do always!

Tzng, 2022-08-22:
This is really too slow to download

SectionTN, 2022-08-22:
> ### Temporarily Workaround:
> - Go to `node_modules/react-native-vision-camera/android/build/downloads`
> - Copy all `*.tar.gz` to a safe folder like `/home/${yourUsername}/gradleThings`
> - When you [...]

this workaround working for me

billnbell, 2022-08-29:
This 2.10.* is a mess. Do no harm. When fabric is turned off it should continue to work !!

agostinhodev, 2022-09-01:
I'm facing the same problem here. Every time I run `expo run:android` for the first time, the speed is so fast! However, if I need to install a native package and run the same command again, it turns so slow. I believe that they restrict the download according your IP if you run this command for several times. I tried again using other VPN IP and it runs so fast. But it's so boring to do every time you need to build your app :(

mandeepsingh-byjus, 2022-09-07:
> this workaround working for me

This worked for me even for ```react-native-reanimated```
Thanks @SectionTN for adding here, even it is a hack but saves a lot of time while build

Orange9000, 2022-09-21:
> ### Temporarily Workaround:
> * Go to `node_modules/react-native-vision-camera/android/build/downloads`
> * Copy all `*.tar.gz` to a safe folder like `/home/${yourUsername}/gradleThings`
> * When you [...]

You may also add it as a script to your `package.json`

For example:
`cp /Users/{{your username}}/projs/boost_1_76_0.tar.gz /Users/{{your username}}/projs/{{your project folder}}/node_modules/react-native-vision-camera/android/build/downloads/`

SectionTN, 2022-09-21:
I'm not familiar with scripts, but your workaround is helpful too, thanks for posting it here!

robertolaru, 2022-10-05:
This problem is caused when installing a new npm dependency. The packages inside node_modules get reset to their default state. Those large packages that Gradle re-downloads are downloaded to `node_modules/expo-modules-core/android/build`, so when installing a new npm package, the contents of that directory are deleted, forcing Gradle to re-download them on the next build.

I have opened an issue about this on Expo's repo: https://github.com/expo/expo/issues/19422

DaveWoody, 2022-11-09:
Still an issue.

agostinhodev, 2022-11-13:
@SectionTN  Your hack suggestion works here out of box! Tks!

vzts, 2022-11-16:
Now since with https://github.com/mrousavy/react-native-vision-camera/pull/1306 (applied at [v2.15.2](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v2.15.2))
you only need to add as such:
`export REACT_NATIVE_DOWNLOADS_DIR="$HOME/react-native-downloads"`
into your shell profile.
Run `run-android`, and it will download the needed content to the specified path.
The downloaded content will be reused next time.

vzts, 2022-11-16:
Now since with https://github.com/mrousavy/react-native-vision-camera/pull/1306 (applied at [v2.15.2](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v2.15.2)), you only need to add as such:
`export REACT_NATIVE_DOWNLOADS_DIR="$HOME/react-native-downloads"`
into your shell profile.

Run `run-android`, and it will download the needed content to the specified path.
The downloaded content will be reused next time when you run it.

obasille, 2022-11-17:
Hi! I'm running on Windows, I've set `REACT_NATIVE_DOWNLOADS_DIR` environment variable to a valid directory but I'm still having the issue.
The files are still being downloaded in `android\build\downloads`

react-native: 0.69.7
react-native-reanimated: 2.10.0
react-native-vision-camera: 2.15.2

josej14, 2022-12-05:
React-native-reanimated faced the same issue ([ref here](https://github.com/software-mansion/react-native-reanimated/issues/3523)). They resolved this with [these code changes](https://github.com/software-mansion/react-native-reanimated/pull/3588/files). Please, let us know if you're planning a new release with a fix soon to skip an additional script/patch on our app. Thanks!

Jaybee4real, 2023-02-01:
Hello, this is still an issue in 2023, are we expecting an update?, is the matter being investigated?

jimitshah-dp, 2023-02-08:
Can we look into this issue, or have a workaround.
It takes 45-55 mins to run react-native run-android

bhavzie, 2023-02-11:
Is this being looked into?

docvhcmc, 2023-02-14:
jusst remove it and it works for me  0_0

ngotra2710, 2023-02-20:
> I'm facing the same problem here. Every time I run `expo run:android` for the first time, the speed is so fast! However, if I need to install a native package and run the same command again, it [...]

This help, thanks. I install proton VPN and it fast as ***

vanenshi, 2023-02-24:
`react-native-reanimated` had this problem once, they seems to fixed it in next version https://github.com/software-mansion/react-native-reanimated/issues/3523
 
 maybe we can use their approach to fix this

leiMsnow, 2023-04-21:
> ### Temporarily Workaround:
> * Go to `node_modules/react-native-vision-camera/android/build/downloads`
> * Copy all `*.tar.gz` to a safe folder like `/home/${yourUsername}/gradleThings`
> * When you [...]

thanks, it's working for me,  i added a scripts to automate these copy operations:

change **package.json scripts** :
1. android
`
   "android": "react-native run-android && chmod +x ./scripts/copy-gradle-things.sh && ./scripts/copy-gradle-things.sh run"
`
2. posiinstall 
`
"postinstall": "chmod +x ./scripts/copy-gradle-things.sh && ./scripts/copy-gradle-things.sh install"
`

3. add copy-gradle-things.sh

```

#!/bin/bash

echo "copy gralde ache files: $1"

if [ "$1" = "run" ]
then
  echo "start copy gradle cache to localfile"
  mkdir -p gradleThings/react-native-vision-camera
  cp -r ./node_modules/react-native-vision-camera/android/build/downloads/ ./gradleThings/react-native-vision-camera/
fi

if [ "$1" = "install" ]
then
  echo "start copy localfile cache to gradle"
  cp -r ./gradleThings/react-native-vision-camera/ ./node_modules/react-native-vision-camera/android/build/downloads/
fi
```

sheltonsuen, 2023-07-13:
This issue has been fixed by the #1306, and you can define a env: `REACT_NATIVE_DOWNLOADS_DIR` to change the downlaod dir to a global place so that it will not downlaod every time you run.

By the way, although the v3-rc will not have the kind of problem, but it's still in rc and will only support ios for now, think twice before you upgrade

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

closed 2023-09-30 by mrousavy