margelo/react-native-vision-camera #1150: 🐛 Black screen flashes for a second before camera appears

vladimir-vasilev, 2022-07-19:
### What were you trying to do?

I want to use back Camera

### Reproduceable Code

```tsx
const [hasPermission, setHasPermission] = useState(false);
  const devices = useCameraDevices();
  const device = devices.back;

  useEffect(() => {
    (async () => {
      const status = await Camera.requestCameraPermission();
      setHasPermission(status === 'authorized');
    })();
  }, []);

  return device && hasPermission
    ? <Camera style={StyleSheet.absoluteFill} device={device} isActive={true} />
    : null;
```

### What happened instead?

Black screen flashes for a second before camera appears. Reproduces on mNY devices and simulators, on iOS and android.

"react-native": "0.68.2"

### Device

iPhone 11 (iOS 15.0), Samsung Galaxy S20 FE (Android 12)

### VisionCamera Version

2.13.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

thomas-coldwell, 2022-07-19:
You are rendering `null` and the request for camera permission is async so it will render initially without displaying the camera component and then (if it has the permission) render the camera component. A simple way to get around this would be to check the permission before navigating to the screen where you have the camera in the first place or just display a permission place holder rather than `null`?

vladimir-vasilev, 2022-07-19:
yes, but I see the issue in Camera component, when it's rendered. I tried with different screen as well.

rmyr, 2022-08-11:
Having the same issue, short flickers (multiple) of black that actually also overflows its parent container which looks kinda ugly (even if the camera is given props for dimensions like 100px that are easily within the bounds of the parent containerl).

mssoliveira, 2022-12-22:
This is happening because the status of hasPermission when opening the screen is false, it is still querying the authorization request.

I'm having the same problem and I'm trying to solve for the permission to be accessed when the app starts.

kashandanish, 2023-05-16:
I faced the same issue and resolved it by checking the permission status before asking permission.

```typescript
if (await Camera.getCameraPermissionStatus() !== 'authorized') {
    if (await Camera.requestCameraPermission() === 'denied') {
        setSettingsModalOpen(true)
    }
}
```

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

closed 2023-09-30 by mrousavy