margelo/react-native-vision-camera #1154: ❓ proper way to handle afterimage when re-enabling camera component

jpike88, 2022-07-20:
### Question

I have my camera component using `@react-navigation/native-stack`. When the recording is stopped, another view is pushed on the stack. The camera component isActive is then set to false using the `useIsFocused` hook. If I pop that new view and come back to the camera component, there's a short delay where the last image shows before the camera 'resets' and starts showing the current camera stream.

What's the recommended/official way of avoiding this without doing something ugly (i.e. destroying camera component, or using some sort of obscuring view with a timeout?)

### VisionCamera Version

2.14.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-07-20:
Hey, I'm guessing that's just the delay of the camera - can't really do much about that. Maybe your screen is frozen (react-freeze)?

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

closed 2023-09-30 by mrousavy