margelo/react-native-vision-camera #1157: takephoto too slow and takeSnapshot bad quality❓

wilav-dev, 2022-07-21:
### Question

I want to take document photos and selfies on Android to validate identity. I'm combining this lib with some ocr and face frame processors.
If I use takePhoto, it has a delay and users can move the camera before capture is done. If I use takeSnapshot, the resolution is not the best to analyze the image later in the backend.

My alternative is to open an external camera and let users to take any picture without validations from the frontend, but want to use this library if possible, for a better UX.

RN 0.67.4

### What I tried

camera.current.takePhoto() but has a delay and image is taken after that delay
camera.current.takeSnapshot() but resolution is not good enough

### VisionCamera Version

2.13.5

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

jonxssc, 2022-10-31:
+1 same problem, couldnt find a solution to take photos instant like for example snapchat camera

versusvoid, 2022-12-06:
Local-patched library to use [ZERO_SHUTTER_LAG mode](https://developer.android.com/reference/androidx/camera/core/ImageCapture#CAPTURE_MODE_ZERO_SHUTTER_LAG()). It does reduce `takePhoto()` duration on average, but it still varies between 100ms - 600ms

jonxssc, 2022-12-06:
> but it still varies between 100ms - 600ms

I switched to expo camera package, it doesnt have the problem and is a bit faster, but takes also some time, maybe its because of what you said, i will test it

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

closed 2023-09-30 by mrousavy

seba9999, 2024-10-03:
Hi there, late to the party but I still experience this lag ( about 1 to 1.5 sec ) ...

@versusvoid Where did you used this patch ?

jkaufman, 2024-10-03:
@seba9999 That suggestion is out of date. See [this ticket](https://github.com/mrousavy/react-native-vision-camera/issues/3143) for further discussion. I'm not sure if further improvement is possible.