margelo/react-native-vision-camera #1158: App stops responding after some time while recording video

UmerQur3shi, 2022-07-21:
### What were you trying to do?

I started recording video and the app stopped responding after 3 minutes (buttons stopped responding, timer stopped but the camera view was working only). Tried on multiple android devices and got the same issue at different time duration. On emulator it was working fine.

### Reproduceable Code

```tsx
{isFocused? (
            <Camera
                style={{width,flex:1}}
                device={cameraState === 'back' ? backCamera : frontCamera}
                ref={camera}
                isActive={true}
                video={true}
                audio={true}
            />
        ):null}
```

### What happened instead?

Buttons and timer on screen freezes but camera view remains normal.

### Device

Xiaomi redmi note 10 pro, Xiaomi mi 9t

### VisionCamera Version

2.14.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

closed 2022-07-22 by UmerQur3shi

jkerb8, 2023-02-13:
I have received reports of users with Android devices experiencing similar issues. I was also able to reproduce the issue on a Pixel 6. Is there any solution at this time?

UmerQur3shi, 2023-02-13:
@jkerb8 I think I was using timer/interval to display the recording duration while recording video which was causing issue, do you have a timer/interval running in that component?

jkerb8, 2023-02-14:
@UmerQur3shi I happen to be using an interval to upload video clips in the background so that must be it. I'll create an alternate solution and report back with my results. Thank you!!

jkerb8, 2023-02-14:
For me, the issue wasn't due to the interval. I removed all intervals and timers and the issue still prevailed. I had a small FlatList that was overlayed over the video camera feed that showed the upload progress of the user's recently-recorded video clips. Even though each clip progress indicator was it's own component and did not re-render the component with the camera feed on progress updates, it was causing app freezes. For some reason, changing this view from a FlatList to a ScrollView fixed the problem for me. Hopefully this helps someone!