margelo/react-native-vision-camera #1159: 'react/bridging/CallbackWrapper.h' file not found

lvothnrv, 2022-07-21:
### What were you trying to do?

Hello here since I switched to react-native 0.69.2 I have this bug that drives me crazy I tested everything nothing happens I even created a react-native project project and just installed react-native-vision- camera I have the same problem..

[image]

### What happened instead?

I just can't start my application anymore.

### Device

iPhone 13

### VisionCamera Version

2.14.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

dmarkow, 2022-07-22:
Looks like a [fix was merged in react-native](https://github.com/facebook/react-native/pull/34214/files). Until it's released, adding this to my `post_install` block in `ios/Podfile` fixed it (I have no idea if the issue affects Android too, we're only targeting iOS right now).

```ruby
installer.target_installation_results.pod_target_installation_results.each do |pod_name, target_installation_result|
  target_installation_result.native_target.build_configurations.each do |config|
    # For third party modules who have React-bridging dependency to search correct headers
    config.build_settings['HEADER_SEARCH_PATHS'] ||= '$(inherited) '
    config.build_settings['HEADER_SEARCH_PATHS'] << '"$(PODS_ROOT)/Headers/Private/React-bridging/react/bridging" '
    config.build_settings['HEADER_SEARCH_PATHS'] << '"$(PODS_CONFIGURATION_BUILD_DIR)/React-bridging/react_bridging.framework/Headers" '
  end
end
```

lvothnrv, 2022-07-24:
> Looks like a [fix was merged in react-native](https://github.com/facebook/react-native/pull/34214/files). Until it's released, adding this to my `post_install` block in `ios/Podfile` fixed it (I [...]

Hi thank you for your comment but I don't know if it's me who misplaces the piece of code you gave me but it doesn't work can you show me your code please...

dmarkow, 2022-07-24:
@lvothnrv In you `ios/Podfile`, there should be a line that starts with `post_install do |installer|`. Put this inside that.

lvothnrv, 2022-07-24:
> @lvothnrv In you `ios/Podfile`, there should be a line that starts with `post_install do |installer|`. Put this inside that.

I did that but it doesn't work..

  post_install do |installer|
    installer.target_installation_results.pod_target_installation_results.each do |pod_name, target_installation_result|
      target_installation_result.native_target.build_configurations.each do |config|
        # For third party modules who have React-bridging dependency to search correct headers
        config.build_settings['HEADER_SEARCH_PATHS'] ||= '$(inherited) '
        config.build_settings['HEADER_SEARCH_PATHS'] << '"$(PODS_ROOT)/Headers/Private/React-bridging/react/bridging" '
        config.build_settings['HEADER_SEARCH_PATHS'] << '"$(PODS_CONFIGURATION_BUILD_DIR)/React-bridging/react_bridging.framework/Headers" '
      end
    end
    react_native_post_install(installer)
    __apply_Xcode_12_5_M1_post_install_workaround(installer)
  end

dmarkow, 2022-07-24:
@lvothnrv Did you run `pod install` again after adding this? If so, then I'm not sure what else to add...

lvothnrv, 2022-07-24:
> 

No I didn't.. thank you very much you really save my life again thank you my friend.

lvothnrv, 2022-07-24:
> @lvothnrv Did you run `pod install` again after adding this? If so, then I'm not sure what else to add...

No I didn't.. thank you very much you really save my life again thank you my friend.

babeone, 2022-11-06:
Upgrading to react-native 0.69.3 seems to resolve the problem

ziga-hvalec, 2022-11-23:
> Looks like a [fix was merged in react-native](https://github.com/facebook/react-native/pull/34214/files). Until it's released, adding this to my `post_install` block in `ios/Podfile` fixed it (I [...]

This worked on react native 0.70.6 and vision camera 2.15.2.

flaviocoppola, 2023-01-17:
I added this block to the Podfile, ran pod install, but when I try to generate the build it gives me "Build file not found"

Osamasomy, 2023-08-30:
facing same problem any solution?

sravis02, 2023-09-14:
also still got this

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

closed 2023-09-30 by mrousavy

sekhuat, 2024-01-14:
I also faced the same issue and I am using react-native-vision-camera ver 2.16.1. I solved the issue by adding `s.dependency "ReactCommon"` into the `node_modules/react-native-vision-camera/VisionCamera.podspec` at the last line as shown below.

```
Pod::Spec.new do |s|
  ...
  s.dependency "React-callinvoker"
  s.dependency "React"
  s.dependency "React-Core"
  s.dependency "ReactCommon"    <== Add this line
end
```

After made the changes, I execute `pod install` command then recompile the app again. Hope this will help those who using V2 of react-native-vision-camera.