margelo/react-native-vision-camera #1162: Shouldn't a Samsung S21 Ultra "supportParallelProcessing"❓

Lucianofc138, 2022-07-25:
### Question

I'm developing an app that needs to record video while sends frames to another devices through WebRTC (or something similar), so I was checking the supportsParallelProcessing attribute in the cameras of a few android phones. 

Considering what the [guide](https://mrousavy.com/react-native-vision-camera/docs/guides/devices#the-supportsparallelvideoprocessing-prop) say:

> On newer Android devices this value is always true.

I though it would work for both cameras on a Samsung S21 Ultra (2021 phone), but it only worked for the back camera.

- Is this a normal thing?
- What is considered a "newer" android phone? 
- Is there an easy was to check the INFO_SUPPORTED_HARDWARE_LEVEL of the cameras in a phone?

Here the results from trying in other phones:

| Phone (year)                                        | Front camera | Back camera |
|---------------------------------------------------|--------------------|-------------------|
| Samsung S21 Ultra (2021)                  |        false        |        true        |
| Huawei Mate 20 Pro (2018)                 |        false        |        true        |
| Motorola moto g7 power 64/4(2019)    |        true         |        true        |
| Xiaomi Redmi 9 (2019)                        |        true          |        true       |
| Huawei P30 lite  (2020)                        |        false        |        false      |
| Samsung Galaxy A21 (2020)               |        false        |        false       |

As you can see, both Xiaomi Redmi 9 and Motorola moto g7, suppport parallel processing, which kinda surprised me since the Samsung S21 is much newer and high-end.

### VisionCamera Version

2.13.5

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-07-26:
Oh that's very interesting. It really sucks that Android doesn't allow parallel video processing for every phone, while every iPhone since the iPhone 5 (I think) has that.

Here's where I check the supported hardware level: https://github.com/mrousavy/react-native-vision-camera/blob/f289f91773f89b520710d4632aabe8c50a4f57b0/android/src/main/java/com/mrousavy/camera/CameraViewModule.kt#L188

Lucianofc138, 2022-07-26:
Yes, also I would have thought the hardware level would have linked to the phone, instead of one for each camera device, but apparently that's not the case.

Also about Android, yeah, that's really frustrating. I think it could be because those cameras need some extra customization with the Camera2 API instead of CameraX, since those phone should be able to process and record at the same time, because it's done by Instagram with filter, but I don't know I'm  just guessing.

Anyway, **could you tell me which Android phone was used to test the library?** since I would like to have some ideas on which phone I could use for the app. Thanks for your help!

mrousavy, 2022-07-26:
I used a Huawai P10 to build VisionCamera.

erayoezmue, 2022-07-26:
Had the same problem with Pixel 6 on Android 12. Removing the Preview of the image allowed to record video and analyze at the same time. For preview, video and analyzing at the same time the level "LEVEL_3 (or better) camera device." is needed, as stated in the documentation. 

https://developer.android.com/training/camerax/architecture

Lucianofc138, 2022-07-26:
@mrousavy Thanks!
@erayoezmue Unfortunately I need to do the 3 at the same time for the app I'm working at, but that's still nice to know. Thanks!

erayoezmue, 2022-07-27:
@Lucianofc138, well me too... Just tried it on a Redmi Note 9 - also didn't work on that one. Downloaded Camera2 API Checker at indeed my Pixel 6 and Redmi Note 9 are only Level "Full" not "LEVEL_3". So you can forget it for many of new android devices also... 

Maybe we should look for workarounds for Android? Like grabbing the video screencast of the view? Doing whole screencast also won't work for most Android devices, as on newer versions the user would get asked every time you start the recording,...

erayoezmue, 2022-07-27:
@Lucianofc138: here a list of some Camera2API Reports https://www.airbeat.com/camera2/report_list

Lucianofc138, 2022-07-27:
@erayoezmue ooohh yeah, some of those i put in the list above could be just "Full", since I just check the attibute supportsParallelProcessing for each device. However, I think is not just about being new or not, since the Motorola moto g7 power (2019) has "LEVEL_3" (I checked it with Camera2 Probe). I think it could be about how much customization each company put over the basic android, because Motorola runs the most basic Stock Android, which is the most basic version without any customization. I'm not sure tho, just an idea.

I think the same about searching a workaround. I was thinking about making a long running frame processor, that could receive the ImageProxies one by one, and make them into a video, but I'm not sure if that's possible yet... Also you would be obligated to process any other frame processor at the same frame-rate that you want to record.

Thanks for the list of reports!

maqeelqureshi, 2022-11-10:
Same Back camera issue on following devices,

OnePlus6T (A6013)
Samsung Galaxy A22 5G

mrousavy, 2023-09-30:
Hey  this has been fixed in V3 as we no longer have that prop thanks to a full rewrite! 🥳

closed 2023-09-30 by mrousavy