margelo/react-native-vision-camera #1163: react-native-vision-camera:downloadDoubleConversion🐛

UsamaIkraam0099, 2022-07-26:
### What were you trying to do?

I try to build the android project through "npx react-native run-android" but Im unable to build the project.
[image]

### What happened instead?

Build Failed.

### Relevant log output

```shell
> Task :react-native-vision-camera:downloadDoubleConversion
Download https://github.com/google/double-conversion/archive/v1.1.6.tar.gz

> Task :react-native-vision-camera:downloadDoubleConversion FAILED

Deprecated Gradle features were used in this build, making it incompatible with Gradle 8.0.

You can use '--warning-mode all' to show the individual deprecation warnings and determine if they come from your own scripts or plugins.

See https://docs.gradle.org/7.2/userguide/command_line_interface.html#sec:command_line_warnings
420 actionable tasks: 3 executed, 417 up-to-date

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-vision-camera:downloadDoubleConversion'.
> de.undercouch.gradle.tasks.download.org.apache.http.conn.ConnectTimeoutException: Connect to codeload.github.com:443 [codeload.github.com/13.127.152.42] failed: connect timed out
```

### Device

LG G6 (android 8)

### VisionCamera Version

2.9.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

UsamaIkraam0099, 2022-07-26:
@mrousavy 

Any suggestion would be appreciated.

Thanks

JBuijgers, 2022-07-26:
Are you able to ping codeload.github.com from the host (e.g. your PC or a CI/CD server) that is performing the build?

UsamaIkraam0099, 2022-07-27:
Hi @JBuijgers 

Thanks for your response. I solved that.

closed 2022-07-27 by UsamaIkraam0099

jhsonmac, 2022-09-08:
> Oi@JBuijgers
> 
> Obrigado pela sua resposta. Eu resolvi isso.

hello, how did you solve this problem?

MohamedAhmed122, 2022-09-12:
@UsamaIkraam0099  how did you solve it? I have the same issue

Applizone, 2022-10-28:
I've though chain of error to run react native after installing this module. I got ```downloadDoubleConversion``` at first. What did I do is I download manually by click the link ```Download https://github.com/google/double-conversion/archive/v1.1.6.tar.gz``` from console and paste it in browser URL. Then it download to local folder. After that, I copy and paste it on ```\node_modules\react-native-vision-camera\android\build\downloads``` and run the ```npx react-native run-android```.

Suprisingly, one after another comes the download error. The ```folly-2021.06.28.00.tar.gz``` and ```glog-0.3.5.tar.gz``` and finally the huge problem [#1283](https://github.com/mrousavy/react-native-vision-camera/issues/1283)

The problem is on react-native-reanimated version. Below are the step that valid with me from the issue on previous paragraph:

> delete ```node_modules```
> edit ```package.json```
> "react": "17.0.2",
> "react-native": "0.68.4",
> "react-native-reanimated": "2.10.0", #not using "^2.10.0"
> "react-native-vision-camera": "^2.15.1",
> 
> yarn cache clean --force
> yarn install
> cd android
> ./gradlew clean
> yarn android
> npx react-native run-android

Part of the chain error when downloading with something like ```java.net.UnknownHostException: No such host is known (github.com)```
Task :react-native-vision-camera:downloadDoubleConversion
Download https://github.com/google/double-conversion/archive/v1.1.6.tar.gz

Task :react-native-reanimated:downloadGlog
Download https://github.com/google/glog/archive/v0.3.5.tar.gz

Task :react-native-vision-camera:downloadFolly
Download https://github.com/facebook/folly/archive/v2021.06.28.00.tar.gz

See which part of the problem occur; react-native-vision-camera or react-native-reanimated when you finally run. Download manually and paste it in ```\node_modules\react-native-vision-camera\android\build\downloads``` or ```\node_modules\react-native-reanimated\android\build\downloads```

Finally, my app can run again. Wow. What an effort.