margelo/react-native-vision-camera #1164: 🐛 react-native 0.69.3 android build crash

BartoszDombrow, 2022-07-26:
### What were you trying to do?

I want to build an android application, but I got this error
I am working on RN 0.69.3

### Reproduceable Code

```tsx
FAILURE: Build failed with an exception.

* Where:
Build file 'C:\Users\48791\OneDrive\Pulpit\sewio-mobile\node_modules\react-native-vision-camera\android\build.gradle' line: 231

* What went wrong:
A problem occurred evaluating project ':react-native-vision-camera'.
> Expected directory 'C:\Users\48791\OneDrive\Pulpit\sewio-mobile\node_modules/react-native/android' to contain exactly one file, however, it contains more than one file.
```

### What happened instead?

App can not build and npx react-native run-android crash. If i disable frame processors in android/gradle.propeties app builds correctly.

### Device

Samsung S8

### VisionCamera Version

2.14.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

JulianErnest, 2022-07-28:
Same issue on 

React Native - 0.69.1
React Native Vision Camera - 2.14.0
React Native Reanimated - 2.9.1
Vision Camera Code Scanner - 2.14.1

thangovn, 2022-07-28:
same issue

JulianErnest, 2022-07-29:
Build was successful after running npx patch-package react-native-vision-camera
Must have the patch installed though, see nobane's answer above. 
Weird cause I was running the 2.14.0 one which should've had the fix.

gtzippy, 2022-07-29:
Same issue for me: 
    "react": "18.0.0",
    "react-native": "0.69.3",
    "react-native-reanimated": "^2.9.1",
    "react-native-vision-camera": "^2.14.0"
    
 I ran the npx patch call and nothing changed

jbrodriguez, 2022-07-29:
should really think about merging this one https://github.com/mrousavy/react-native-vision-camera/pull/1116/files

thangovn, 2022-07-30:
add disableFrameProcessors=true in gradle.properties will work on android

spsaucier, 2022-08-01:
We are still using frame processors, so we can't turn that flag off.

onesmartguy, 2022-08-01:
@spsaucier did you find anything on this? I'm stuck with the same issue, also need to keep frame processors on.

spsaucier, 2022-08-01:
I haven’t - hoping the relevant PR will be merged very soon (
https://github.com/mrousavy/react-native-vision-camera/pull/1116).

TiKevin83, 2022-08-30:
This should be fixed as of the latest release, #1116 was superseded by #1186 and merged in the latest release

closed 2023-08-18 by mrousavy