margelo/react-native-vision-camera #1166: 🐛 'react/bridging/CallbackWrapper.h' file not found (iOS)

mocanSergiu666, 2022-07-27:
### What were you trying to do?

I installed the package and after this the project doesn't build on iOS and it throws this error. On Android everything works as expected.

### What happened instead?

The app crashes on iOS platform with the next error: 'react/bridging/CallbackWrapper.h' file not found.

### Device

iPhone 13 Pro (iOS 15.5)

### VisionCamera Version

2.14.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

deolive, 2022-07-27:
Probably You use `react-native 0.69.0 - 0.69.2`
This error fixed in `react-native 0.69.3`, try
`
npx react-native upgrade
`

mocanSergiu666, 2022-07-27:
Yes, i am using react-native 0.69.2. I'll do the upgrade and will tell you if it fixed my issue. Thanks for the reply.

mocanSergiu666, 2022-07-27:
@deolive thanks, your hint fixed the issue.

closed 2022-07-27 by mocanSergiu666