margelo/react-native-vision-camera #1170: Problem when I try call frameProcessor from useScanBarcodes 🐛

huandrey, 2022-07-28:
### What were you trying to do?

I`m working in a project that i need use the feature of the qrcode.
When I use the lib react-native-vision-camera in the version 2.13.0, own this problem:
"Frame processors are not enable"
[image]
So, when use the lib in the version 2.14.0, I own this problem:
[image]
[image]
[image]

I already view this issues and still the same problem
- https://github.com/mrousavy/react-native-vision-camera/issues/626
- https://github.com/mrousavy/react-native-vision-camera/issues/560
- https://github.com/mrousavy/react-native-vision-camera/issues/957
- https://github.com/mrousavy/react-native-vision-camera/issues/1012

### What happened instead?

"Frame processors are not enable"

### Device

POCO F3 XIAOMI

### VisionCamera Version

2.14.0 e 2.13.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-07-29:
Which REA and RN version?

huandrey, 2022-07-29:
> Which REA and RN version?
[image]

mrousavy, 2022-07-29:
Please share your Reanimated version.

huandrey, 2022-07-29:
> Please share your Reanimated version.
@mrousavy 
[image]

mrousavy, 2022-08-01:
Did you read the error?

"Frame Processors are disabled because REA v2 does not exist"

closed 2022-08-01 by mrousavy