margelo/react-native-vision-camera #1173: 🐛 Frame Processor threw an error: Exception in HostObject::get(propName:width):

chornos13, 2022-07-30:
### What were you trying to do?

So I need to read what frame width/height running currently

[image]

it's work fine in debug but in release it crash

### Reproduceable Code

```tsx
const updateFrameSize = (width: number, height: number) => {
    refScanArea.current = {
      x: width * 0.06,
      y: height * 0.2,
      width: width * 0.45,
      height: height * 0.5,
    }

    setFrameWidth(width)
    setFrameHeight(height)
  }

  const frameProcessor = useFrameProcessor(frame => {
    'worklet'
    const barcodes = scanBarcodes(frame, [BarcodeFormat.ALL_FORMATS], {
      checkInverted: true,
    })
    runOnJS(updateFrameSize)(frame.width, frame.height)

    if (barcodes.length === 0) {
      return
    }
  }, [])
```

### What happened instead?

I can't read the frame.width or frame.height

### Relevant log output

```shell
Type	Time	PID	Tag	Message
	08:43:54.170	7467	VisionCamera	Frame Processor threw an error: Exception in HostObject::get(propName:width): java.lang.NoSuchMethodError: no non-static method "Landroidx/camera/core/k2;.getWidth()I"
	08:43:54.172	7467	ReactNativeJS	Frame Processor threw an error: Exception in HostObject::get(propName:width): java.lang.NoSuchMethodError: no non-static method "Landroidx/camera/core/k2;.getWidth()I"
	08:43:54.175	2584	ChimeraSrvcProxy	NullBinder for android.net.action.RECOMMEND_NETWORKS triggering remote TransactionTooLargeException due to Service without Chimera impl
	08:43:54.179	7467	unknown:ReactNative	console.error: Frame Processor threw an error: Exception in HostObject::get(propName:width): java.lang.NoSuchMethodError: no non-static method "Landroidx/camera/core/k2;.getWidth()I", stack:
Error@-1
t@74:250
o@72:330
construct@-1
<unknown>@71:437
s@71:637
f@71:1781

	08:43:54.250	7467	skia	onFlyCompress
	08:43:54.540	1793	GnssHAL_GnssInterface	gnssSvStatusCb: a: input svInfo.flags is 8
	08:43:54.589	1793	GnssHAL_GnssInterface	gnssSvStatusCb: b: input svInfo.flags is 8
	08:43:54.718	1937	netmgr	Failed to open QEMU pipe 'qemud:network': Invalid argument
	08:43:54.786	3082	WorkerManager	dispose()

	08:43:54.788	3082	A	(REDACTED) Queue length for executor %s is now %d. Perhaps some tasks are too long, or the pool is too small.

	08:43:54.951	1942	wifi_forwarder	RemoteConnection failed to initialize: RemoteConnection failed to open pipe
	08:43:55.536	1793	GnssHAL_GnssInterface	gnssSvStatusCb: a: input svInfo.flags is 8
	08:43:55.536	1793	GnssHAL_GnssInterface	gnssSvStatusCb: b: input svInfo.flags is 8
	08:43:56.378	7467	VisionCamera	Frame Processor threw an error: Exception in HostObject::get(propName:width): java.lang.NoSuchMethodError: no non-static method "Landroidx/camera/core/k2;.getWidth()I"
	08:43:56.384	7467	ReactNativeJS	Frame Processor threw an error: Exception in HostObject::get(propName:width): java.lang.NoSuchMethodError: no non-static method "Landroidx/camera/core/k2;.getWidth()I"
	08:43:56.417	7467	unknown:ReactNative	console.error: Frame Processor threw an error: Exception in HostObject::get(propName:width): java.lang.NoSuchMethodError: no non-static method "Landroidx/camera/core/k2;.getWidth()I", stack:
Error@-1
t@74:250
o@72:330
construct@-1
<unknown>@71:437
s@71:637
f@71:1781
```

### Device

Emulator

### VisionCamera Version

Pixel 3a API 29

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

chornos13, 2022-07-30:
Ok I found the solution, just using progruad for that class, thanks!

closed 2022-07-30 by chornos13

SharifCoinMENA, 2022-10-24:
can you share your proguard please? I am facing same issue

ashrazdev, 2022-11-14:
can you share your proguard please? I am also facing same issue

chornos13, 2022-11-14:
`
-keep class androidx.camera.core.** { *; }
`

just add this 😄