margelo/react-native-vision-camera #1174: ❓ In example can I trigger andorid plugin from react?

gtzippy, 2022-07-30:
### Question

I have the example up and running on my android phone. The frameprocessor is working and logging the stuff from the react-native code, but I am wondering if I can trigger the java in the android ExampleFrameProcessorPlugin? I already have a robust android app that does what I want with an image, so if I can trigger that code by using the frameprocessor that would be amazing. Thanks for any guidance!

### What I tried

I have tried logging to the console in java and touching a file on the phone in the ExampleFrameProcessorPlugin but it I cannot find any evidence of the log or the file being updated on my phone.

### VisionCamera Version

example project

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

gtzippy, 2022-07-30:
I retract this question. I just did a reclone and setup of the example project and now it all works. I was very meticulous throughout this process, so I guess that I messed something up the first time. Now I am able to trigger things in the Android code.

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

closed 2023-09-30 by mrousavy