margelo/react-native-vision-camera #1177: ❓ Phone auto-locks while recording is occuring

jpike88, 2022-08-03:
### Question

My iPhone 11 pro is set to 30 seconds auto-lock. When recording, the auto-lock kicks in. I can't find anything in this plugin to help regulate this. 

Shouldn't this plugin keep the device awake while video is recording, or at least expose an api to do that?

### VisionCamera Version

2.14.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

guthriet1230, 2022-08-17:
I'm also experiencing this issue. I'll note that there are packages to rectify this (ie react-native-keep-awake) so i think there is a way, but those packages arent downloading properly and i believe are deprecated. 

need help too!

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

closed 2023-09-30 by mrousavy

BboyKeen, 2024-01-17:
Hey!

Thanks for this awesome package.
I'm not sure it's fixed. I can see my iPhone going to sleep using v3 while recording a video.

I think I can use something like isIdleTimerDisabled equivalent in JS as a workaround.
 
I use React Native without Expo

mrousavy, 2024-01-18:
Hey - thank you.

I believe all apps with active `AVCaptureSession`/`AVCapturePreviewLayer` instances do not go to sleep, this is the first time I heard about this - does this happen in the example app as well?

BboyKeen, 2024-01-18:
Yes I can reproduce in the example app using the main branch. For the record I'm using an iPhone 13 (iOS 17.2).

However to reproduce I have to comment this part in CaptureButton to make sure the recording continues without holding down the button (same case in my app)

```
case State.END:
case State.FAILED:
case State.CANCELLED: {
  // exit "recording mode"
  try {
    if (pressDownDate.current == null) throw new Error('PressDownDate ref .current was null!')
    const now = new Date()
    const diff = now.getTime() - pressDownDate.current.getTime()
    pressDownDate.current = undefined
    if (diff < START_RECORDING_DELAY) {
      // user has released the button within 200ms, so his intention is to take a single picture.
      await takePhoto()
    } else {
      // user has held the button for more than 200ms, so he has been recording this entire time.
      await stopRecording()
    }
  } finally {
    setTimeout(() => {
      isPressingButton.value = false
      setIsPressingButton(false)
    }, 500)
  }
  return
}
```

Let me know if you want me to dig further

BboyKeen, 2024-01-18:
Looks like Amazon recommends setting the idleTimer off on their example app for broadcasting too. 

https://docs.aws.amazon.com/fr_fr/ivs/latest/LowLatencyUserGuide/broadcast-ios-getting-started.html#broadcast-ios-disable-idle-timer

mrousavy, 2024-01-19:
Created a PR to fix this: #2421