margelo/react-native-vision-camera #1178: 🐛 Starting camera with frame processor is much slower than just starting a camera

ex3ndr, 2022-08-03:
### What were you trying to do?

Open a screen with a camera that has a frame processor (barcode scanner). I tried to delay setting the frame processor, but it became slower; setting `frameProcessorFps` doesn't affect slowness.

### Reproduceable Code

```tsx
import * as React from 'react';
import { ActivityIndicator, View } from 'react-native';
import { runOnJS } from 'react-native-reanimated';
import { Camera, useFrameProcessor } from 'react-native-vision-camera';
import { BarcodeFormat, scanBarcodes, Barcode } from 'vision-camera-code-scanner';
import { useFastBackCamera } from '../../utils/useFastBackCamera';

export const CameraScannerComponent = React.memo((props: { handler: (barcode: Barcode) => void }) => {

    // Resolve Device
    const device = useFastBackCamera();

    // Scanner
    const frameProcessor = useFrameProcessor(frame => {
        'worklet';
        const detectedBarcodes = scanBarcodes(frame, [BarcodeFormat.QR_CODE], { checkInverted: true });
        if (detectedBarcodes.length > 0) {
            runOnJS(props.handler)(detectedBarcodes[0]);
        }
    }, [props.handler]);

    const [inited, setInited] = React.useState(false);
    const onInitedCallback = React.useCallback(() => {
        setInited(true)
    }, []);

    return (
        <View style={{ flexGrow: 1, flexBasis: 0, alignSelf: 'stretch', backgroundColor: 'black', alignItems: 'center', justifyContent: 'center' }}>
            {!device && (<ActivityIndicator />)}
            {!!device && (
                <Camera
                    style={{ flexGrow: 1, flexBasis: 0, alignSelf: 'stretch' }}
                    device={device}
                    isActive={true}
                    onInitialized={onInitedCallback}
                    frameProcessor={frameProcessor}
                />
            )}
        </View>
    );
});
```

### What happened instead?

First frames of a camera a frozen for a second or so.

### Device

iPhone 13 Pro Max

### VisionCamera Version

2.14.0

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

ex3ndr, 2022-08-03:
Actually when i try to set `frameProcessor` later -  camera seems to be just restarted.

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

closed 2023-09-30 by mrousavy