margelo/react-native-vision-camera #1179: 🐛 [device/parallel-video-processing-not-supported] It seems I can't use frameProcessors for some Samsung devices

keShraa, 2022-08-03:
### What were you trying to do?

Hi guys,

As mentioned in the title, I can't use **frameProcessors** for some Samsung devices (Samsung A51, A52, A53...).
I'm using this awesome library `react-native-vision-camera` combined with `vision-camera-code-scanner` to scan barcodes for my application.

It works perfectly with my others devices (Google Pixel 5A, iPhone X...) but with those Samsung devices, the camera component doesn't even initialize and I have this error message:

> [device/parallel-video-processing-not-supported] The given **LEGACY Camera device does not support parallel video processing** (`video={true}` + `frameProcessor={...}`). Disable either `video` or `frameProcessor`. To find out if a device supports parallel video processing, check the `supportsParallelVideoProcessing` property on the CameraDevice. See https://mrousavy.github.io/react-native-vision-camera/docs/guides/devices#the-supportsparallelvideoprocessing-prop for more information.]

I understand that the device **doesn't support parallel video processing** `device.supportsParallelVideoProcessing = false` but I should be able to _disable the video_ and _activate frame processors_. (Cf documentation)
> Camera devices provide the [supportsParallelVideoProcessing property](https://mrousavy.com/react-native-vision-camera/docs/api/interfaces/CameraDevice#supportsparallelvideoprocessing) which determines whether the device supports using Video Recordings (video={true}) and Frame Processors (frameProcessor={...}) at the same time.
If this property is false, you can either enable video, or add a frameProcessor, but not both.

### Reproduceable Code

```tsx
import { Camera, useCameraDevices, sortFormats } from 'react-native-vision-camera';
import { useScanBarcodes, BarcodeFormat } from 'vision-camera-code-scanner';

export const CameraComponent = props => {
    const [frameProcessor, barcodes] = useScanBarcodes([...]);

    // Camera configuration (ref, isFocused, isActive, device, format...)
    [...]

    if (device == null || permission === 'denied') return [...]

    return (
        <Camera
            ref={camera}
            device={device}
            isActive={isActive}

            // Disable video and should be able to use frameProcessor
            photo={true}
            video={false}
            audio={false}
            frameProcessor={frameProcessor}
            frameProcessorFps={frameProcessorFps || 1}

            flash={supportsFlash ? flash : 'off'}
            lowLightBoost={device?.lowLightBoost}
            fps={fps}
            {...(preset
                ? { preset }
                : {
                      format: format,
                      hdr: format?.supportsPhotoHDR,
                  })}
            onInitialized={onInitialized}
            onError={onError}
            style={StyleSheet.absoluteFill}
        />
    )
}
```

### What happened instead?

This gives me a black screen if the device's `supportsParallelVideoProcessing` property if `false`. I thought that I would fix this by disabling the video, but nope 😅 ! Can you help me with this ?
Tell me if you need more information. 🙏🏻 

### Relevant log output

```shell
[device/parallel-video-processing-not-supported] The given **LEGACY Camera device does not support parallel video processing** (`video={true}` + `frameProcessor={...}`). Disable either `video` or `frameProcessor`. To find out if a device supports parallel video processing, check the `supportsParallelVideoProcessing` property on the CameraDevice. See https://mrousavy.github.io/react-native-vision-camera/docs/guides/devices#the-supportsparallelvideoprocessing-prop for more information.]
```

### Device

Samsung Galaxy A53 5G (SM-A536B/DS)

### VisionCamera Version

2.14.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

joaonew, 2022-11-29:
Did you find any workaround for this issue?

keShraa, 2023-01-17:
No unfortunately, sorry for late answer 😅

mrousavy, 2023-09-30:
This has been fixed with the full rewrite in VisionCamera V3 as we no longer need the parallel video processing prop 🥳

closed 2023-09-30 by mrousavy