margelo/react-native-vision-camera #1181: 🐛

peterzadori, 2022-08-04:
### What were you trying to do?

Hello

I'm having issues using callbacks as arguments for my exported function. 

I would like to use something like this in my frame processor:

`__detectRectangles(frame, (result, error) => {
                // callback
            });`

However, I don't know how to cast the arguments into `RCTResponseSenderBlock` or how to call the callback from my Swfit code.

### What happened instead?

Can't call provided callback

### Device

iOS

### VisionCamera Version

2.13.3

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-08-05:
Everything should be synchronous, callbacks will introduce async callback hell. I think it would be possible to support them though if really needed

eddy-lau, 2022-08-06:
You can do it in Objective-C. I tried with React Native 0.69 and OK.

```obj-c
@implementation YourProcessor

+ (void)asyncProcessFrame:(Frame * _Nonnull)frame args:(NSArray * _Nonnull)args callback:(void (^ _Nonnull)(NSDictionary<NSString *, id> * _Nullable, NSError * _Nullable))callback 
{
   /* async operation */
   /* ... */
   callback(dict, [NSNull null]);
}

static inline id processFrame(Frame* frame, NSArray* args) {
    
    RCTResponseSenderBlock responseSender = args[1];
    
    [YourProcessor asyncProcessFrame:frame args:args callback:^(NSDictionary<NSString *,id> * _Nullable result, NSError * _Nullable error) {
        
        responseSender(@[
            error != nil ? error : [NSNull null],
            result != nil ? result : [NSNull null]
        ]);
        
    }];
    
    return @[];
}

VISION_EXPORT_FRAME_PROCESSOR(processFrame)

@end

```

peterzadori, 2022-08-06:
> You can do it in Objective-C. I tried with React Native 0.69 and OK.
>
> ```objc
> @implementation YourProcessor
>
> + (void)asyncProcessFrame:(Frame * _Nonnull)frame args:(NSArray * _Nonnull)args [...]

Thanks, I'm going to try this and report back

mrousavy, 2023-09-30:
Hey - callbacks don't really make sense here as FPs are synchronous. You can't really have long running tasks here, this will just stall the Camera

closed 2023-09-30 by mrousavy