margelo/react-native-vision-camera #1184: 🐛 Expected directory '.../node_modules/react-native/android' to contain exactly one file, however, it contains more than one file.

criss02-cs, 2022-08-05:
### What were you trying to do?

When I try to build my application he gaves me this error.
React-native: 0.69.3

### Reproduceable Code

```tsx
* Where:
Build file 'C:\Users\CristianoA\Desktop\LoginReactTypescript\node_modules\react-native-vision-camera\android\build.gradle' line: 231

* What went wrong:
A problem occurred evaluating project ':react-native-vision-camera'.
> Expected directory 'C:\Users\CristianoA\Desktop\LoginReactTypescript\node_modules/react-native/android' to contain exactly one file, however, it contains more than one file.
```

### What happened instead?

It doesn't buil my application, when i remove this package application works perfectly

### Relevant log output

```shell
* Where:
Build file 'C:\Users\CristianoA\Desktop\LoginReactTypescript\node_modules\react-native-vision-camera\android\build.gradle' line: 231

* What went wrong:
A problem occurred evaluating project ':react-native-vision-camera'.
> Expected directory 'C:\Users\CristianoA\Desktop\LoginReactTypescript\node_modules/react-native/android' to contain exactly one file, however, it contains more than one file.
```

### Device

Realme 9 4G

### VisionCamera Version

2.14.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

spsaucier, 2022-08-08:
See https://github.com/mrousavy/react-native-vision-camera/pull/1116

NastuzziSamy, 2022-08-16:
Hi, I had to apply this patch to make it work with 2.14.1

```sh
cd <project-directory>
mkdir patches
mv ~/Downloads/react-native-vision-camera+2.14.1.txt patches/react-native-vision-camera+2.14.1.patch
npm i patch-package
npx patch-package
```

Hope this helps you :)

[react-native-vision-camera+2.14.1.txt](https://github.com/mrousavy/react-native-vision-camera/files/9348629/react-native-vision-camera%2B2.14.1.txt)

anderpaz, 2022-08-24:
> Hi, I had to apply this patch to make it work with 2.14.1
> 
> ```shell
> cd <project-directory>
> mkdir patches
> mv ~/Downloads/react-native-vision-camera+2.14.1.txt patches/react-native-vision-camera+2.14.1.patch
> npm i patch-package
> npx patch-package
> ```
> 
> Hope this helps you :)
> 
> Based on the PR merged: https://github.com/mrousavy/react-native-vision-camera/pull/1144/files
> 
> [react-native-vision-camera+2.14.1.txt](https://github.com/mrousavy/react-native-vision-camera/files/9348629/react-native-vision-camera%2B2.14.1.txt)

We have a forecast for the release of this release with the correction?

francois-spectre, 2022-10-12:
for me it works now with "^2.14.1"

princefishthrower, 2022-10-19:
@anderpaz - What version of react native are you on? I'm desperately trying to get this library installed. I've upgraded to react native 0.70.0, maybe that is the issue?

closed 2023-08-18 by mrousavy