margelo/react-native-vision-camera #1185: 🐛 Android build failure

mocanSergiu666, 2022-08-06:
### What were you trying to do?

I want to build the Android project. Because I am using react-native-reanimated in combination with react-native-vision-camera, it crashes.

"react": "18.0.0",
"react-native": "0.69.3",
"react-native-reanimated": "^2.9.1",
"react-native-vision-camera": "^2.14.0"

### What happened instead?

Build crashes with the error: 

* What went wrong:
A problem occurred evaluating project ':react-native-vision-camera'.
> Expected directory '.../node_modules/react-native/android' to contain exactly one file, however, it contains more than one file.

### Device

Pixel 5 API 30 (Emulator)

### VisionCamera Version

2.14.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

lfrallon, 2022-08-06:
This might also be related to this issue #1118.

spsaucier, 2022-08-08:
See https://github.com/mrousavy/react-native-vision-camera/pull/1116

rkostrab, 2022-08-11:
I was also upset while failing builds... Then I noticed a new rn-vision-camera release and tried it before throwing my laptop through the window...

No need for any patches from StackOverflow or other github issues. Remove them.

if you have:
`"react-native": "0.69.3",`
update your libs to versions:
`"react-native-vision-camera": "^2.14.1",` (".1" supports RN 0.69)
`"vision-camera-code-scanner": "^0.2.0"`

Make sure you did every step of the installation instructions:
- using JDK11 (org.gradle.java.home=/Library/Java/JavaVirtualMachines/zulu-11.jdk/Contents/Home in gradle.properties)
- babel.config.js:
```
[
  'react-native-reanimated/plugin',
  {
    globals: ['__scanCodes'],
  },
]
```
put into project/build.gradle:
```
buildscript {
    ext {
        ...
        VisionCameraCodeScanner_targetSdkVersion = 31
        VisionCameraCodeScanner_compileSdkVersion = 31
    }
}
```

**Then rm -rf node modules, yarn install, gradlew clean, get rid of garbage of the kitchen, etc,...**

`npx react-native run-android` and pray...

mocanSergiu666, 2022-08-15:
[rkostrab](https://github.com/rkostrab) I updated the package and now the build works. Thanks.

closed 2022-08-15 by mocanSergiu666