margelo/react-native-vision-camera #1188: 🐛  [ANDROID] Build error

hminhman, 2022-08-12:
### What were you trying to do?

I use react native 0.69.4 with new architecture and hermes.
Before I install react-native-vision-camera and vision-camera-code-scanner, android build success. But after install them, android throw an error when build: [**fcntl(): Bad file descriptor**]. This issue mention path: {project-path}/android/app/build/react-ndk/exported/x86/libfb.so . This path is defined in react-native-safe-area-context/android/src/main/jni/Android.mk (LOCAL_SRC_FILES), but everything is ok util I install camera vision. This is how they define path: `LOCAL_SRC_FILES := $(wildcard $(LOCAL_PATH)/*.cpp) $(wildcard $(GENERATED_SRC_DIR)/codegen/jni/*.cpp)`. Please suggest me the way to fix this problem.
Thank you very much.

### Reproduceable Code

```tsx
npx react-native init MyApp --template react-native-template-typescript

- turn new architecture and hermes on

yarn add react-native-safe-area-context react-native-vision-camera vision-camera-code-scanner

npx react-native run-android (or build with android studio)
```

### What happened instead?

Build error
[image]

### Relevant log output

```shell
Build command failed.
Error while executing process {user}/Library/Android/sdk/ndk/21.4.7075529/ndk-build with arguments {NDK_PROJECT_PATH=null APP_BUILD_SCRIPT={project}/node_modules/react-native-safe-area-context/android/src/main/jni/Android.mk APP_ABI=x86 NDK_ALL_ABIS=x86 NDK_DEBUG=1 NDK_OUT={project}/node_modules/react-native-safe-area-context/android/build/intermediates/cxx/Debug/3w1h2923/obj NDK_LIBS_OUT={project}/node_modules/react-native-safe-area-context/android/build/intermediates/cxx/Debug/3w1h2923/lib APP_CFLAGS+=-Wall APP_CFLAGS+=-Werror APP_CFLAGS+=-fexceptions APP_CFLAGS+=-frtti APP_CFLAGS+=-DWITH_INSPECTOR=1 APP_CPPFLAGS+=-std=c++17 APP_PLATFORM=android-21 APP_STL=c++_shared NDK_TOOLCHAIN_VERSION=clang GENERATED_SRC_DIR={project}/android/app/build/generated/source PROJECT_BUILD_DIR={project}/android/app/build REACT_ANDROID_DIR={project}/node_modules/react-native/ReactAndroid REACT_ANDROID_BUILD_DIR={project}/node_modules/react-native/ReactAndroid/build clean}
Android NDK: ERROR:{project}/node_modules/react-native-safe-area-context/android/src/main/jni/Android.mk:fb: LOCAL_SRC_FILES points to a missing file    
Android NDK: Check that {project}/android/app/build/react-ndk/exported/x86/libfb.so exists  or that its path is correct   

fcntl(): Bad file descriptor
{user}/Library/Android/sdk/ndk/21.4.7075529/build/core/prebuilt-library.mk:45: *** Android NDK: Aborting    .  Stop.
```

### Device

Pixel 5A api 31 (emulator, just build, not run yet)

### VisionCamera Version

2.14.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

NastuzziSamy, 2022-08-16:
Hi, fixed the issue here with a temporary patch

https://github.com/mrousavy/react-native-vision-camera/issues/1184#issuecomment-1216250548

hminhman, 2022-08-18:
Thanks @NastuzziSamy. I will try it soon and feedback to you.

hminhman, 2022-08-19:
I build andoid successfully with your change, @NastuzziSamy. Thank you and hope #1184 release soon.

closed 2022-08-19 by hminhman