margelo/react-native-vision-camera #1189: 🐛 Build unsuccessful

V00ster1, 2022-08-12:
### What were you trying to do?

I was trying to build an example app

### Reproduceable Code

```tsx
$ cd ios && bundle install
```

### What happened instead?

It crashed when trying to install json

### Relevant log output

```shell
$ cd ios && bundle install
Calling `DidYouMean::SPELL_CHECKERS.merge!(error_name => spell_checker)' has been deprecated. Please call `DidYouMean.correct_error(error_name, spell_checker)' instead.
Fetching gem metadata from https://rubygems.org/.......
Using rexml 3.2.5
Using CFPropertyList 3.0.4
Using concurrent-ruby 1.1.9
Using i18n 1.8.10
Using minitest 5.14.4
Using thread_safe 0.3.6
Using tzinfo 1.2.9
Using activesupport 5.2.6
Using public_suffix 4.0.6
Using addressable 2.8.0
Using httpclient 2.8.3
Fetching json 2.5.1
Installing json 2.5.1 with native extensions
Gem::Ext::BuildError: ERROR: Failed to build gem native extension.

current directory:
C:/tools/ruby31/lib/ruby/gems/3.1.0/gems/json-2.5.1/ext/json/ext/generator
C:/tools/ruby31/bin/ruby.exe -I C:/tools/ruby31/lib/ruby/3.1.0 -r
./siteconf20220812-3076-bumsjh.rb extconf.rb
creating Makefile

current directory:
C:/tools/ruby31/lib/ruby/gems/3.1.0/gems/json-2.5.1/ext/json/ext/generator
make DESTDIR\= clean

current directory:
C:/tools/ruby31/lib/ruby/gems/3.1.0/gems/json-2.5.1/ext/json/ext/generator
make DESTDIR\=
generating generator-x64-mingw-ucrt.def
compiling generator.c
process_begin: CreateProcess(NULL, gcc -I.
-I/C/tools/ruby31/include/ruby-3.1.0/x64-mingw-ucrt
-I/C/tools/ruby31/include/ruby-3.1.0/ruby/backward
-I/C/tools/ruby31/include/ruby-3.1.0 -I. -DJSON_GENERATOR
-D__USE_MINGW_ANSI_STDIO=1 -DFD_SETSIZE=2048 -D_WIN32_WINNT=0x0600
-D__MINGW_USE_VC2005_COMPAT -D_FILE_OFFSET_BITS=64 -O3 -fno-fast-math
-fstack-protector-strong -m64 -o generator.o -c generator.c, ...) failed.
make (e=2): The system cannot find the file specified.

make: *** [Makefile:246: generator.o] Error 2

make failed, exit code 2

Gem files will remain installed in
C:/tools/ruby31/lib/ruby/gems/3.1.0/gems/json-2.5.1 for inspection.
Results logged to
C:/tools/ruby31/lib/ruby/gems/3.1.0/extensions/x64-mingw-ucrt/3.1.0/json-2.5.1/gem_make.out

  C:/tools/ruby31/lib/ruby/3.1.0/rubygems/ext/builder.rb:95:in `run'
  C:/tools/ruby31/lib/ruby/3.1.0/rubygems/ext/builder.rb:44:in `block in make'
  C:/tools/ruby31/lib/ruby/3.1.0/rubygems/ext/builder.rb:36:in `each'
  C:/tools/ruby31/lib/ruby/3.1.0/rubygems/ext/builder.rb:36:in `make'
C:/tools/ruby31/lib/ruby/3.1.0/rubygems/ext/ext_conf_builder.rb:63:in `block
in build'
  C:/tools/ruby31/lib/ruby/3.1.0/tempfile.rb:317:in `open'
  C:/tools/ruby31/lib/ruby/3.1.0/rubygems/ext/ext_conf_builder.rb:26:in `build'
C:/tools/ruby31/lib/ruby/3.1.0/rubygems/ext/builder.rb:161:in
`build_extension'
C:/tools/ruby31/lib/ruby/3.1.0/rubygems/ext/builder.rb:195:in `block in
build_extensions'
  C:/tools/ruby31/lib/ruby/3.1.0/rubygems/ext/builder.rb:192:in `each'
C:/tools/ruby31/lib/ruby/3.1.0/rubygems/ext/builder.rb:192:in
`build_extensions'
  C:/tools/ruby31/lib/ruby/3.1.0/rubygems/installer.rb:853:in `build_extensions'
C:/tools/ruby31/lib/ruby/gems/3.1.0/gems/bundler-2.2.27/lib/bundler/rubygems_gem_installer.rb:66:in
`build_extensions'
C:/tools/ruby31/lib/ruby/gems/3.1.0/gems/bundler-2.2.27/lib/bundler/rubygems_gem_installer.rb:26:in
`install'
C:/tools/ruby31/lib/ruby/gems/3.1.0/gems/bundler-2.2.27/lib/bundler/source/rubygems.rb:199:in
`install'
C:/tools/ruby31/lib/ruby/gems/3.1.0/gems/bundler-2.2.27/lib/bundler/installer/gem_installer.rb:54:in
`install'
C:/tools/ruby31/lib/ruby/gems/3.1.0/gems/bundler-2.2.27/lib/bundler/installer/gem_installer.rb:16:in
`install_from_spec'
C:/tools/ruby31/lib/ruby/gems/3.1.0/gems/bundler-2.2.27/lib/bundler/installer/parallel_installer.rb:186:in
`do_install'
C:/tools/ruby31/lib/ruby/gems/3.1.0/gems/bundler-2.2.27/lib/bundler/installer/parallel_installer.rb:171:in
`install_serially'
C:/tools/ruby31/lib/ruby/gems/3.1.0/gems/bundler-2.2.27/lib/bundler/installer/parallel_installer.rb:97:in
`call'
C:/tools/ruby31/lib/ruby/gems/3.1.0/gems/bundler-2.2.27/lib/bundler/installer/parallel_installer.rb:71:in
`call'
C:/tools/ruby31/lib/ruby/gems/3.1.0/gems/bundler-2.2.27/lib/bundler/installer.rb:262:in
`install_in_parallel'
C:/tools/ruby31/lib/ruby/gems/3.1.0/gems/bundler-2.2.27/lib/bundler/installer.rb:209:in
`install'
C:/tools/ruby31/lib/ruby/gems/3.1.0/gems/bundler-2.2.27/lib/bundler/installer.rb:89:in
`block in run'
C:/tools/ruby31/lib/ruby/gems/3.1.0/gems/bundler-2.2.27/lib/bundler/process_lock.rb:12:in
`block in lock'
C:/tools/ruby31/lib/ruby/gems/3.1.0/gems/bundler-2.2.27/lib/bundler/process_lock.rb:9:in
`open'
C:/tools/ruby31/lib/ruby/gems/3.1.0/gems/bundler-2.2.27/lib/bundler/process_lock.rb:9:in
`lock'
C:/tools/ruby31/lib/ruby/gems/3.1.0/gems/bundler-2.2.27/lib/bundler/installer.rb:71:in
`run'
C:/tools/ruby31/lib/ruby/gems/3.1.0/gems/bundler-2.2.27/lib/bundler/installer.rb:23:in
`install'
C:/tools/ruby31/lib/ruby/gems/3.1.0/gems/bundler-2.2.27/lib/bundler/cli/install.rb:60:in
`run'
C:/tools/ruby31/lib/ruby/gems/3.1.0/gems/bundler-2.2.27/lib/bundler/cli.rb:252:in
`block in install'
C:/tools/ruby31/lib/ruby/gems/3.1.0/gems/bundler-2.2.27/lib/bundler/settings.rb:131:in
`temporary'
C:/tools/ruby31/lib/ruby/gems/3.1.0/gems/bundler-2.2.27/lib/bundler/cli.rb:251:in
`install'
C:/tools/ruby31/lib/ruby/gems/3.1.0/gems/bundler-2.2.27/lib/bundler/vendor/thor/lib/thor/command.rb:27:in
`run'
C:/tools/ruby31/lib/ruby/gems/3.1.0/gems/bundler-2.2.27/lib/bundler/vendor/thor/lib/thor/invocation.rb:127:in
`invoke_command'
C:/tools/ruby31/lib/ruby/gems/3.1.0/gems/bundler-2.2.27/lib/bundler/vendor/thor/lib/thor.rb:392:in
`dispatch'
C:/tools/ruby31/lib/ruby/gems/3.1.0/gems/bundler-2.2.27/lib/bundler/cli.rb:31:in
`dispatch'
C:/tools/ruby31/lib/ruby/gems/3.1.0/gems/bundler-2.2.27/lib/bundler/vendor/thor/lib/thor/base.rb:485:in
`start'
C:/tools/ruby31/lib/ruby/gems/3.1.0/gems/bundler-2.2.27/lib/bundler/cli.rb:25:in
`start'
C:/tools/ruby31/lib/ruby/gems/3.1.0/gems/bundler-2.2.27/exe/bundle:49:in
`block in <top (required)>'
C:/tools/ruby31/lib/ruby/gems/3.1.0/gems/bundler-2.2.27/lib/bundler/friendly_errors.rb:128:in
`with_friendly_errors'
C:/tools/ruby31/lib/ruby/gems/3.1.0/gems/bundler-2.2.27/exe/bundle:37:in `<top
(required)>'
  C:/tools/ruby31/bin/bundle:25:in `load'
  C:/tools/ruby31/bin/bundle:25:in `<main>'

An error occurred while installing json (2.5.1), and Bundler cannot continue.

In Gemfile:
  cocoapods-check was resolved to 1.1.0, which depends on
    cocoapods was resolved to 1.10.2, which depends on
      cocoapods-core was resolved to 1.10.2, which depends on
        algoliasearch was resolved to 1.27.5, which depends on
          json
```

### Device

Windows 10

### VisionCamera Version

Release 2.14.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Veragin, 2022-10-24:
try `sudo bundle install`

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

closed 2023-09-30 by mrousavy