margelo/react-native-vision-camera #1190: Execution failed for task ':react-native-vision-camera:compileDebugKotlin'. 🐛

HenrikZabel, 2022-08-13:
### What were you trying to do?

I wanted to start my app after installing react-native-vision-camera.

### Reproduceable Code

```tsx
npx react-native run-android
```

### What happened instead?

* What went wrong:
Execution failed for task ':react-native-vision-camera:compileDebugKotlin'.
> Compilation error. See log for more details

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 2m 12s

    at makeError (C:\HENRIK\App\my_app\my_app\node_modules\@react-native-community\cli-platform-android\node_modules\execa\index.js:174:9)
    at C:\HENRIK\App\my_app\my_app\node_modules\@react-native-community\cli-platform-android\node_modules\execa\index.js:278:16
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at async runOnAllDevices (C:\HENRIK\App\my_app\my_app\node_modules\@react-native-community\cli-platform-android\build\commands\runAndroid\runOnAllDevices.js:109:5)
    at async Command.handleAction (C:\HENRIK\App\my_app\my_app\node_modules\@react-native-community\cli\build\index.js:192:9)
info Run CLI with --verbose flag for more details.

### Relevant log output

```shell
"react-native": "0.67.1",
"react-native-vision-camera": "v2.13.1",
```

### Device

Android Emulator Google Pixel 4 XL

### VisionCamera Version

v2.13.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Easy47, 2022-08-22:
Same problem here.

namhihi237, 2022-10-15:
same problem

princefishthrower, 2022-10-19:
I have reason to believe that it is something sour with RN 0.67.X versions and this lib, I am on 0.67.2 and can't build successfully for Android.

But if you look at many comments in https://github.com/mrousavy/react-native-vision-camera/issues/1012 people claim that 2.13.0 'works fine'. Many of them fail to state what RN version they are using though 🙄

HenrikZabel, 2022-11-26:
Solved with updating all dependencies including React Native and React Native Vision Camera.

Versions:
    "react-native": "0.70.6",
    "react-native-vision-camera": "^2.15.2"

Now it is working. Don't know exactly what the problem was/is

closed 2022-11-26 by HenrikZabel

SantoshYadavSridix, 2024-04-23:
Please, Upgrade to the latest version of dependencies.
In my case, i got this error because i am using 1.4.0 instead of 1.6.0
Execution failed for task ':react-native-vision-camera-face-detector:compileDebugKotlin'.