margelo/react-native-vision-camera #1192: Is there a way to capture what is shown in the camera compnent and pass that down as a frame?❓

gtzippy, 2022-08-13:
### Question

Is there a way for me to capture the image shown in the camera view and pass that down as a frame instead of everything the camera sees?

Let's say the camera component does not use absolute fill and is set to be a 600x400 rectangle. The camera sees a frame containing much more than the view shows, but I only want what is visable in the view to be passed to my plugin. Can this be done? 

### What I tried

I've tried to figure out a way to draw a rectangle with known coordinates over the camera view. This doesn't work because I can't find anything to just draw a simple green rectangle on the camera view. Also, the resolution of the picture that my phone takes and the imageproxy that shows up in the java plugin are different, so I can't cheat and just use the upper lefthand corner of the camera view as my 0,0. 

### VisionCamera Version

example project

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-08-15:
Hi, I wanted to add a `transformToCameraCoordinates` method, but haven't got around to doing that yet. For now you can get the view dimensions, get the camera format dimensions and calculate the aspect ratio crop accordingly (bit of tricky maths)

gtzippy, 2022-08-18:
I don't have a problem with tricky math. The logging shows my phone capturing the image as 4032x2268 but the size of the ImageProxy in the java plugin is only 3024x3024. I have not figured out how to get the top left of the captured camera image to be the top left of the ImageProxy. I was hoping to find some mechanism for forcing a resolution on the camera but I have not seen it. I might be blind though. I have been doing most of this with a 1-year-old sleeping in my lap in the middle of the night.

shiv71, 2022-08-30:
@gtzippy were you able to figure it out? I'm also struggling with this.

vitzaoral, 2022-11-14:
Hi, something new about _transformToCameraCoordinates_ ? I tried draw frame to screen but no luck, it's no consistent for Android and iOS 😢

mrousavy, 2022-11-14:
Hi - no not yet. We are planning V3, but want to raise money from companies using VisionCamera to fund the development of all of the planned features. If you're interested, reach out to me.

abdelrahman-muntaser, 2023-09-17:
Hello , any updates regarding this issue right now, or any workaround ?

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

closed 2023-09-30 by mrousavy