margelo/react-native-vision-camera #1193: 🐛 Bulid failed: Configure project :react-native-gradle-plugin

mujehoxe, 2022-08-14:
### What were you trying to do?

Want to scan qr code

### What happened instead?

unable to build after adding frame processor code and dependencies

### Relevant log output

```shell
info Running jetifier to migrate libraries to AndroidX. You can disable it using "--no-jetifier" flag.
Jetifier found 1039 file(s) to forward-jetify. Using 4 workers...
info JS server already running.
info Installing the app...

> Configure project :react-native-gradle-plugin
e: /home/o/Documents/clj/repos/react-native/node_modules/react-native-gradle-plugin/build.gradle.kts:9:19: Unresolved reference: configurationcache
e: /home/o/Documents/clj/repos/react-native/node_modules/react-native-gradle-plugin/build.gradle.kts:43:7: Unresolved reference: serviceOf

FAILURE: Build failed with an exception.

* Where:
Build file '/home/o/Documents/clj/repos/react-native/node_modules/react-native-gradle-plugin/build.gradle.kts' line: 9

* What went wrong:
Script compilation errors:

  Line 09: import org.gradle.configurationcache.extensions.serviceOf
                             ^ Unresolved reference: configurationcache

  Line 43:       serviceOf<ModuleRegistry>().getModule("gradle-tooling-api-builders").classpath.asFiles.first()
                 ^ Unresolved reference: serviceOf

2 errors

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 5s

error Failed to install the app. Make sure you have the Android development environment set up: https://reactnative.dev/docs/environment-setup.
Error: Command failed: ./gradlew app:installDebug -PreactNativeDevServerPort=8081
e: /home/o/Documents/clj/repos/react-native/node_modules/react-native-gradle-plugin/build.gradle.kts:9:19: Unresolved reference: configurationcache
e: /home/o/Documents/clj/repos/react-native/node_modules/react-native-gradle-plugin/build.gradle.kts:43:7: Unresolved reference: serviceOf

FAILURE: Build failed with an exception.

* Where:
Build file '/home/o/Documents/clj/repos/react-native/node_modules/react-native-gradle-plugin/build.gradle.kts' line: 9

* What went wrong:
Script compilation errors:

  Line 09: import org.gradle.configurationcache.extensions.serviceOf
                             ^ Unresolved reference: configurationcache

  Line 43:       serviceOf<ModuleRegistry>().getModule("gradle-tooling-api-builders").classpath.asFiles.first()
                 ^ Unresolved reference: serviceOf

2 errors

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 5s

    at makeError (/home/o/Documents/clj/repos/react-native/node_modules/execa/index.js:174:9)
    at /home/o/Documents/clj/repos/react-native/node_modules/execa/index.js:278:16
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at async runOnAllDevices (/home/o/Documents/clj/repos/react-native/node_modules/@react-native-community/cli-platform-android/build/commands/runAndroid/runOnAllDevices.js:109:5)
    at async Command.handleAction (/home/o/Documents/clj/repos/react-native/node_modules/@react-native-community/cli/build/index.js:192:9)
info Run CLI with --verbose flag for more details.
```

### Device

genymotion emulator

### VisionCamera Version

2.14.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

oknpc, 2022-11-05:
i had this problem, anyone find a solution?

joaogithub, 2022-11-11:
Same issue here. Any idea?

Thanks

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy