margelo/react-native-vision-camera #1196: 🐛

deevus, 2022-08-17:
### What were you trying to do?

Record a video on iOS. 

### What happened instead?

The preview is mirrored as expected, however the resulting video is unchanged (i.e. still mirrored).

It looks as though it might have been a sneaky iOS change at some point in the last month or 2. We haven't released a new major version of our app since March, however some users have recently reported this issue in the last few weeks. 

### Device

iPhone Xs Max iOS 15.6, iPhone 13 Pro Max iOS 15.x

### VisionCamera Version

2.12.0, 2.14.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

deevus, 2022-09-07:
I don't know how I missed adding a title, but I'm not surprised this was missed.

deevus, 2022-09-07:
@mrousavy Any ideas why this would be happening? Or a temporary fix?

deevus, 2022-09-07:
The following patch appears to fix the issue in my limited testing. 

```diff
diff --git a/node_modules/react-native-vision-camera/ios/CameraView+Orientation.swift b/node_modules/react-native-vision-camera/ios/CameraView+Orientation.swift
index 5c6db1f..880c934 100644
--- a/node_modules/react-native-vision-camera/ios/CameraView+Orientation.swift
+++ b/node_modules/react-native-vision-camera/ios/CameraView+Orientation.swift
@@ -49,10 +49,10 @@ extension CameraView {
         // Run those updates on cameraQueue since they can be blocking.
         self.captureSession.outputs.forEach { output in
           output.connections.forEach { connection in
-            if connection.isVideoMirroringSupported {
-              connection.automaticallyAdjustsVideoMirroring = false
-              connection.isVideoMirrored = isMirrored
-            }
+            /*if connection.isVideoMirroringSupported {*/
+              /*connection.automaticallyAdjustsVideoMirroring = false*/
+              /*connection.isVideoMirrored = isMirrored*/
+            /*}*/
             connection.setInterfaceOrientation(connectionOrientation)
           }
         }
```

HorbachAndrii, 2023-07-05:
Will be enough to set false to isVideoMirrored in the [CameraView+Orientation.swift](https://github.com/mrousavy/react-native-vision-camera/blob/3ed019d018cce9ff20afb0fcc81146732419f17e/ios/CameraView%2BOrientation.swift#L39) file
`connection.isVideoMirrored = false`

https://github.com/mrousavy/react-native-vision-camera/blob/3ed019d018cce9ff20afb0fcc81146732419f17e/ios/CameraView%2BOrientation.swift#L39

mrousavy, 2023-07-06:
I wonder what `automaticallyAdjustsVideoMirroring` actually does?

HorbachAndrii, 2023-07-06:
> I wonder what `automaticallyAdjustsVideoMirroring` actually does?

As I understand correctly the [`automaticallyAdjustsVideoMirroring`](https://developer.apple.com/documentation/avfoundation/avcaptureconnection/1387082-automaticallyadjustsvideomirrori) automatically adjusts the video mirroring based on the capture device's position, while [`isVideoMirrored`](https://developer.apple.com/documentation/avfoundation/avcaptureconnection/1389172-isvideomirrored) allows you to explicitly set whether the video should be mirrored or not, overriding the automatic behavior.

I think it is nice to have a property for the activation mirror.

mrousavy, 2023-07-06:
I believe in all of the Cameras I have used so far, I want the selfie cam to be mirrored (which is exactly what Snapchat and Instagram are doing). If the resulting video/photo is not mirrored, it feels weird and looks off.

I can add this prop to the backlog but will put this on very low priority as there are just a million things to do with V3 right now

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

Video Mirroring is default for front camera

closed 2023-09-30 by mrousavy

mrousavy, 2024-07-02:
Hey all! I just spent some time to build a new feature: I created a PR to add a `isMirrored` prop to Camera: https://github.com/mrousavy/react-native-vision-camera/pull/3036

With this prop you can enable or disable output mirroring 🎉🥳

If you appreciate my time and dedication towards building new features and constantly improving this library, please [**consider sponsoring me on GitHub**](https://github.com/sponsors/mrousavy/)  :)