margelo/react-native-vision-camera #1199: Translating bounds from Camera frame to a specific place in screen❓

JesusLozada1, 2022-08-18:
### Question

I'm trying to track the face's movement in the screen, however when I'm displaying the bounds on the phone screen its position is relative to the frame itself and not the screen. I'm trying to figure out how to scale between the frame and the screen within a certain place.

### What I tried

I've tried dividing the bound's **x** and **y** by a certain amount that works with my screen, however it's not an exact match.

### VisionCamera Version

2.14.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

codeapp17, 2022-10-12:
Are you able to find solution for this ?

JesusLozada1, 2022-11-29:
@codeapp17 not yet, i have not found a way to come up with a formula that would make the frame's dimensions and phone screen dimensions proportionate. there are too many variable factors that make it hard to make a concrete formula to correctly display face's bounds.

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

closed 2023-09-30 by mrousavy