margelo/react-native-vision-camera #1202: ❓ If frameProcessorFps is 5, shouldn't frameProcessor run 5 times per second?

Noma98, 2022-08-22:
### Question

If frameProcessorFps is 5, shouldn't frameProcessor run 5 times per second? Assigning 1 to frameProcessorFps calls exactly once every second. However, if I assign a number greater than 2, it is called only once or twice a second.

Please see the code below (this is an excerpt of my project code made using reaction-native-vision-camera). The text value recognized in runOnJs is saved as ocr and the processRecognizedText is executed whenever the ocr value changes to useEffect. At this time, I guessed that the frame processor could not run as many times as I wanted per second because it took a long time to run processRecognizedText, but when I checked it, it took 0.001 seconds.

If so, I think processRecognizedText should be excuted 100 times if frameProcessorFps is set to 100, but why is it only excuted 1~2 times?
I would appreciate it if you could check if I misunderstood the function of frame processor Fps and if there was anything wrong set up! Thank you.
```
import React, { useEffect, useState } from 'react';
import 'react-native-reanimated';
import { runOnJS } from 'react-native-reanimated';
import {
  Camera,
  useCameraDevices,
  useFrameProcessor,
} from 'react-native-vision-camera';
import { scanOCR } from 'vision-camera-ocr';
(...)

const DataCaptureView = ({ navigation, route }: IProps) => {
(...)
  const [enabled, setEnabled] = useState(true);
  const [ocr, setOcr] = useState();
  const devices = useCameraDevices();
  const device = devices[cameraPosition];

  const frameProcessor = useFrameProcessor(frame => {
    'worklet';
    const data = scanOCR(frame);
    runOnJS(setOcr)(data?.result?.text);
  }, []);

  useEffect(() => {
    if (!Boolean(ocr)) {
      return;
    }
    enabled && processRecognizedText(ocr);
  }, [ocr, enabled]);

  const processRecognizedText = (text: string) => {
    (...)
  };

  return (
    <StyledView>
      {device !== undefined && hasPermission ? (
          <Camera
            style={{ flex: 1, width: '100%' }}
            frameProcessor={frameProcessor}
            device={device}
            isActive={isAppForeground && isFocused}
            frameProcessorFps={5}
          />
      ) : (
        <StyledView />
      )}
    </StyledView>
  );
};
const StyledView = styled.View`
  flex: 1;
`;
export default DataCaptureView;
```

### What I tried

아래 코드를 보면, runOnJs에 인식된 텍스트 값을 ocr이라는 상태로 저장해놓고 useEffect로 ocr 값이 변할때 마다 proccessRecognizedText를 실행합니다. 이때 processRecognizedText 실행시 소요되는 시간이 길어서 1초당 원하는 횟수만큼 frameprocessor가 실행이 되지 않는 건가 추측했지만, 확인해보니 해당 함수는 0.001초가 걸립니다.

제가 frameProcessorFps의 기능을 잘못 이해했는지, 잘못 설정한 점은 없는지 확인해 주시면 감사하겠습니다! frameProcessorFps이 5이면 1초에 5번 frameProcessor가 실행되어져야 하는 거 아닌가요? frameProcessorFps에 1을 할당하면 정확히 1초에 한 번씩 호출합니다. 그러나 내가 2보다 큰 숫자를 할당하면 1초에 1~2번만 호출됩니다.

### VisionCamera Version

2.13.5

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

hikmert, 2022-10-06:
I have exactly the same issue, setting frameprocessorfps to , let's say 30, and I only receive around 12-13 frames from native Android code. Any ideas?

thomas-coldwell, 2022-10-21:
Hey everyone, I think I might have some insight onto your issues here:
- Firstly, on Android you can set the Frame Processor FPS as high as you like but internally it will be capped at the FPS CameraX decides on for its ImageAnalysis use case (in my testing this was around 15-20fps). There are ways to improve this FPS, but it requires setting up some custom capture requests with Camera2 through CameraX's interop API - maybe this is something we could think about sorting out for V3 so you can disable some camera features to improve the ImageAnalysis use case @mrousavy ?
- Secondly, if you're running this on iOS I just fixed an issue that you can see in this PR #1288 that was causing it to skip ever other frame (if the frame processor callback was able to execute in real-time)

mrousavy, 2022-10-24:
> but it requires setting up some custom capture requests with Camera2 through CameraX's interop API

CameraX is limiting us more and more. We should talk about using Camera2 instead

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

closed 2023-09-30 by mrousavy