margelo/react-native-vision-camera #1203: ❓ Reduce release bundle size

PavelRazumov, 2022-08-22:
### Question

Hi, team.

I use this library only for reading barcodes. My question is is there an opportunity to reduce the size of the bundle of the final release application? As an example for comparison, I created two applications using npx react-native init https://reactnative.dev/docs/environment-setup (version 0.66.0), and for one of them I added the react-native-vision camera library (version 2.13.2) with reanimated (version 2.4.0) and launched the release build with the minifyEnabled true flag

// **Empty app**

[image]

// App with react-native-vision-camera

[image]

Is there any solution to optimize the releaser apk build ?

### What I tried

1. Create two empty projects with help npx react-native init 
2. One of them add dependencies react-native-vision-camera, react-native-reanimated and vision-camera-code-scanner
3. Launching release build for both app with flag minifyEnabled true.
4. Check apk's result

### VisionCamera Version

2.13.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

PavelRazumov, 2022-08-22:
I try add compile flag like -Os or -O3 (https://gcc.gnu.org/onlinedocs/gcc/Optimize-Options.html#Optimize-Options) in build.gradle in react-native-vision-camera, but unfortunately it didn't work

babaiyu, 2022-09-06:
In my case, i was set `minSdkVersion = 23` and apk size is increased. Finally i change it to <b>21</> .

 > I will back later to share the information source

PavelRazumov, 2022-09-06:
@babaiyu, hi, 

I don't understand, what do you mean about minSdkVersion and boostsplah? 
In my question i want to reduce size of react-native-vision-camera only, and my minSdkVersion is 23 it's actual version of Android 6 (out minimal support version)

babaiyu, 2022-09-06:
> @babaiyu, hi,
> 
> I don't understand, what do you mean about minSdkVersion and boostsplah? In my question i want to reduce size of react-native-vision-camera only, and my minSdkVersion is 23 it's actual version of Android 6 (out minimal support version)

Nevermind with bootsplash, it just example case to other libraries hehe.

https://stackoverflow.com/a/68219448
I found this question from stackoverflow, please redirect to link above. I hope it can help to.

Sorry before, my bad english.

PavelRazumov, 2022-09-06:
ok, thank you, i'll check it later and return with result

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy