margelo/react-native-vision-camera #1205: ❓

seriousStar, 2022-08-22:
### Question

How to set the zoom without any gesture?

### What I tried

I checked the example of vision camera, but it doesn't demonstrate the way to use zoom API, instead it uses gesture to zoom in & out, I just wanted to know how to zoom in & out whenever the user clicks the zoom options. Please check the attachment.
As you can see, there are 3 options - 0.5x, 1x, 3x. 

So there are 2 questions here:
- Is there any method to zoom in & out without using gesture? If so, what's the method? and pls provide me the example.
- Is it possible to zoom out as 0.5x? if so, how to do?
[image]

### VisionCamera Version

^2.14.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

adembacajmemmo, 2022-08-29:
Did you try to play with zoom field? Try to change that number (honestly didn't try it before, but I guess it should work)

adembacajmemmo, 2022-08-30:
[image]

There is zoom, you can also find documentation for it here: https://mrousavy.com/react-native-vision-camera/docs/guides/zooming

Except animation zoom, you can change that field by yourself

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy