margelo/react-native-vision-camera #1207: ❓ How to change the cameras smoothly?

seriousStar, 2022-08-23:
### Question

How to change the cameras smoothly?

### What I tried

I checked the example of this repo, but couldn't find to switch across multiple cameras.
For example, iPhone 13 pro has 3 cameras, and I wanted to know how to switch between them.
Please provide me the code snippets if it's possible.

### VisionCamera Version

^2.14.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

franciscocavalli, 2022-09-25:
Hi, have you find a solution?
Thank you

seriousStar, 2022-10-09:
Not yet.

Sid-Turner-Ellis, 2023-01-27:
What have you done for a workaround to this?

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy