margelo/react-native-vision-camera #1209: ❓ Why I am getting only denied from requestCameraPermission

JaLe29, 2022-08-24:
### Question

Hi,

(this is my first contact with react native).

I am stuck on camera setup. I am getting always `denied` response from `await Camera.requestCameraPermission()` I dont know why? I never clicked on decline button. 

I tested on android emulator, my real device.

Full code:
```
	const getRequestCameraPermission = async () => {
		console.log('CLICK')
		const newCameraPermission = await Camera.requestCameraPermission();
		ToastAndroid.show(newCameraPermission, ToastAndroid.SHORT);
	}
```

This function is fired after click on button.

### What I tried

google
issues
SO

### VisionCamera Version

2.14.1

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

kicksent, 2022-09-03:
Have you looked at the settings in the phone/emulator you are using?

xeroxoid, 2022-12-11:
We are experiencing the same issue on Android. Permission is always `denied` and the prompt to allow camera and/or microphone is never shown.

isaacparsons-dw, 2023-01-11:
I too had this issue. Permission was always 'denied' even after a fresh instance of the app installed onto a physical device. No prompt either - just denies straight away. 

In the end i've had to request the permission manually through `PermissonsAndroid` from React Native. 

Here is a helper function to request permissons with Android:

`  const requestAndroidCameraPermission = useCallback(async () => {
    try {
      const permissionGranted = await PermissionsAndroid.request(PermissionsAndroid.PERMISSIONS.CAMERA, {
        title: '<your title here>',
        message:
          '<your message here>',
        buttonNegative: 'Deny',
        buttonPositive: 'Allow',
      });
      // then access permission status
      if (granted === PermissionsAndroid.RESULTS.GRANTED) {
        // permissons have been accepted - update a useState() here or whatever your usecase is :)
      }
    } catch (err) {
      console.warn(err);
    }
  }, []);
`

Conditionally fire above function based on OS... in case anyone needs it
`  if (Platform.OS === 'android') {
    requestAndroidCameraPermission()
  }`

Hopefully the above is helpful

buenaondalab, 2023-02-07:
> I too had this issue. Permission was always 'denied' even after a fresh instance of the app installed onto a physical device. No prompt either - just denies straight away.
> 
> In the end i've had to request the permission manually through `PermissonsAndroid` from React Native.
> 
> Here is a helper function to request permissons with Android:
> 
> ```ts
> import { PermissionsAndroid } from 'react-native';
> const requestAndroidCameraPermission = useCallback(async () => {
>     try {
>       const permissionGranted = await PermissionsAndroid.request(PermissionsAndroid.PERMISSIONS.CAMERA, {
>         title: '<your title here>',
>         message:
>           '<your message here>',
>         buttonNegative: 'Deny',
>         buttonPositive: 'Allow',
>       });
>       // then access permission status
>       if (granted === PermissionsAndroid.RESULTS.GRANTED) {
>         // permissons have been accepted - update a useState() here or whatever your usecase is :)
>       }
>     } catch (err) {
>       console.warn(err);
>     }
>   }, []);
> ```
> 
> Conditionally fire above function based on OS... in case anyone needs it
> 
> ```ts
> if (Platform.OS === 'android') {
>     requestAndroidCameraPermission()
>   }
> ```
> 
> Hopefully the above is helpful

Same issue here. Solved using `PermissionsAndroid`. Thanks to @isaacparsons-dw.
We still waiting for an official solution.

sekhuat, 2023-03-07:
I guessed the proper way is to use the requestCameraPermission method from the module itself.

```
  const devices = useCameraDevices();
  const device = devices.back;
  
  React.useEffect(() => {
    if(device) {
      Camera.requestCameraPermission()
      .then(result => {
        console.log('requestCameraPermission: ', result);
      })
      .catch(err => {
        console.log('requestCameraPermission Err: ', err);
      });
    }
  }, [device]);
```

sahinbezeng, 2023-07-05:
Hey guys,

Probably you haven't added the permission to `AndroidManifest.xml`

Same happened to me so i just added the line 
`<uses-permission android:name="android.permission.CAMERA" />` to  `AndroidManifest.xml`
and problem solved.

davidfeldt, 2023-08-09:
I only had this issue on Android when requesting more than one permission.  For recording video (use of camera) and audio, there is only the prompt for camera permission and there is no audio permission prompt.  

My solution was to use the following:
```
import {PermissionsAndroid} from 'react-native';
const getAllPermissions = useCallback(async () => {
    PermissionsAndroid.requestMultiple([
      PermissionsAndroid.PERMISSIONS.CAMERA,
      PermissionsAndroid.PERMISSIONS.RECORD_AUDIO,
    ]).then(result => {
      if (
        result['android.permission.CAMERA'] === 'granted' &&
        result['android.permission.RECORD_AUDIO'] === 'granted'
      ) {
        setLoading(devices);
      } else {
        Alert.alert(
          'Permission Not Granted',
          'You need to grant this app permission to use the camera to record audio and video',
          [
            {
              text: 'Cancel',
              onPress: () => navigation.pop(),
            },
            {
              text: 'Grant',
              onPress: () => Linking.openSettings(),
            },
          ],
        );
        setLoading(null);
      }
    });
  }, [navigation, devices]);

  useEffect(() => {
    getAllPermissions();
  }, [getAllPermissions, devices]);
```

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy

manish0707, 2023-10-02:
> I too had this issue. Permission was always 'denied' even after a fresh instance of the app installed onto a physical device. No prompt either - just denies straight away.
>
> In the end i've had to [...]

What about IOS?

mrousavy, 2023-10-03:
This has been fixed now btw, Permissions dont return denied on android anymore

KamranBarlas111, 2023-10-04:
@mrousavy It's not fixed yet. I'm getting permission **denied** on asking camera permission 1st time and without selecting any option from the permission dialog.

Kata-C, 2023-10-10:
> Hey guys,
> 
> Probably you haven't added the permission to `AndroidManifest.xml`
> 
> Same happened to me so i just added the line `<uses-permission android:name="android.permission.CAMERA" />` to `AndroidManifest.xml` and problem solved.

This worked for me!