margelo/react-native-vision-camera #1211: 🐛

fobin, 2022-08-24:
### What were you trying to do?

Trying to use FrameProcessor from `vision-camera-code-scanner`. Not using Hermes. 

This was was similar issue but states specifically that using Hermes causes it so I guess this is not duplicate: #997.

All the basics tried: rm -rf node_modules, gradlew clean, yarn react-native start --reset-cache..

`"react-native": "0.68.2",`
`"react-native-vision-camera": "^2.14.1",`
`"vision-camera-code-scanner": "^0.2.0"`

### Reproduceable Code

```tsx
const [frameProcessor, barcodes] = useScanBarcodes([BarcodeFormat.QR_CODE], {
    checkInverted: true,
  });

....

<Camera
        style={StyleSheet.absoluteFill}
        device={device}
        isActive={true}
        frameProcessor={frameProcessor}
        frameProcessorFps={'auto'}
      />
```

### What happened instead?

Crash on Android. iOS works fine. 

### Relevant log output

```shell
art_sigsegv_fault 0x00000075c2a4b58c
art::FaultManager::HandleFault(int, siginfo*, void*) 0x00000075c2a4b3f8
art::SignalChain::Handler(int, siginfo*, void*) 0x000000787a860328
<unknown> 0x000000788562c89c
__gxx_personality_v0 0x000000753512ec90
__gxx_personality_v0 0x000000753512ec90
unwind_phase2 0x000000750c397728
_Unwind_Resume 0x000000750c3977f8
vision::FrameProcessorRuntimeManager::registerPlugin(facebook::jni::alias_ref<facebook::jni::detail::JTypeFor<vision::JFrameProcessorPlugin, facebook::jni::JObject, void>::_javaobject*>)::$_4::operator()(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long) const 0x000000750c363dec
decltype(std::__ndk1::forward<vision::FrameProcessorRuntimeManager::registerPlugin(facebook::jni::alias_ref<facebook::jni::detail::JTypeFor<vision::JFrameProcessorPlugin, facebook::jni::JObject, void>::_javaobject*>)::$_4&>(fp)(std::__ndk1::forward<facebook::jsi::Runtime&>(fp0), std::__ndk1::forward<facebook::jsi::Value const&>(fp0), std::__ndk1::forward<facebook::jsi::Value const*>(fp0), std::__ndk1::forward<unsigned long>(fp0))) std::__ndk1::__invoke<vision::FrameProcessorRuntimeManager::registerPlugin(facebook::jni::alias_ref<facebook::jni::detail::JTypeFor<vision::JFrameProcessorPlugin, facebook::jni::JObject, void>::_javaobject*>)::$_4&, facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long>(vision::FrameProcessorRuntimeManager::registerPlugin(facebook::jni::alias_ref<facebook::jni::detail::JTypeFor<vision::JFrameProcessorPlugin, facebook::jni::JObject, void>::_javaobject*>)::$_4&, facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*&&, unsigned long&&) 0x000000750c363be0
facebook::jsi::Value std::__ndk1::__invoke_void_return_wrapper<facebook::jsi::Value>::__call<vision::FrameProcessorRuntimeManager::registerPlugin(facebook::jni::alias_ref<facebook::jni::detail::JTypeFor<vision::JFrameProcessorPlugin, facebook::jni::JObject, void>::_javaobject*>)::$_4&, facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long>(vision::FrameProcessorRuntimeManager::registerPlugin(facebook::jni::alias_ref<facebook::jni::detail::JTypeFor<vision::JFrameProcessorPlugin, facebook::jni::JObject, void>::_javaobject*>)::$_4&, facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*&&, unsigned long&&) 0x000000750c363b2c
std::__ndk1::__function::__alloc_func<vision::FrameProcessorRuntimeManager::registerPlugin(facebook::jni::alias_ref<facebook::jni::detail::JTypeFor<vision::JFrameProcessorPlugin, facebook::jni::JObject, void>::_javaobject*>)::$_4, std::__ndk1::allocator<vision::FrameProcessorRuntimeManager::registerPlugin(facebook::jni::alias_ref<facebook::jni::detail::JTypeFor<vision::JFrameProcessorPlugin, facebook::jni::JObject, void>::_javaobject*>)::$_4>, facebook::jsi::Value (facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long)>::operator()(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*&&, unsigned long&&) 0x000000750c363aa4
std::__ndk1::__function::__func<vision::FrameProcessorRuntimeManager::registerPlugin(facebook::jni::alias_ref<facebook::jni::detail::JTypeFor<vision::JFrameProcessorPlugin, facebook::jni::JObject, void>::_javaobject*>)::$_4, std::__ndk1::allocator<vision::FrameProcessorRuntimeManager::registerPlugin(facebook::jni::alias_ref<facebook::jni::detail::JTypeFor<vision::JFrameProcessorPlugin, facebook::jni::JObject, void>::_javaobject*>)::$_4>, facebook::jsi::Value (facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long)>::operator()(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*&&, unsigned long&&) 0x000000750c362b3c
<unknown> 0x000000753898efbc
<unknown> 0x000000752de5ec18
<unknown> 0x000000752e0a10c4
<unknown> 0x000000752e08b29c
```
```

### Device

Samsung A23

### VisionCamera Version

2.14.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Productivix, 2022-09-12:
hi, 
works with : 
`"dependencies": {
    "react": "18.0.0",
    "react-native": "0.69.4",
    "react-native-reanimated": "^2.9.1",
    "react-native-vision-camera": "^2.14.1",
    "vision-camera-code-scanner": "^0.2.0"
  },`

letrungtrung, 2022-12-29:
hello bro, it still not resolved yet? any update or workaround ?

nc-hung, 2022-12-30:
This error occurs from package dependencies  ' **react-native-vision-camera** '.
refer to this link may help to solve the problem: https://mrousavy.com/react-native-vision-camera/docs/guides/troubleshooting/
with me just follow step 1.

nc-hung, 2022-12-30:
> hello bro, it still not resolved yet? any update or workaround ?

anh giải quyết xong lỗi này chưa ạ

angelica-snowit, 2022-12-30:
> This error occurs from package dependencies ' **react-native-vision-camera** '. refer to this link may help to solve the problem: https://mrousavy.com/react-native-vision-camera/docs/guides/troubleshooting/ with me just follow step 1.

As the first message, we already have read and tried the troubleshooting guide.

nc-hung, 2022-12-30:
> 
Have you deleted this file yet? ' package-lock.json '

yousrasd, 2023-01-20:
Any update on this? We are running into the same issue, and tried the troubleshooting guide without success.

404-html, 2023-02-04:
Exception captured with `adb logcat -b crash` when using frame processor with [vision-camera-ocr](https://github.com/aarongrider/vision-camera-ocr):

```
02-04 00:38:27.362 25087 25311 F libc    : Fatal signal 6 (SIGABRT), code -1 (SI_QUEUE) in tid 25311 (pool-37-thread-), pid 25087 (com.digdig)02-04 00:38:28.145 25314 25314 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
02-04 00:38:28.145 25314 25314 F DEBUG   : Build fingerprint: 'google/sargo/sargo:12/SP2A.220505.008/8782922:user/release-keys'
02-04 00:38:28.145 25314 25314 F DEBUG   : Revision: 'MP1.0'
02-04 00:38:28.145 25314 25314 F DEBUG   : ABI: 'arm64'
02-04 00:38:28.145 25314 25314 F DEBUG   : Timestamp: 2023-02-04 00:38:27.499392482+0100
02-04 00:38:28.145 25314 25314 F DEBUG   : Process uptime: 0s
02-04 00:38:28.145 25314 25314 F DEBUG   : Cmdline: com.digdig
02-04 00:38:28.145 25314 25314 F DEBUG   : pid: 25087, tid: 25311, name: pool-37-thread-  >>> com.digdig <<<
02-04 00:38:28.145 25314 25314 F DEBUG   : uid: 10354
02-04 00:38:28.145 25314 25314 F DEBUG   : signal 6 (SIGABRT), code -1 (SI_QUEUE), fault addr --------
02-04 00:38:28.145 25314 25314 F DEBUG   :     x0  0000000000000000  x1  00000000000062df  x2  0000000000000006  x3  0000006ec3a90470
02-04 00:38:28.146 25314 25314 F DEBUG   :     x4  00000000ebad808a  x5  00000000ebad808a  x6  00000000ebad808a  x7  00000000ebad808b
02-04 00:38:28.146 25314 25314 F DEBUG   :     x8  00000000000000f0  x9  d959baad6b56da4e  x10 0000000000000000  x11 ffffff80fffffbdf
02-04 00:38:28.146 25314 25314 F DEBUG   :     x12 0000000000000001  x13 0000000000000000  x14 0000000000000004  x15 0000000000000004
02-04 00:38:28.146 25314 25314 F DEBUG   :     x16 00000073b6246050  x17 00000073b6222db0  x18 0000006be5b76000  x19 00000000000061ff
02-04 00:38:28.146 25314 25314 F DEBUG   :     x20 00000000000062df  x21 00000000ffffffff  x22 fffffffffffffffe  x23 0000006ec3a906d0
02-04 00:38:28.146 25314 25314 F DEBUG   :     x24 0000000000000001  x25 000000715b6c8110  x26 0000000000000001  x27 000000717b883410
02-04 00:38:28.146 25314 25314 F DEBUG   :     x28 0000007060cdc218  x29 0000006ec3a904f0
02-04 00:38:28.146 25314 25314 F DEBUG   :     lr  00000073b61d5aa0  sp  0000006ec3a90450  pc  00000073b61d5acc  pst 0000000000000000
02-04 00:38:28.146 25314 25314 F DEBUG   : backtrace:
02-04 00:38:28.146 25314 25314 F DEBUG   :       #00 pc 000000000004facc  /apex/com.android.runtime/lib64/bionic/libc.so (abort+164) (BuildId: cd7952cb40d1a2deca6420c2da7910be)
02-04 00:38:28.146 25314 25314 F DEBUG   :       #01 pc 00000000000af7e4  /data/app/~~6IjAKqvztq-YRuqz-hGcfA==/com.digdig-Y5PWgZEWf771L5za6Vbbng==/lib/arm64/libc++_shared.so (BuildId: ece72a2ebc3774a1be9fd21271258acd3bcdfaa7)
02-04 00:38:28.146 25314 25314 F DEBUG   :       #02 pc 00000000000aedc8  /data/app/~~6IjAKqvztq-YRuqz-hGcfA==/com.digdig-Y5PWgZEWf771L5za6Vbbng==/lib/arm64/libc++_shared.so (BuildId: ece72a2ebc3774a1be9fd21271258acd3bcdfaa7)
02-04 00:38:28.146 25314 25314 F DEBUG   :       #03 pc 00000000000aebf8  /data/app/~~6IjAKqvztq-YRuqz-hGcfA==/com.digdig-Y5PWgZEWf771L5za6Vbbng==/lib/arm64/libc++_shared.so (__gxx_personality_v0+200) (BuildId: ece72a2ebc3774a1be9fd21271258acd3bcdfaa7)
02-04 00:38:28.146 25314 25314 F DEBUG   :       #04 pc 000000000022bd4c  /data/app/~~6IjAKqvztq-YRuqz-hGcfA==/com.digdig-Y5PWgZEWf771L5za6Vbbng==/lib/arm64/libreanimated.so (unwind_phase2+156) (BuildId: 82c0e59bc1713ddfef44d9aed5062b13b6812f7d)
02-04 00:38:28.146 25314 25314 F DEBUG   :       #05 pc 000000000022be08  /data/app/~~6IjAKqvztq-YRuqz-hGcfA==/com.digdig-Y5PWgZEWf771L5za6Vbbng==/lib/arm64/libreanimated.so (_Unwind_Resume+52) (BuildId: 82c0e59bc1713ddfef44d9aed5062b13b6812f7d)
02-04 00:38:28.146 25314 25314 F DEBUG   :       #06 pc 0000000000181c04  /data/app/~~6IjAKqvztq-YRuqz-hGcfA==/com.digdig-Y5PWgZEWf771L5za6Vbbng==/lib/arm64/libreanimated.so (facebook::jsi::Function::call(facebook::jsi::Runtime&, facebook::jsi::Value const*, unsigned long) const+144) (BuildId: 82c0e59bc1713ddfef44d9aed5062b13b6812f7d)
02-04 00:38:28.146 25314 25314 F DEBUG   :       #07 pc 00000000001b3884  /data/app/~~6IjAKqvztq-YRuqz-hGcfA==/com.digdig-Y5PWgZEWf771L5za6Vbbng==/lib/arm64/libreanimated.so (reanimated::ShareableValue::toJSValue(facebook::jsi::Runtime&)::$_2::operator()(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long)+428) (BuildId: 82c0e59bc1713ddfef44d9aed5062b13b6812f7d)
02-04 00:38:28.146 25314 25314 F DEBUG   :       #08 pc 00000000001b36b4  /data/app/~~6IjAKqvztq-YRuqz-hGcfA==/com.digdig-Y5PWgZEWf771L5za6Vbbng==/lib/arm64/libreanimated.so (decltype(std::__ndk1::forward<reanimated::ShareableValue::toJSValue(facebook::jsi::Runtime&)::$_2&>(fp)(std::__ndk1::forward<facebook::jsi::Runtime&>(fp0), std::__ndk1::forward<facebook::jsi::Value const&>(fp0), std::__ndk1::forward<facebook::jsi::Value const*>(fp0), std::__ndk1::forward<unsigned long>(fp0))) std::__ndk1::__invoke<reanimated::ShareableValue::toJSValue(facebook::jsi::Runtime&)::$_2&, facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long>(reanimated::ShareableValue::toJSValue(facebook::jsi::Runtime&)::$_2&, facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*&&, unsigned long&&)+128) (BuildId: 82c0e59bc1713ddfef44d9aed5062b13b6812f7d)
[... cut ...]
02-04 00:38:28.146 25314 25314 F DEBUG   :       #69 pc 000000000020a958  /apex/com.android.art/lib64/libart.so (nterp_helper+5720) (BuildId: 
a49c773ef6221a996ecea990e9753caa)
02-04 00:38:28.146 25314 25314 F DEBUG   :       #70 pc 00000000002027a4  [anon:dalvik-classes.dex extracted in memory from /data/app/~~6IjAKqvztq-YRuqz-hGcfA==/com.digdig-Y5PWgZEWf771L5za6Vbbng==/base.apk] (androidx.camera.core.ImageAnalysisAbstractAnalyzer$$ExternalSyntheticLambda1.run+28)
02-04 00:38:28.146 25314 25314 F DEBUG   :       #71 pc 000000000206e5b8  /memfd:jit-cache (deleted)
```

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy

mrousavy, 2023-10-06:
Hey!

JFYI; **VisionCamera V3 now includes a QR/Barcode Scanner!** 😍  Check out the [CodeScanner Documentation](https://www.react-native-vision-camera.com/docs/guides/code-scanning) 🚀 

Try upgrading to V3 if you can, it has a much more stable yet flexible foundation.

If you appreciate me dedicating my free time to improving VisionCamera and implementing features like the Code Scanner, please consider [**sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy) to show your support.