margelo/react-native-vision-camera #1212: 🐛 After recording many videos on Pixel 5, video stops recording

davidlcorbitt, 2022-08-25:
### What were you trying to do?

I was trying to record many short videos on my Pixel 5.

### Reproduceable Code

```tsx
I think the easiest way to reproduce this code is to clone the simple react-native repository I created to reproduce the bug: https://github.com/reality-platforms/recording-error-example/
```

### What happened instead?

After recording anywhere from 30-60 recordings without error, the video will suddenly stop recording properly, usually halfway through a recording.  The video preview will start pulsing about once a second.  Audio continues to be saved correctly to that specific recording, but video stops saving suddenly.  After that recording finishes and another one begins, no video or audio is saved to the next recording (although no error is thrown until the recording stops).  When the recording is stopped, `onRecordingError` receives the following error:
```
{
    "cause": {
        "message": "[capture/no-valid-data] The recording failed because no valid data was produced to be recorded.
            This error is generated when the essential data for a recording to be played correctly is missing, for example, a recording must contain at least one key frame. The application will need to clean up the output file, such as deleting the file.",
        "stacktrace": "com.mrousavy.camera.NoValidDataError: [capture/no-valid-data] The recording failed because no valid data was produced to be recorded.
            This error is generated when the essential data for a recording to be played correctly is missing, for example, a recording must contain at least one key frame. The application will need to clean up the output file, such as deleting the file.
                at com.mrousavy.camera.CameraView_RecordVideoKt$startRecording$1.accept(CameraView+RecordVideo.kt:62)
                at com.mrousavy.camera.CameraView_RecordVideoKt$startRecording$1.accept(CameraView+RecordVideo.kt:52)
                at androidx.camera.video.Recorder$RecordingRecord.lambda$updateVideoRecordEvent$6$androidx-camera-video-Recorder$RecordingRecord(Recorder.java:2624)
                at androidx.camera.video.Recorder$RecordingRecord$$ExternalSyntheticLambda6.run(Unknown Source:4)
                at android.os.Handler.handleCallback(Handler.java:942)
                at android.os.Handler.dispatchMessage(Handler.java:99)
                at android.os.Looper.loopOnce(Looper.java:201)
                at android.os.Looper.loop(Looper.java:288)
                at android.app.ActivityThread.main(ActivityThread.java:7898)
                at java.lang.reflect.Method.invoke(Native Method)
                at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
                at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:936)"
    },
    "code": "capture/no-valid-data",
    "message": "[capture/no-valid-data] The recording failed because no valid data was produced to be recorded. This error is generated when the essential data for a recording to be played correctly is missing, for example, a recording must contain at least one key frame. The application will need to clean up the output file, such as deleting the file.",
    "userInfo": null
}
```

The majority of the recordings after the first messed up recording also seem to be empty and throw an error.  On occasion a recording will be able to save, but usually the next recording will also throw an error.

### Relevant log output

```shell
I/ReactNativeJS: Starting a new recording
D/RecordVideo: file is "/data/user/0/<bundle-id>/cache/VisionCamera-20220824_1655345588239996310333588.mp4"
D/Recorder: Transitioning Recorder internal state: IDLING --> PENDING_RECORDING
D/Recorder: Transitioning Recorder internal state: PENDING_RECORDING --> RECORDING
D/Recorder: Sent VideoRecordEvent class androidx.camera.video.VideoRecordEvent$Start
D/VideoEncoder: Start on 68:46:01.125
D/MediaCodec: keep callback message for reclaim
W/Codec2Client: query -- param skipped: index = 1342179345.
W/Codec2Client: query -- param skipped: index = 2415921170.
W/Codec2Client: query -- param skipped: index = 1610614798.
D/CCodecBufferChannel: [c2.qti.avc.encoder#78] Query input allocators returned 0 params => BAD_INDEX (6)
D/CCodecBufferChannel: [c2.qti.avc.encoder#78] Using basic input block pool with poolID 1 => got 1 - OK (0)
D/CCodecBufferChannel: [c2.qti.avc.encoder#78] Query output allocators returned 0 params => BAD_INDEX (6)
I/CCodecBufferChannel: [c2.qti.avc.encoder#78] Created output block pool with allocatorID 16 => poolID 2238 - OK
D/CCodecBufferChannel: [c2.qti.avc.encoder#78] Configured output block pool ids 2238 => OK
D/VideoEncoder: Transitioning encoder internal state: CONFIGURED --> STARTED
D/Camera2CameraImpl: {Camera@9d1261d[id=0]} Use case androidx.camera.video.VideoCapture-044dd978-12c3-4df2-a41b-de7bad972427242771473 UPDATED
D/UseCaseAttachState: Active and attached use case: [androidx.camera.video.VideoCapture-044dd978-12c3-4df2-a41b-de7bad972427242771473, androidx.camera.core.Preview-e92b375b-187c-4698-aefb-9a1dc3a4c408133334584] for camera: 0
D/CaptureSession: Attempting to submit CaptureRequest after setting
D/CaptureSession: Issuing request for session.
D/ReactNativeJS: Skipping autofocusing on center for Android device
D/Recorder: Video source has transitioned to state: ACTIVE_STREAMING
D/Recorder: Transitioning Recorder internal state: RECORDING --> STOPPING
D/VideoEncoder: Transitioning encoder internal state: STARTED --> STOPPING
D/VideoEncoder: Stop on 68:46:04.724
D/Camera2CameraImpl: {Camera@9d1261d[id=0]} Issue capture request
[... 152 lines cut ...]
        mtime: Wed Aug 24 2022 16:55:34 GMT-0600 (MDT),
        name: 'VisionCamera-20220824_1655345588239996310333588.mp4',
        path: '/data/user/0/<bundle-id>/cache/VisionCamera-20220824_1655345588239996310333588.mp4',
        size: 0,
        isFile: [Function: isFile],
        isDirectory: [Function: isDirectory] } ]
D/TrafficStats: tagSocket(161) with statsTag=0xffffffff, statsUid=-1
D/TrafficStats: tagSocket(161) with statsTag=0xffffffff, statsUid=-1
I/ReactNativeJS: Starting a new recording
```

### Device

Pixel 5

### VisionCamera Version

2.14.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

awcchungster, 2022-08-25:
I get this bug on iPhone 11 too

ShaikRizwan0343, 2022-12-26:
yes i'm getting the same error!!,did you find any solution for this

jaiminbhut, 2023-01-23:
Same error!

izuutech, 2023-05-26:
Did you sort this? Any solution?

Prasanna7254, 2023-08-02:
Wipe the Data on the Device using Android Studio.
Go to Device Manager in the Android Studio, and in the Action section, you will see this option.

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

closed 2023-09-30 by mrousavy